raspibolt/raspibolt #663: welcome message script electrs version

bmml1, 2020-11-26:
Hi,

after upgrading the electrs to 0.8.6 the script somehow does not work for the versioning of electrum.

The request to the electrum server from the script is:

`electrspi=$(echo '{"jsonrpc": "2.0", "method": "server.version", "id": 0}' | netcat 127.0.0.1 50001 -q 1 | jq -r '.result[0]' | awk '{print "v"$2}')`

Error from electrs states:
`electrs[11664]: 2020-11-26T12:05:54.347+00:00 - WARN - rpc #0 server.version [] failed: Error: invalid params: []`

Seems as if the server expects an params array?

https://electrum.readthedocs.io/en/latest/protocol.html?highlight=server.version#server-version

@lucasmoten Any idea?

bmml1, 2020-11-26:
I think this works:

`test=$(echo '{ "jsonrpc": "2.0", "method": "server.version", "params": [ "4.0.6", "1.4" ], "id": 0 }' | netcat 127.0.0.1 50001 -q 1 | jq -r '.result[0]' | awk '{print "v"$2}')`

`echo $test
v0.8.6
`

This commit is probably the reason
https://github.com/romanz/electrs/commit/8f2f53303a62321e3ccd1a8dc42b46c63629a03f

lucasmoten, 2020-12-02:
Confirmed an issue with the script for Electrs 0.8.6 forward

Stadicus, 2020-12-13:
Thanks for your contribution @bmml1 & the code review @lucasmoten! :+1:

closed 2020-12-13 by Stadicus