raspibolt/raspibolt #664: Unable to connect to Bitcoin Explorer

bvrulez, 2020-11-27:
[This](https://stadicus.github.io/RaspiBolt/raspibolt_55_explorer.html) is the page where you explain how to set up Bitcoin Explorer.

I am running it and get this output (notice I have a different hostname). It seems to connect and serve:

```
btcrpcexplorer@RPi4:~/btc-rpc-explorer $ npm run start

> btc-rpc-explorer@2.0.1 start /home/btcrpcexplorer/btc-rpc-explorer
> node ./bin/www

Loading env file: /home/btcrpcexplorer/btc-rpc-explorer/.env
  www Set max_old_space_size to 1024 MB +0ms
  www Express server listening on 127.0.0.1:3002 +22ms
  btcexp:app Starting BTC RPC Explorer, v2.0.1 (commit: '6d70d916e5', date: 2020-05-28) +0ms
  btcexp:app Connecting to RPC node at 127.0.0.1:8332 +0ms
  btcexp:app Verifying RPC connection... +31ms
  btcexp:app Loading mining pools config +7ms
  btcexp:app RPC Connected: version=200100 subversion=/Satoshi:0.20.1/, parsedVersion(used for RPC versioning)=0.20.1, protocolversion=70015, chain=main, services=[NETWORK, WITNESS, NETWORK_LIMITED] +69ms
  btcexp:app Loading historical data for chain=main +0ms
  btcexp:app Skipping performance-intensive task: fetch UTXO set summary. This is skipped due to the flag 'slowDeviceMode' which defaults to 'true' to protect slow nodes. Set this flag to 'false' to enjoy UTXO set summary details. +1ms
  btcexp:app Skipping performance-intensive task: fetch last 24 hrs of blockstats to calculate transaction volume. This is skipped due to the flag 'slowDeviceMode' which defaults to 'true' to protect slow nodes. Set this flag to 'false' to enjoy UTXO set summary details. +1ms

```

My firewall is open for this port:

```
admin@RPi4:~ $ sudo ufw status
Status: active

To                         Action      From
--                         ------      ----
50002                      ALLOW       Anywhere                   # allow Electrum SSL
22                         ALLOW       Anywhere                   # allow SSH
3002                       ALLOW       192.168.100.0/24           # BTC RPC Explorer
50002 (v6)                 ALLOW       Anywhere (v6)              # allow Electrum SSL
22 (v6)                    ALLOW       Anywhere (v6)              # allow SSH

```

But I can not connect with Firefox using `http://192.168.100.220:3002/`.

My `.env` is this (I removed the Passwort). The server is headless (Raspberry Pi 4):

```
btcrpcexplorer@RPi4:~/btc-rpc-explorer $ cat .env
# Optional logging settings, uncomment one of these example lines or add your own.
# See comments after each for more info.
# Default: "btcexp:app,btcexp:error"
#DEBUG=*  # Enable all logging, including middleware
#DEBUG=btcexp:*  # Enable all logging specific to btc-rpc-explorer
#DEBUG=btcexp:app,btcexp:error  # Default
DEBUG=btcexp:app,btcexp:error,www

# The active coin. Only officially supported value is "BTC".
# Default: BTC
#BTCEXP_COIN=BTC

# Host/Port to bind to
# Defaults: shown
#BTCEXP_HOST=127.0.0.1
# Ben
BTCEPX_HOST=0.0.0.0
BTCEXP_PORT=3002

# Bitcoin RPC Credentials (URI -OR- HOST/PORT/USER/PASS)
# Defaults:
#   - [host/port]: 127.0.0.1:8332
#   - [username/password]: none
#   - cookie: '~/.bitcoin/.cookie'
#   - timeout: 5000 (ms)
#BTCEXP_BITCOIND_URI=bitcoin://rpcusername:rpcpassword@127.0.0.1:8332?timeout=10000
BTCEXP_BITCOIND_HOST=127.0.0.1
BTCEXP_BITCOIND_PORT=8332
BTCEXP_BITCOIND_USER=RPi4
BTCEXP_BITCOIND_PASS=
#BTCEXP_BITCOIND_COOKIE=/path/to/bitcoind/.cookie
BTCEXP_BITCOIND_RPC_TIMEOUT=10000

# Select optional "address API" to display address tx lists and balances
# Options: electrumx, blockchain.com, blockchair.com, blockcypher.com
# If electrumx set, the BTCEXP_ELECTRUMX_SERVERS variable must also be
# set.
# Default: none
#BTCEXP_ADDRESS_API=(electrumx|blockchain.com|blockcypher.com)
BTCEXP_ADDRESS_API=electrumx

# Optional ElectrumX Servers. See BTCEXP_ADDRESS_API. This value is only
# used if BTCEXP_ADDRESS_API=electrumx
#BTCEXP_ELECTRUMX_SERVERS=tls://electrumx.server.com:50002,tcp://127.0.0.1:50001,...
BTCEXP_ELECTRUMX_SERVERS=tcp://127.0.0.1:50001

# Set number of concurrent RPC requests. Should be lower than your node's "rpcworkqueue" value.
# Note that Bitcoin Core's default rpcworkqueue=16.
# Default: 10
#BTCEXP_RPC_CONCURRENCY=10

# Disable app's in-memory RPC caching to reduce memory usage
# Default: false (i.e. in-memory cache **enabled**)
#BTCEXP_NO_INMEMORY_RPC_CACHE=true

# Optional redis server for RPC caching
# Default: none
#BTCEXP_REDIS_URL=redis://localhost:6379

# Default: hash of credentials
#BTCEXP_COOKIE_SECRET=0000aaaafffffgggggg

# Whether public-demo aspects of the site are active
# Default: false
#BTCEXP_DEMO=true

# Set to false to enable resource-intensive features, including:
# UTXO set summary querying
# (default value is true, i.e. resource-intensive features are disabled)
#BTCEXP_SLOW_DEVICE_MODE=false

# Privacy mode disables:
# Exchange-rate queries, IP-geolocation queries
# Default: false
BTCEXP_PRIVACY_MODE=true

# Don't request currency exchange rates
# Default: true (i.e. no exchange-rate queries made)
#BTCEXP_NO_RATES=true

# Password protection for site via basic auth (enter any username, only the password is checked)
# Default: none
#BTCEXP_BASIC_AUTH_PASSWORD=mypassword

# Enable to allow access to all RPC methods
# Default: false
BTCEXP_RPC_ALLOWALL=false

# Custom RPC method blacklist
# Default: (see config.js)
#BTCEXP_RPC_BLACKLIST=signrawtransaction,sendtoaddress,stop,...

# Optional API keys
# Default: none
#BTCEXP_GANALYTICS_TRACKING=UA-XXXX-X
#BTCEXP_SENTRY_URL=https://00000fffffff@sentry.io/XXXX
#BTCEXP_IPSTACK_APIKEY=000000fffffaaaaa

# Optional value for "max_old_space_size"
# Default: 1024
#BTCEXP_OLD_SPACE_MAX_SIZE=2048

# Show tools list in a sub-nav at top of screen
# Default: true
BTCEXP_UI_SHOW_TOOLS_SUBHEADER=true
```

I don't know if it is related, but regarding the Electrum wallet, I also can not connect to it via clearnet. It just works over the onion address.

lucasmoten, 2020-12-02:
This part here in the .env file 
```
# Host/Port to bind to
# Defaults: shown
#BTCEXP_HOST=127.0.0.1
# Ben
BTCEPX_HOST=0.0.0.0
BTCEXP_PORT=3002
```
I believe the default was 127.0.0.1 which means it is only bound to local interface. 
With the 0.0.0.0 it should be binding to all allowing external access with your ufw rules.
Have you restarted it since making the change to .env and your ufw rules?

bvrulez, 2020-12-02:
thanks for your reply. you are right with your assumptions, I think the same. I had restarted, but it had no effect.

dominicus, 2021-03-06:
The guide [adds a firewall rule](https://stadicus.github.io/RaspiBolt/raspibolt_55_explorer.html#firewall) to allow calls to Block Explorer from local network:
`sudo ufw allow from 192.168.0.0/16 to any port 3002 comment 'allow BTC RPC Explorer from local network'`
If your gateway assigns IP addresses to local devices using a prefix different from `192.168.0.xxx`, you will not be able to connect using the above rule.
To fix the issue, the command to add the firewall rule needs to use the IP prefix matching your local network.

bvrulez, 2021-03-06:
@dominicus Thanks, but of course I adjusted this to my local LAN.

Stadicus, 2021-11-27:
If you didn't solve this issue yet: check out the updated RaspiBolt version 3. The NGINX reverse proxy config and BTC RPC Explorer are now part of the main guide and well tested. 

Closing this issue for now. Feel free to reopen it if further assistance is required.

closed 2021-11-27 by Stadicus