raspibolt/raspibolt #665: Unable to connect BitBox App via TOR or clearnet to Electrs server

bvrulez, 2020-11-27:
I tried to use the BitBoxApp with the `bla.onion:50002` address from my local Electrs server in `/var/lib/tor/hidden_service_electrs/hostname`. 

It is in the tutorial [here](https://stadicus.github.io/RaspiBolt/raspibolt_50_electrs.html#connect-bitboxapp): 

I get: 

```
Fehlgeschlagen:
dial tcp: lookup bla.onion: no such host
```

My proxy works. I know this because I can connect to `localhost:9050` with the browser and I get a message that this is a proxy for SOCKS, not HTTP. So this should be fine. I also tried with a running TOR browser and proxy over port `9150`.

But I can't connect to the Electrs server with BitBox. 

An Electrum client is on the same Mac and is connected successfully via TOR at the same time. 

I also tried clearnet with `local IP:50002` and it was successfull in downloading the remote certificate for TLS. Where is this certificate loaded from? From the Electrs server? So there must be a connection? 

But connecting ultimately failed due to a response timeout. 

```
Fehlgeschlagen:

response timeout
``` 

So I can not add my local Electrs server to BitBoxApp.

Stadicus, 2020-12-13:
Hm, hard to tell what's the issue. You already covered the basics by making sure Tor works correctly on your Mac, and that you can connect Electrum to the RaspiBolt over Tor as well.

When adding the onion to the BitBoxApp, did you also try to download the SSL certificate first? 

Using SSL with Tor is acutally redundant and since I published the tutorial, the BitBoxApp now also supports connecting without SSL. Can you try the following?

* add the direct port `50001` to Tor in `/etc/tor/torrc`:
  ```
  HiddenServiceDir /var/lib/tor/hidden_service_electrs_nossl/
  HiddenServiceVersion 3
  HiddenServicePort 50002 127.0.0.1:50001
  ```
* Try to connect to this new onion in `sudo cat /var/lib/tor/hidden_service_electrs_nosssl/hostname` with the BitBoxApp:
  1. enter onion address
  2. click directly on "Check" and then "Add" (skipping the "download remote cert" part)

bvrulez, 2020-12-13:
I had the same problem from a Ubuntu 20.04 within a VM on a Windows host. I can connect with Electrum wallet app over tor proxy localhost:9050 to both of my local full nodes. I was not able to connect from the BitBox App. Connection times out as above.

But: **Downloading the certificat and then validating the connection surprisingly works here.** I am pretty sure it did not work 16 days ago on the Mac. I remember beeing only able to receive a certificate via clearnet. 

I had to restart the BitBox app after activating the tor proxy. Might be good to add a hint to do that. 

I will check again on the Mac. Also with the direct port suggestion. Thanks!

bvrulez, 2020-12-13:
After updating to the current version of the BitBox app in **macOS I am now also able to download the certificate** and then validate and add the server (my own full node electrs). 

I just realized that I need an actual BitBox to use this app. :)

closed 2020-12-13 by bvrulez

Stadicus, 2020-12-14:
> I just realized that I need an actual BitBox to use this app. :)

Oh, that's a good point. I could stress that a bit more, I guess... :-D

bvrulez, 2020-12-14:
Maybe I will get one for the multisig. This is probably the future. For now I am very glad having two Raspis running and preparing to sweep two old wallets onto a Ledger. There is a lot to know even for such a basic setup already.

Stadicus, 2020-12-15:
Have a look at this in regards to multisig. It's great if done properly, but most hardware wallets make it less secure than just using singlesig: https://shiftcrypto.ch/blog/how-nearly-all-personal-hardware-wallet-multisig-setups-are-insecure/

bvrulez, 2020-12-15:
Yes, I meant, if I ever want to use multisig (what will probably be the case in the future) I will buy a BitBox. :)