raspibolt/raspibolt #669: Consider transitioning next release to 64 Bit

lucasmoten, 2020-12-02:
As reported by opennoms in https://github.com/lightningnetwork/lnd/issues/4811, as Lightning Node database grows in size, variant depending upon usage, eventually it will reach a point where it is too large for 32 bit process to handle.  For some reason this threshold on the file size is around 1GB in size.  Once the threshold is past, compaction requires running on a 64 bit platform.

Prior to hitting the limits, the database can be compacted following guidance opennoms has summarized here 
https://openoms.gitbook.io/lightning-node-management/lnddatabasecompaction

lucasmoten, 2020-12-03:
The 64bit lite version of Raspberry Pi OS is available here
https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2020-08-24/

2pac1, 2020-12-11:
I cant figure out has 64bit came out of beta or not?
If it did (which I doubt), why is it not available on the main download section.

Switching to 64bit for the guide is good plan.

Stadicus, 2020-12-13:
That's a good input, thanks @lucasmoten. I actually considered switching to the 64bit Ubuntu server image, but it's quite a pain to setup the most basic things. 

Will have a look at the Raspi OS 64bit for the next larger overhaul.

raelschiffler, 2021-03-18:
Anxiously waiting for the tutorial to upgrade Raspi OS to 64bit. Hope I don't need to close any of my channels to perform this upgrade.

Covalent5, 2021-03-19:
I've successfully transitioned to the 64bit Ubuntu Server. I am in the process of writing the necessary changes that need to happen to achieve a successful installation of all parts.

Swissnode, 2021-05-28:
> 
> 
> That's a good input, thanks @lucasmoten. I actually considered switching to the 64bit Ubuntu server image, but it's quite a pain to setup the most basic things.
> 
> Will have a look at the Raspi OS 64bit for the next larger overhaul.

this would be amazing... not only that but also how to transition from a 32bit running system to a 64bit one, is it a simple case of setting up a second node and copying over the .lnd folder? Be GREAT to have a guide as simple as the rest of your wonderful guide!!

openoms, 2021-05-28:
Yes, the transition only means that to build a new 64 bit system and copy the lnd folder over to the new node. 

Would make sure everything works on the new system by creating (and disposing) an empty wallet first.

Also make sure to not run the lnd database on two places at the same time.

```
sudo system stop lnd
sudo systemctl disable lnd
```
before copying the folder over to stop it from restarting.

An lnd node is not active until unlocked with the password.

The latest (v1.7) release of the RaspiBlitz is running with the 64bit Raspberry OS: https://downloads.raspberrypi.org/raspios_arm64/images/ without any issues so far.

See the thread: https://github.com/rootzoll/raspiblitz/issues/1199

JssDWt, 2021-06-15:
> Yes, the transition only means that to build a new 64 bit system and copy the lnd folder over to the new node.
> 
> Would make sure everything works on the new system by creating (and disposing) an empty wallet first.
> 
> Also make sure to not run the lnd database on two places at the same time.
> 
> ```
> sudo system stop lnd
> sudo systemctl disable lnd
> ```
> 
> before copying the folder over to stop it from restarting.
> 
> An lnd node is not active until unlocked with the password.
> 
> The latest (v1.7) release of the RaspiBlitz is running with the 64bit Raspberry OS: https://downloads.raspberrypi.org/raspios_arm64/images/ without any issues so far.
> 
> See the thread: [rootzoll/raspiblitz#1199](https://github.com/rootzoll/raspiblitz/issues/1199)

According to docs.lightning.engineering, copying the lnd folder between different architectures is untested and should be avoided:
https://docs.lightning.engineering/lightning-network-tools/lnd/safety#migrating-a-node-to-a-new-device

Are you sure this will work?

openoms, 2021-06-15:
@JssDWt from 32bit arm to 64bit arm (as the situation was the same with my raspiblitz nodes) I have done it multiple times without any problems.

martinatime, 2021-06-16:
This node has accomplished this feat https://www.reddit.com/r/TheLightningNetwork/comments/n45gsr/cornelius_is_now_64bit_i_feel_like_i_just_did/

xanoni, 2021-07-26:
I'm still on 32 bit. Curious, what practical advantages does the 64 bit upgrade bring for our use case, besides the ability to handle larger `channel.db` files? 

I'm not afraid of the upgrade, but trying to understand if it's worth it. There's still room in my `channel.db` :)

martinatime, 2021-07-27:
I can tell you that my node has had its channel.db grow about 10GB in less than a week. At this rate I'm going to have to upgrade in a few months or risk losing my node. I'll probably be following the link that I posted before sooner rather than later.

I believe Raspbian has only had a 64bit version for a short time and it is still in beta https://www.raspberrypi.org/forums/viewtopic.php?t=275370

There are 64bit OSes for the RPi such as DietPi.

xanoni, 2021-07-28:
> I can tell you that my node has had its channel.db grow about 10GB in less than a week. At this rate I'm going to have to upgrade in a few months or risk losing my node. I'll probably be following the link that I posted before sooner rather than later.

Maybe you could try this ... I don't think it fixes everything, but it should get a bit better.

```
 1043 ; If set to true, then we'll prune a channel if only a single edge is seen as
 1044 ; being stale. This results in a more compact channel graph, and also is helpful
 1045 ; for neutrino nodes as it means they'll only maintain edges where both nodes are
 1046 ; seen as being live from it's PoV.
 1047
 1048 routing.strictgraphpruning=true
```

raelschiffler, 2021-08-16:
Just finished upgrading to Raspberry OS 64-Bit (Buster).
Bitcoin Core and LND running smooth, I even managed to migrate to LiT thanks to the new tutorial.
Only issue is Electrs, it's not building. I opened an issue in their project explaining better: https://github.com/romanz/electrs/issues/457

I'd like to share something I did: the 1 TB SSD I had running somehow decreased it's speed to ~12 MB/sec. I even tested it with the SATA cable on my desktop and the speed was still the same, so I went to the "Update External Drive" page in the Bonus Section. I checked my office (and my pocket) and realised I had 2 extra SSDs with 512 GB capacity each and no money to buy a good quality 1 TB new drive, so I wrote a shell script to sync half of the files from the old drive to each new drive and then create symbolic links to those files in the second new one, making everything look like it's only in one place, but physically spread in two drives. It worked, but bitcoind accused the DB was corrupted, so I had to reindex everything. It didn't consume internet data, but took 5 days to reindex. Anyway, it saved me a few bucks and I'm happy it worked. 
Another thing is that  Raspy couldn't power both SSDs at the same time, fortunately I also had in my office an externally powered USB 3.0 hub. Speeds tested, got ~300 MB/s in each.
If anyone could check the issue I opened to Electrs, any help is welcome.

xanoni, 2021-08-17:
> I'd like to share something I did: the 1 TB SSD I had running somehow decreased it's speed to ~12 MB/sec.

Have you tried `$ sudo fstrim -v /dev/sdX` or just `$ sudo fstrim -av`?

```bash
# not getting much bang because cron does this all the time
$ sudo fstrim -va
/mnt/ssd: 0 B (0 bytes) trimmed on /dev/mapper/ssd
/mnt/old_boot: 204 MiB (213933056 bytes) trimmed on /dev/mmcblk0p1
/boot: 211.5 MiB (221756928 bytes) trimmed on /dev/sda1
/mnt/old_root: 0 B (0 bytes) trimmed on /dev/mapper/old_root
```

It's a bit tricky because many drives (typically if connected via USB) need some extra massaging to accept the SCSI commands over USB. See my post here: https://github.com/rootzoll/raspiblitz/issues/1805#issuecomment-892380571

There's also a systemd script that triggers `fstrim` once per week:
```bash
/usr/lib/systemd/system/fstrim.service
/usr/lib/systemd/system/fstrim.timer
```

I believe you turn that on via `sudo systemctl enable fstrim.timer`.

raelschiffler, 2021-08-19:
> Have you tried `$ sudo fstrim -v /dev/sdX` or just `$ sudo fstrim -av`?

Thank you for answering.
I believe something's missing:

> admin ~ ₿ sudo fstrim -v /dev/sdb
> [sudo] password for admin: 
> fstrim: /dev/sdb: not a directory
> admin ~ ₿ sudo fstrim -v /dev/sdb1
> fstrim: /dev/sdb1: not a directory
> admin ~ ₿ sudo mount -a
> admin ~ ₿ lsblk -o NAME,MOUNTPOINT,UUID,FSTYPE,SIZE,LABEL,MODEL
> NAME        MOUNTPOINT   UUID                                 FSTYPE   SIZE LABEL  MODEL
> sda                                                                    477G        USB3.0
> └─sda1      /mnt/extNVME XXXXXXXXXXXXXXXXXXXXXXXXXXXXXX     477G        
> sdb                                                                  953.9G        D_1TB     **### <--This is the lazy guy**
> └─sdb1                   XXXXXXXXXXXXXXXXXXXXXXXXXXXXXX ext4   953.9G        
> sdc                                                                  465.8G        USB3.0
> └─sdc1      /mnt/ext     XXXXXXXXXXXXXXXXXXXXXXXXXXXXXX ext4   465.7G        
> mmcblk0                                                               29.8G        
> ├─mmcblk0p1 /boot        XXXXXXXXXXXXXXXXXXXXXXXXXXXXXX                            vfat     256M boot   
> └─mmcblk0p2 /            XXXXXXXXXXXXXXXXXXXXXXXXXXXXXX ext4    29.6G rootfs 
> 
After editing FSTAB to auto-mount the drive:

> admin ~ ₿ sudo mount -a
> admin ~ ₿ lsblk -o NAME,MOUNTPOINT,UUID,FSTYPE,SIZE,LABEL,MODEL
> NAME        MOUNTPOINT   UUID                                 FSTYPE   SIZE LABEL  MODEL
> sda                                                                    477G        USB3.0
> └─sda1      /mnt/extNVME XXXXXXXXXXXXXXXXXXXXXXXXXXXXXX ext4     477G        
> sdb                                                                  953.9G        D_1TB
> └─sdb1      /mnt/extOLD  XXXXXXXXXXXXXXXXXXXXXXXXXXXXXX ext4   953.9G        
> sdc                                                                  465.8G        USB3.0
> └─sdc1      /mnt/ext     XXXXXXXXXXXXXXXXXXXXXXXXXXXXXX ext4   465.7G        
> mmcblk0                                                               29.8G        
> ├─mmcblk0p1 /boot        XXXXXXXXXXXXXXXXXXXXXXXXXXXXXX                            vfat     256M boot   
> └─mmcblk0p2 /            XXXXXXXXXXXXXXXXXXXXXXXXXXXXXX ext4    29.6G rootfs 
> admin ~ ₿ sudo fstrim -v /dev/sdb1
> fstrim: /dev/sdb1: not a directory
> admin ~ ₿ sudo fstrim -v /dev/sdb
> fstrim: /dev/sdb: not a directory
> admin ~ ₿ sudo fstrim -av
> /boot: 222.1 MiB (232928256 bytes) trimmed on /dev/mmcblk0p1
> /: 21 GiB (22503153664 bytes) trimmed on /dev/mmcblk0p2
> admin ~ ₿ sudo fstrim -v /mnt/extOLD 
> fstrim: /mnt/extOLD: the discard operation is not supported
>

raelschiffler, 2021-08-19:
The SSD-USB3 case I'm using was preventing the system from executing fstrim. I connected the drive to the SATA cable in my desktop and the command took a few minutes to execute as follows:

```
~$ sudo fstrim -v /media/linus/XXXXXXXXXXXXXXXXXXXXXXXXXXXXXX/
/media/linus/XXXXXXXXXXXXXXXXXXXXXXXXXXXXXX/: 444,9 GiB (477659525120 bytes) trimmed

```
Anyway, speed is still mediocre, although it's 3x what it used to be:
```
$ sudo hdparm -t --direct /dev/sdc

/dev/sdc:
 Timing O_DIRECT disk reads: 100 MB in  3.00 seconds =  33.33 MB/sec
```

xanoni, 2021-08-19:
> Anyway, speed is still mediocre, although it's 3x what it used to be

@raelschiffler I assume it's just worn down beyond repair then. How much free space? I'm very curious if speeds would go up massively once more data is deleted and you `fstrim` again. Theory suggests there could be a big speed increase, but SSD controllers are weird and unpredictable in practice.

Maybe even reformat the drive (`mkfs.ext4 /dev/sdc`) if you already migrated all the data to the new disks ...

raelschiffler, 2021-08-20:
@xanoni YOU WERE RIGHT!!!!
Formatting and re-fstrimming recovered the normal speed! 
Now how can I use the scheduled `fstrim` if the enclosure does not support it?

raelschiffler, 2021-08-20:
CAN YOU BELIEVE IT TRIMMED ALMOST 1TB???

$ sudo fstrim -v /media/linus/XXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
[sudo] password for linus: 
/media/linus/XXXXXXXXXXXXXXXXXXXXXXXXXXXXXX: 937,9 GiB (1006990123008 bytes) trimmed

raelschiffler, 2021-08-20:
I managed to allow the enclosure to trim the drive by following https://www.glump.net/howto/desktop/enable-trim-on-an-external-ssd-on-linux tutorial.
Very easy, anyone should try to avoid speed and data loss using a raspy.

xanoni, 2021-08-20:
> Very easy, anyone should try to avoid speed and data loss using a raspy.

Yeah man, `trim` / `unmap` / `discard` is a bit of trouble with USB-connected drives (not sure why the Linux Kernel doesn't automatically do this for popular devices), but it is very much worth it!

I suggest RaspiPi adds udev settings/configuration/scripts at least for the recommended SSDs (SanDisk, Samsung, ...).

Stadicus, 2021-11-27:
I think the original topic of this issue has been taken care of with the recent update to RaspiBolt v3. I'm closing this issue, but feel free to reopen it if you think it's still unsolved.

closed 2021-11-27 by Stadicus