raspibolt/raspibolt #671: Shutdown Gracefully

Covalent5, 2020-12-08:
Can anyone tell me how can I gracefully shutdown my node for relocation?

Many thanks in advance 🖖🏻

lucasmoten, 2020-12-08:
Shut down services ...  You may not have all these, but this is the order i do for my node

```bash
$ sudo systemctl stop runthenumbers.service
$ sudo systemctl stop ridethelightning.service
$ sudo systemctl stop lnd.service
$ sudo systemctl stop btcrpcexplorer.service
$ sudo systemctl stop electrs.service
$ sudo systemctl stop bitcoind.service
```

And then shut down
```bash
$ sudo shutdown now
```

Covalent5, 2020-12-08:
Thanks for the reply. So just using `sudo shutdown -h now` won't stop all services running?

2pac1, 2020-12-11:
It is not recommended that you do shutdown -h before you stopped bitcoind and electrs.
I presume that stopping the service as per post above would be the same as dong:
`bitcoin-cli stop`
While you do that, you monitor the log with:
`sudo tail -f /mnt/ext/bitcoin/debug.log` so you verify that it has been shut down cleanly.

Only then you power down the pi.

Covalent5, 2020-12-15:
Awesome. Thank you 🖖🏻

closed 2020-12-15 by Covalent5

uskrusader, 2022-03-16:
> Shut down services ... You may not have all these, but this is the order i do for my node
> 
> ```shell
> $ sudo systemctl stop runthenumbers.service
> $ sudo systemctl stop ridethelightning.service
> $ sudo systemctl stop lnd.service
> $ sudo systemctl stop btcrpcexplorer.service
> $ sudo systemctl stop electrs.service
> $ sudo systemctl stop bitcoind.service
> ```
> 
> And then shut down
> 
> ```shell
> $ sudo shutdown now
> ```

I just finished the raspibolt guide and I need to shut it all down to plug in a UPS/backup battery.
Should I also use the commands above? (all except runthenumbers, I don't believe that was in the most recent raspibolt guide)

.....and when I power it back up, do I simply reverse the order of commands and replace "stop" with "restart"?

thanks!