raspibolt/raspibolt #679: Systemd and BitcoinD Autostart on boot > exit error on 64bit raspbian

2pac1, 2020-12-18:
Hey guys, 

 After wasting my whole day I am sharing this:
 If you use the 64bit beta, you won't be able to setup systemd following the guide.
 At least I was not. I got back to 32bit OS, works like a charm.
 Thought I share.

2pac1, 2020-12-19:
bitcoind ran manually worked, it just didn't want to start from systemd.
Maybe I messed up something with paths as I was booting from SSD, but I tried multiple times from scratch.

Stadicus, 2020-12-19:
Thanks for your feedback. I plan to move over to 64bit Raspbian for the next major overhaul, so it's good to know what to expect... :D

2pac1, 2020-12-19:
Wasted whole day on this.
It's definitely that. Once i switched back to 32bit and this time I created
/name/ext folder which I used for symbolic linking, so I follow guide more,
it worked right away.

LightningTwo, 2021-01-06:
Hello, I am currently following the guide using Raspberry Pi OS 64 bit, and have no problems with starting bitcoind at boot.
I am booting directly from the SSD (no SD card inserted), put the `/ext` folder directly in root (`/`), and changed all `/mnt/ext` in this guide to `/ext`. I installed the 64 bit version of Bitcoin Core. It is now syncing the blockchain. If I encounter any problems installing lnd, electrs, etc I will report it here.

Stadicus, 2021-11-27:
I think the original topic of this issue has been taken care of with the recent update to RaspiBolt v3. I'm closing this issue, but feel free to reopen it if you think it's still unsolved.

closed 2021-11-27 by Stadicus