raspibolt/raspibolt #681: Slow down on progress towards the end - is it normal?

dellagustin, 2020-12-20:
Hello All, I have been running the node for Initial Block Download for more or less 4 day now.
The Pi is connected through cable to my modem, I have 150Mb connection and a 1TB HDD, the HDD speed test gave me aroun 100MB/sec.
The progress reached 0.886212 now, but I have the impression it is progressing slower over time.
Is that normal? If yes, might worth mentioning in the guide.

It would be also good to have a section on how to do the IDB in a more powerful PC then copy it to the PI (over net or directly to the connected storage drive).

Pantamis, 2020-12-21:
It is hard to tell because it depends of many things. But it is commonly reported.

Maybe you just flushed the UTXO set from RAM to HDD. In that case, Core must read back all the UTXO needed to validate transaction and update the UTXO set from the HDD, which is very slow. If you follow the guide, you should have extend dbcache to 2000MB to reduce this issue, you can't really do better than doing the IBD on another machine with more RAM indeed.

Maybe the average size of blocks starts to increase a lot because of Segwit adoption where you are currently in the Ledger. This implies bigger transactions and blocks are harder to process.

Finally, the most plausible explanation is that you start to process blocks which are after the one you assumed valid. In Core 0.20.0, it is the block 0000000000000000000f2adce67e49b0b6bdeb9de8b7c3d7e93b21e7fc1e819d at height 623950. All blocks after this one are not assumed valid, you must validate all transactions (and checks all signatures and all scripts in particular). This parameter is set by default in all Core release and is moved forward at each release by Core devs (to a block deep enough in the blockchain). You can change it for a more recent block at your own risk (which is very low honestly).

dellagustin, 2020-12-22:
@Pantamis thank you for the detailed answer.
I don't know for sure anymore, but I have the impression it had already slowed down before height 623950.
I did set the dbcache to 2000MB.
I have a Pi with 8GB of ram, should it help if I increased the cache to 6BG or something like that? I it already finished now, but I can try that if I have to do it again.

Pantamis, 2020-12-22:
It is highly possible that at the time you looked at the progress bar, Core was flushing UTXOs to disk so making no progress for a signifiant amount of time. This could give the impression it slows down... Anyway whatever

With 8GB RAM you can safely set the dbcache to 6-6.5 GB (7 GB if you like to live dangerously haha, that's my setup when I reindex on my laptop with 8GB RAM)

Now that you are synced with mainnet network no need to change the setting anymore. However, if you have to reindex because of database corruption (chainstate or block index) it will definitively be useful to do it with high dbcache value. Use an UPS to prevent power outage if you can (this is really an embarassing situation when it happens).

However, I am not sure if you can use more than 2GB RAM when running a 32-bit OS (maybe this is totally irrelevant, I just doubt)

Stadicus, 2021-11-27:
Closing this issue due to inactivity. Feel free to reopen it if further assistance is required.

closed 2021-11-27 by Stadicus