raspibolt/raspibolt #685: Update Node JS to LTS = that is 14 right now

2pac1, 2020-12-26:
When installing btc explorer, there is old LTS Node JS version (12) in the guide:
Current: curl -sL https://deb.nodesource.com/setup_12.x | bash -

Proposed change: curl -sL https://deb.nodesource.com/setup_14.x | bash -

LightningTwo, 2021-01-14:
I installed BTC RPC Explorer with NodeJS 14 and did not encounter any problems so far.
You could also change the command to `curl -sL https://deb.nodesource.com/setup_lts.x | bash -`
which always gets the latest LTS version.

Stadicus, 2021-11-27:
I think the original topic of this issue has been taken care of with the recent update to RaspiBolt v3. I'm closing this issue, but feel free to reopen it if you think it's still unsolved.

closed 2021-11-27 by Stadicus