raspibolt/raspibolt #686: bitcoind autostarter AFTER network is up

2pac1, 2020-12-26:
Current configs for autoboot: https://stadicus.github.io/RaspiBolt/raspibolt_30_bitcoin.html#autostart-on-boot

After=network.target

Maybe it would be better to change in accordance with
https://www.freedesktop.org/wiki/Software/systemd/NetworkTarget/

to: network-online.target

Would that be better suited for this purpose?

Stadicus, 2021-11-27:
Good point, sorry to come back to this so late... :)

All services are very resilient against starting while the network is not yet ready. I'm not sure that this significantly improves node stability or performance, and would need some testing.

Do you think it's worth it?

2pac1, 2021-11-27:
Let's leave as it is

Stadicus, 2021-11-27:
Ack, and thanks for the quick reply.

closed 2021-11-27 by Stadicus