raspibolt/raspibolt #689: Bitcoin Core 0.21.0

Covalent5, 2021-01-16:
Just upgraded to the latest Bitcoin core version and now my TOR V3 address score is stuck at 4. Also, I get 0 inbound connections. Outbound connections are solid 10. 

Any ideas?

Covalent5, 2021-01-16:
[image]

Is this normal? Bounding to 2 ports on 127.0.0.1?

LightningTwo, 2021-01-16:
I'm not sure, but it could be that there aren't many Tor V3 Bitcoin nodes on the network yet. 
What version of Tor are you running? You can check this with `tor --version`. The latest (stable) version is `0.4.4.6`. There is a problem with ARM 32 bit systems not updating, and are stuck on `0.3.5.x`.

Covalent5, 2021-01-16:
> I'm not sure, but it could be that there aren't many Tor V3 Bitcoin nodes on the network yet.
>
> What version of Tor are you running? You can check this with `tor --version`. The latest (stable) [...]

I am on 0.3.5.12 I think. But before the upgrade I was having non-Tor inbound connections.

Covalent5, 2021-01-16:
It's also barely transmitting data... [image]

LightningTwo, 2021-01-16:
> I am on 0.3.5.12 I think. But before the upgrade I was having non-Tor inbound connections as well.

Are you sure you had inbound connection over ipv4 or ipv6? If you followed this guide, then you only accept inbound connections via Tor. Check your firewall with `sudo ufw status` if you opened port `8333`, and check this on your router too, if you want non-Tor inbound connections.

I just updated by node to `0.21.0` as well, and I'm also not getting inbound connections over Tor yet.

Covalent5, 2021-01-16:
> > I am on 0.3.5.12 I think. But before the upgrade I was having non-Tor inbound connections as well.
> 
> 
> 
> Are you sure you had inbound connections over ipv4 or ipv6? If you followed this guide, then you only accept inbound connections via Tor. Check your firewall with `sudo ufw status` if you opened port `8333`, and check this on your router too, if you want non-Tor inbound connections.
> 
> 
> 
> I just updated my node to `0.21.0` as well, and I'm also not getting inbound connections over Tor yet. 

My bad. You are right. It was all over Tor. So I am guessing we wait until the network adopts 0.21 a bit more then.

Covalent5, 2021-01-17:
> Are you sure you had inbound connections over ipv4 or ipv6? If you followed this guide, then you only accept inbound connections via Tor. Check your firewall with `sudo ufw status` if you opened [...]

Did you get any inbound connections?

lucasmoten, 2021-01-17:
You wouldn't have had entries for bitcoin in /etc/tor/torrc previously 

Note that [Bitcoin 0.21.0 release notes](https://bitcoincore.org/en/releases/0.21.0/) state
* This release adds support for Tor version 3 hidden services, and rumoring them over the network to other peers using [BIP155](https://github.com/bitcoin/bips/blob/master/bip-0155.mediawiki). Version 2 hidden services are still fully supported by Bitcoin Core, but the Tor network will start [deprecating](https://blog.torproject.org/v2-deprecation-timeline) them in the coming months. ([#19954](https://github.com/bitcoin/bitcoin/pull/19954))
* The Tor onion service that is automatically created by setting the `-listenonion` configuration parameter will now be created as a Tor v3 service instead of Tor v2. The private key that was used for Tor v2 (if any) will be left untouched in the `onion_private_key` file in the data directory (see `-datadir`) and can be removed if not needed. Bitcoin Core will no longer attempt to read it. The private key for the Tor v3 service will be saved in a file named `onion_v3_private_key`. To use the deprecated Tor v2 service (not recommended), the `onion_private_key` can be copied over `onion_v3_private_key`, e.g. `cp -f onion_private_key onion_v3_private_key`. ([#19954](https://github.com/bitcoin/bitcoin/pull/19954))

It's possible that TOR

lucasmoten, 2021-01-17:
For your outbound connections, can you check to see if any of them are onioni v3 address types?

Covalent5, 2021-01-17:
> For your outbound connections, can you check to see if any of them are onioni v3 address types? 

None. All are V2 or ip4 and ip6

svpi11, 2021-01-17:
Same problem here... All is working fine, except 0 connections in. I think, as others said, that my node is not finding enough Tor V3 others nodes yet probably.

Because when I connect to my Bisq software through only my own node, it appears "1" connection in. So apparently all is working fine, just not enough nodes yet.

florianfink, 2021-02-26:
I have the exact same issue. Did anyone solve this?

Covalent5, 2021-03-01:
Nope. But I tested and people can connect with no problem as long as they have my .onion address.

raelschiffler, 2021-03-18:
Same problem here. Bitnodes.io sees and connects to my node via onion but no incoming connectios so far since upgraded to 0.21.0. I can connect to some 0.21.0 nodes but none connects to me...

2pac1, 2021-05-29:
Same here. So far 10 peers. (bitcoin-cli getconnectioncount)
Bitnodes see and can connect via onion.
Let's see after few days what happens.

gabmontes, 2021-06-16:
Same problem here. Bitnodes can connect to the onion address but not to the public IP.

What is required to receive non-Tor incoming connections? Ports are open in `ufw` and port forwarding is properly set in the router.

2pac1, 2021-06-17:
For you to receive direct connection via IP and not via TOR, you need to
have 8333 opened in ufw and you need to have that port forwarded on your
router. But why would you do that?
It's more secure and equally efficient to have it only via TOR and port not
opened, as any port that you do not need to open, you don't open it

raelschiffler, 2021-06-17:
I agree with @2pac1. Moreover, I just found out that in a few countries, mine included, ISPs won't forward connections that weren't started from your modem, no matter if the doors are opened or not. 
As TOR connections are started from your local, it's only an encrypted stream to your ISP, only door-resolved locally in your TOR daemon, therefore they will work. At least works to me, although the System Overview screen will show my node as not reachable publicly, but I have incoming connections through TOR.
Blame the ISPs and their politics to avoid having a server running in clients homes.

Stadicus, 2021-11-27:
I think the original topic of this issue has been taken care of with the recent update to RaspiBolt v3, updated Tor configuration and the latest Bitcoin Core version. I'm closing this issue, but feel free to reopen it if you think it's still unsolved.

closed 2021-11-27 by Stadicus