raspibolt/raspibolt #691: Link to Raspbian Buster Lite is broken

arminmayrhofer, 2021-01-19:
The link contained in "Download the Raspbian Buster Lite disk image" on https://stadicus.github.io/RaspiBolt/raspibolt_20_pi.html#preparing-the-operating-system is broken.

fatrattombala, 2021-01-19:
The Raspberry Pi Foundation now recommends installing Raspberry Pi OS (previously called Raspbian) via the Raspberry Pi Imager. Raspian Buster Lite is now called Rasperry Pi OS Lite (32 Bit), last version was released 2021-01-11.

[image]

Stadicus, 2021-11-27:
I think the original topic of this issue has been taken care of with the recent update to RaspiBolt v3. I'm closing this issue, but feel free to reopen it if you think it's still unsolved.

closed 2021-11-27 by Stadicus