raspibolt/raspibolt #695: [ File '/mnt/ext/bitcoin/bitcoin.conf' is unwritable ]

bhaskarndas, 2021-01-28:
Hello
My whole system ran fine till it was fully synced. Now when I am trying to check the status of the system by running the command I received following issue.

 bitcoin-cli getblockchaininfo
error: Could not connect to the server 127.0.0.1:8332

Make sure the bitcoind server is running and that you are connecting to the correct RPC port.

So I tried to do investigation of bitcoin.conf file by running the following command

sudo nano /mnt/ext/bitcoin/bitcoin.conf

On opening it is showing me the message 
[ File '/mnt/ext/bitcoin/bitcoin.conf' is unwritable ]

lucasmoten, 2021-01-29:
Is the external drive mounted?

bhaskarndas, 2021-01-29:
Yes it does show to me.

Covalent5, 2021-01-29:
Did you recently restated or unplugged the hard drive? If so, it means your system mounted the drive in "read only" mode. 

Try: 

```
$ sudo su - bitcoin
$ cd /mnt/ext
$ touch bitcoin/test.file
$ rm bitcoin/test.file
```

If the above fails try a “refresh” and remount all the file systems listed in fstab:

```
$ sudo mount -a
```

bhaskarndas, 2021-01-30:
Ohh yes it was restarted but there has been several restart and every time it has worked fine. Problem occurred after recently after the complete blockchain was synced. None of the steps above has been helping much. I tried to umount but facing the issue
sudo umount /dev/sda1
umount: /mnt/ext: target is busy.

Covalent5, 2021-01-30:
Make sure stop all serviced before attempting the umount and remount commands.

Stadicus, 2021-11-27:
Closing this issue due to inactivity. Feel free to reopen it if further assistance is required.

closed 2021-11-27 by Stadicus