raspibolt/raspibolt #696: Node Unreachable / No Inbound Peers

penguin4111, 2021-02-03:
I just finished going through the whole RaspiBolt setup guide. The blockchain is completely synced and I have 10 outbound peers, but 0 inbound peers. I have port 8333 enabled on the raspberry pi firewall (even though I dont think the guide specifically pointed out to do this...), and I have port 8333 set up on my router's port forwarding for my raspberry pi's IP address. I am really not sure why I cannot get any inbound peers. Anyone have experience with this or perhaps suggestions of things to check?

Covalent5, 2021-02-05:
Hello there 🖖🏻. Are you on Bitcoin Core Version 0.21.0 by any chance?

penguin4111, 2021-02-05:
> Hello there 🖖🏻. Are you on Bitcoin Core Version 0.21.0 by any chance?
> 
> For the "reachable" issue, type:
> 
> `bitcoin-cli getnetworkinfo`
> 
> Check if the reachable is 'TRUE' on all 3 protocols. Assuming you are using Tor.

Thanks for the reply. Yes, I am running 0.21.0. Getnetworkinfo shows all three protocols as "reachable". I saw your post on reddit and now understand its an issue with Tor v3. I didn't change anything about Tor, but I guess upgrading bitcoin core changes over the Tor version? In any case, maybe you could try connecting to me: glm52zywiqrcxuwswvgjsxr5pfeggil7uci4z5tbpvb4rjyu5hwjhtid.onion

Covalent5, 2021-02-05:
Will do :)

penguin4111, 2021-02-05:
> Can you check!? I connected!

Yes! I see one inbound connection now! Thanks, I feel better now knowing there is nothing wrong with my setup. I wonder how long until we start picking up inbound connections automatically

closed 2021-02-05 by penguin4111

gabmontes, 2021-06-16:
@penguin4111 I'm having the same problem. Did you start receiving incoming connections?