raspibolt/raspibolt #697: Unable to parse settings file /home/admin/.bitcoin/settings.json

mateomar, 2021-02-06:
I am following the RaspiBolt guide. When I initially ran bitcoind, there were no problems. Then I when I got to: "systemctl status bitcoind.service", I got the following output:
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Fri 2021-02-05 03:59:00 GMT; 21s ago
  Process: 552 ExecStart=/usr/local/bin/bitcoind -daemon -pid=/run/bitcoind/bitcoind.pid -conf=/mnt/ext/bitcoin/bitcoin.conf -datadir=/mnt/ext/bitcoin (code=exited, status=0/SUCCESS)
 Main PID: 605 (code=exited, status=1/FAILURE)

Now if I run bitcoind, there is a parse issue because settings.json is empty. 

I saw a similar thread, where your response is linked below. I assume the screenshot is from a different version of the guide, since I can't find that section anywhere. 

Question I have:
Do I need to make any changes to the node?
Where can I find the RaspiBolt instructions for remote downloading and validating the blockchain?

Thank you for creating this guide. It is very detailed and informative. 

May I kindly suggest you go through the guide step-by-step? :-)

[image]

_Originally posted by @Stadicus in https://github.com/Stadicus/RaspiBolt/issues/136#issuecomment-405298495_

lucasmoten, 2021-02-21:
@mateomar I believe the referenced issue #136 is from a version of the guide that was originally developed for Raspberry Pi 3.  The current documentation for setup for a Raspberry Pi 4 is available here https://stadicus.github.io/RaspiBolt/

Looking at your service status ...
```
● bitcoind.service - Bitcoin daemon
Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
Active: activating (auto-restart) (Result: exit-code) since Fri 2021-02-05 03:59:00 GMT; 21s ago
Process: 552 ExecStart=/usr/local/bin/bitcoind -daemon -pid=/run/bitcoind/bitcoind.pid -conf=/mnt/ext/bitcoin/bitcoin.conf -datadir=/mnt/ext/bitcoin (code=exited, status=0/SUCCESS)
Main PID: 605 (code=exited, status=1/FAILURE)
```

I recommend doing some diagnostics.

First, ensure that your external drive is attached, detected, and mounted.  You can list the block devices using this command
```sh
lsblk -o NAME,MOUNTPOINT,UUID,FSTYPE,SIZE,LABEL,MODEL
```

Here is an example of output on my setup
```
NAME   MOUNTPOINT UUID                                 FSTYPE   SIZE LABEL MODEL
sda                                                           953.9G       SABRE
└─sda1 /mnt/ext   1d9e9dee-87c3-4296-94e2-e833b948a19d ext4   953.9G       
mmcblk0
│                                                                58G       
├─mmcblk0p1
│      /boot      4BBD-D3E7                            vfat     256M boot  
└─mmcblk0p2
       /          45e99191-771b-4e12-a526-0779148892cb ext4    57.8G rootfs
```
In the above, `sda1` refers to my external drive, and its mounted on `/mnt/ext`.
The micro sd card is `mmcblk0` and has two partitions, one for `/boot`, and the other as root `/`

mateomar, 2021-02-23:
Thanks for the response. I have the Raspberry Pi 3, so I would need to
download the blockchain separately. Does that version of the guide still
exist?

lucasmoten, 2021-02-27:
The original stable guide release is here
https://github.com/Stadicus/RaspiBolt/releases/tag/1.0

Code for that release
https://github.com/Stadicus/RaspiBolt/tree/1.0

Stadicus, 2021-11-27:
Closing this issue due to inactivity. Feel free to reopen it if further assistance is required.

closed 2021-11-27 by Stadicus