raspibolt/raspibolt #702: If you are upgrading to Electrs 0.8.8, some additional steps...

lucasmoten, 2021-02-28:
The recent release of [Electrs 0.8.8](https://github.com/romanz/electrs/releases/tag/v0.8.8) didn't update the RELEASE-NOTES.md file prior to making the tag.

The determination of versions within the codebase is limited to that depicted in the RELEASE-NOTES.md.  As a result, without manual intervention, the scripts for upgrade will think version 0.8.8 is 0.8.7 and always claim it needs updating.

Following the [guidance for upgrades](https://stadicus.github.io/RaspiBolt/raspibolt_faq.html#how-to-upgrade-electrs), use the following which will checkout the commit that includes the fix to the RELEASE-NOTES.md for the version setting.
```
$ # clone the most recent tagged version
$ git clone --branch ${electrsgit} https://github.com/romanz/electrs.git

$ # Build it
$ cd electrs
$ get checkout d114a6b4a766dd958186fd7f51c884c76e6b5ac7
$ cargo build --locked --release
```

One of the configuration options has been deprecated.  To accommodate this sooner rather than later, you can update the configuration
```
sudo su - bitcoin
nano /mnt/ext/electrs/electrs.conf
```
Locate the section for `# RPC user / password` and change `cookie` to `auth`. Save and exit the file.  This release of Electrs deprecates the cookie option, and the initial quick handling is to switch to auth. In the future, 

Continue with the rest of the update guidance to stop the service, copy over the updated file, and then restart as normal
```
$ # Stop the service, copy the binary over, and start the service back up
$ sudo systemctl stop electrs
$ sudo cp ./target/release/electrs /usr/local/bin/
$ sudo systemctl start electrs
```

luneth, 2021-02-28:
Actually it is advised to use cookie_file cf: [https://github.com/romanz/electrs/blob/master/doc/cookie_deprecation.md](https://github.com/romanz/electrs/blob/master/doc/cookie_deprecation.md).
If you really need to use password based authentication replacing rpcpassword by rpcauth in your bitcoin.conf is the way to go as rpcpassword is soon be deprecated (already is?)
Refs: https://github.com/jlopp/bitcoin-core-config-generator/issues/25 https://bitcoin.org/en/release/v0.12.0#rpc-random-cookie-rpc-authentication

Stadicus, 2021-11-27:
I think the original topic of this issue has been taken care of with the recent update to RaspiBolt v3. I'm closing this issue, but feel free to reopen it if you think it's still unsolved.

closed 2021-11-27 by Stadicus