raspibolt/raspibolt #704: Preventing Electrum utxo leaks

2pac1, 2021-03-02:
Hey guys,
 
Keep in mind that: 
 electrum --oneserver --server raspibolt.local:50002:s
that's fine. That runs electrum pinned to your node.

But if you are using GUI that one should not just uncheck the Connect automatically. But, one should edit the config file of electrum and set:
Guide says: Uncheck “Select server automatically”.
I don't think that is enough.
One should edit config of electrum manually and set:

"auto_connect": false,
"oneserver": true,
 "server": "192.168.18.4:50002:s",
 
 If you just uncheck Connect automatically, and one server is not set to true, it will connect to 1 server but will communicate with different servers for different purposes as well. That is my understanding.
 So best hardening would be:
 1. edit the config file
 2. always run electrum with  electrum --oneserver --server raspibolt.local:50002:s

lucasmoten, 2021-04-16:
:+1: 

I concur with this. I set the config file as described and don't rely on CLI arguments.

Stadicus, 2021-11-27:
The updated RaspiBolt v3 guide explains how to directly change the config file before even starting Electrum. Easier across platforms and safer.

https://raspibolt.org/electrs.html#connect-electrum-wallet

I'm closing this issue, but feel free to reopen it if you think it's still unsolved.

closed 2021-11-27 by Stadicus