raspibolt/raspibolt #712: Automatic Dropbox SCB backup fails

Yasoj, 2021-03-27:
I followed the guide. However, the Dropbox menu is a bit different now, so I created a 'short-lived access token' which seems to be the recommended option (rather than one with no expiration).

When I do the test at the end, the file is not copied in my Dropbox folder.
The output of `sudo journalctl -fu backup-channels` is:

```
Uploading signature: 'channel.backup.md5'...
parse error: Invalid numeric literal at line 1, column 6
Unknown error when uploading...
/usr/local/bin/lnd-channel-backup.sh: line 151: inotifywait: command not found
Copying backup file...
Uploading backup file: 'channel.backup'...
parse error: Invalid numeric literal at line 1, column 6
Unknown error when uploading...
```

IdrisMiles, 2021-04-01:
I ran into this issue to, the fix is:
- in the dropbox app settings, go to the permissions tab
- enable `files.content.write` and `files.metadata.write` (I also enabled `files.content.read` for good measure but don't think that's required)
- save, then refresh the page and hit generate auth token again and use that one

[image]

Yasoj, 2021-04-01:
Perfect, it now works for me, thank you very much!

closed 2021-04-01 by Yasoj