raspibolt/raspibolt #721: Resurrecting an old Raspibolt. Doesn't appear to boot

paintufast, 2021-04-19:
Hello all. I used this guide to create a raspibolt in early 2019 I believe. I haven't touched it for a very long time. I decided to get back at it today and I noticed I can no longer SSH into my device. Upon further inspection I don't believe that it is booting up at all. When I power it on my power device shows a green LED but my raspberry pi shows a solid red light on its face only. I am reading that I may need a fresh install on my SD card perhaps but I am adraid to do anything for fear of losing the coins and details for my node. Can anyone give a brief synopsis or idea on how I should proceed? 

From what I can tell the device is not booting. It doesn't show up on my network now (I have an IP reserver for it, pings all fail and SSH always times out, ethernet lights are not lighting up.) . The device it self just has a solid red rlight on the front that seems to faintly flicker then goes dark about every thirty seconds for a moment before coming back on. I have tried connecting an HDMI cable and cannot get any video output from the unit. I have tried variations of the power connector on and off and also tried removing and reinstalling the SD card but no changes. 

All of that being said. What do I do? 

Should I just flash the SD card and start completely over following the guide again? I think I saved everything necessary to restore it but I am not really sure... I have: Master User Password, Bitcoin RPC Password, LND Wallet Password, LND Seed Password (optional). And a 24 word "test net seed". 

Please help, where do I go from here? I am good with doing the work, I am just confused where to start and no one I know in this space understands any of this stuff to help me so I am coming to you all hat in hands, seeking help! Thanks in advance.

lucasmoten, 2021-04-19:
The red light on your Raspberry Pi itself should be fine, as it indicates power is on.

I would recommend plugging in HDMI display if you can. Otherwise your diagnostics will be limited to seeing whats on the external drive and the SD card using another system.   Depending on the size of your SD card, its possible that it may have been filled to capacity from system logs (/var/log) 

If you were using your node to store coins in the bitcoin wallet, then you'll want to recover that.  If not, then the only harm in starting over is that you will end up resynching the whole blockchain.  If the only coins you had were from test net, then that's no real loss, and newer versions of BitcoinD support signet for testing as well

Stadicus, 2021-11-27:
Closing this issue due to inactivity

closed 2021-11-27 by Stadicus