raspibolt/raspibolt #722: Notes on Installing Bitcoin Core 0.21.0-based Taproot Client 0.1

lucasmoten, 2021-04-19:
Note: This is a Bitcoin Core based release of the most recent version of Bitcoin (v0.21.0) that includes Taproot Activation.  This is provided as guidance for those that want to upgrade to a version of Bitcoin that supports activation of Taproot following the BIP 8, Lot=True activation method.

As "admin" user, stop the lnd and bitcoind system units

```sh
$ sudo systemctl stop lnd
$ sudo systemctl stop bitcoind
```

Download the tarball and verify the checksum
```sh
# change to tmp directory which is cleared on reboot
$ cd /tmp

# download the binary and digital signatures
$ wget https://github.com/BitcoinActivation/bitcoin/releases/download/v0.21.0-taproot0.1/bitcoin-0.21.0-taproot0.1-arm-linux-gnueabihf.tar.gz
$ wget https://github.com/BitcoinActivation/bitcoin/releases/download/v0.21.0-taproot0.1/SHA256SUMS.asc

# check that the reference checksum matches the real checksum
# (ignore the "lines are improperly formatted" warning
$ sha256sum --check SHA256SUMS.asc --ignore-missing
```

The output will look like
```
bitcoin-0.21.0-taproot0.1-arm-linux-gnueabihf.tar.gz: OK
```

Continue with downloading, viewing, and importing public keys used for signing the release
```sh
# TODO: download, view, and import bitcoin mechanic's public key
# bitcoinmechanicca@protonmail.com
# fingerprint D085B36F8DEEDB4DDFEB73D7253F7E29DD06270D

# download, view, and import luke-jr's public key 
$ wget http://luke.dashjr.org/myself/luke-jr.asc
$ gpg --import-options show-only --import --with-fingerprint luke-jr.asc
> gpg: key BD02942421F4889F: 63 signatures not checked due to missing keys
> pub   rsa8192 2012-03-23 [SC] [expires: 2021-06-09]
>       E463 A93F 5F31 17EE DE6C  7316 BD02 9424 21F4 889F
> uid                      Luke Dashjr <luke@dashjr.org>
> uid                      Luke Dashjr <luke-jr+git@utopios.org>
> uid                      Luke Dashjr <luke-jr@dashjr.org>

$ gpg --import < luke-jr.asc
> gpg: key BD02942421F4889F: 63 signatures not checked due to missing keys
> gpg: key BD02942421F4889F: public key "Luke Dashjr <luke@dashjr.org>" imported
> gpg: key 665FC11DD53E9583: 58 signatures not checked due to missing keys
> gpg: key 665FC11DD53E9583: public key "Luke-Jr (Home) <luke@dashjr.org>" imported
> gpg: Total number processed: 2
> gpg:               imported: 2
> gpg: no ultimately trusted keys found
```

Next, verify the file itself
```sh
# verify the file
$ gpg --verify SHA256SUMS.asc
> gpg: Signature made Fri 16 Apr 2021 03:08:05 AM BST
> gpg:                using RSA key D085B36F8DEEDB4DDFEB73D7253F7E29DD06270D
> gpg: Can't check signature: No public key
> gpg: Signature made Fri 16 Apr 2021 03:52:53 AM BST
> gpg:                using RSA key E463A93F5F3117EEDE6C7316BD02942421F4889F
> gpg: Good signature from "Luke Dashjr <luke@dashjr.org>" [unknown]
> gpg:                 aka "Luke Dashjr <luke-jr+git@utopios.org>" [unknown]
> gpg:                 aka "Luke Dashjr <luke-jr@dashjr.org>" [unknown]
> gpg: WARNING: This key is not certified with a trusted signature!
> gpg:          There is no indication that the signature belongs to the owner.
> Primary key fingerprint: E463 A93F 5F31 17EE DE6C  7316 BD02 9424 21F4 889F
```

Extract the binaries
```sh
$ tar -xvf bitcoin-0.21.0-taproot0.1-arm-linux-gnueabihf.tar.gz
```

Install the binaries and restart the services
```sh
$ sudo install -m 0755 -o root -g root -t /usr/local/bin bitcoin-0.21.0-taproot0.1/bin/*
$ sudo systemctl start bitcoind
$ sudo systemctl start lnd
```

lucasmoten, 2021-04-19:
Added verification for both Bitcoin Mechanic and Luke Dash jr

Pantamis, 2021-04-23:
Perfect ! Thanks !

It should be easy also to change btc-rpc-explorer code to add bip8 softforks informations in node details section. I will take a look at this.

VajraOfIndra, 2021-10-05:
I suppose this issue can now be closed as Taproot has been successfully locked-in.

closed 2021-11-14 by Stadicus