raspibolt/raspibolt #723: Can't connect to electrs from Electrum

micto2021, 2021-04-20:
> You can try to narrow your problem.
> 
> First install `ncat` on your Raspberry Pi if you do not have it: `sudo apt install ncat`.
> Run `ncat`, it connects to server and waits for your input, enter provided json, `ncat` prints output. Press `Ctrl`-`C` to exit.
> 
> * Try to query Electrs directly from your Raspberry Pi
> 
> ```shell
> $ ncat localhost 50001
> {"jsonrpc":"2.0", "id":0, "method":"server.version", "params":["1.9.5","0.6"]}
> {"id":0,"jsonrpc":"2.0","result":["electrs 0.8.3","1.4"]}
> ^C
> ```
> 
> * Query Electrs through nginx
> 
> ```shell
> $ ncat --ssl localhost 50002
> {"jsonrpc":"2.0", "id":0, "method":"server.version", "params":["1.9.5","0.6"]}
> {"id":0,"jsonrpc":"2.0","result":["electrs 0.8.3","1.4"]}
> ^C
> ```
> 

I am having a similar problem.  I have been following the Raspibolt instructions.  
I am unable to connect to the electrs from electrum on my PC using the local address or via TOR.  ncat is clearly confirming I screwed something up.  I'm clueless on the 50001 failure, but I guess I screwed up a certificate somewhere for electrs.  I don't have enough knowledge at this point to understand where the breakdown is.  

ncat -v localhost 50001
Ncat: Version 7.70 ( https://nmap.org/ncat )
Ncat: Connection to ::1 failed: Connection refused.
Ncat: Trying next address...
Ncat: Connection refused.

ncat -v --ssl localhost 50002
Ncat: Version 7.70 ( https://nmap.org/ncat )
Ncat: Connection to ::1 failed: Connection refused.
Ncat: Trying next address...
Ncat: Subject: CN=localhost
Ncat: Issuer: CN=localhost
Ncat: SHA-1 fingerprint: 628E 20CE CA6E AF6D 3B63 BEA3 60BC 4C21 EE30 5F96
Ncat: Certificate verification failed (self signed certificate).
Ncat: SSL connection to 127.0.0.1:50002.
Ncat: SHA-1 fingerprint: 628E 20CE CA6E AF6D 3B63 BEA3 60BC 4C21 EE30 5F96

torsocks ncat -v --ssl <mynode..onion 50002
Ncat: Version 7.70 ( https://nmap.org/ncat )
Ncat: Subject: CN=localhost
Ncat: Issuer: CN=localhost
Ncat: SHA-1 fingerprint: 628E 20CE CA6E AF6D 3B63 BEA3 60BC 4C21 EE30 5F96
Ncat: Certificate verification failed (self signed certificate).
Ncat: SSL connection to 127.42.42.0:50002.
Ncat: SHA-1 fingerprint: 628E 20CE CA6E AF6D 3B63 BEA3 60BC 4C21 EE30 5F96

_Originally posted by @micto2021 in https://github.com/Stadicus/RaspiBolt/issues/565#issuecomment-823109348_

Would appreciate any assistance.

micto2021, 2021-04-22:
So I'm still smacking my head against this. I believe I followed the instructions to a T for RaspiBolt. So I'm not yet sure where i went wrong. I see other open issues with this certificate verification error. Expect it's something that someone more knowledgable can actually pinpoint. I am suspecting it's something in the NGINX portion... as up till then everything worked. I can see bitcoind and electrs running as expected. lnd is also running and other than I ahve not yet configured channels. Wallet is unlocked and

I installed electrum wallet on the pi now to see if I can connect locally and I can not. On TOR as well.

$ electrum -v --oneserver --server :50002:s --proxy socks5:127.0.0.1:9050
I | logging | Electrum version: 4.1.2 - https://electrum.org - https://github.com/spesmilo/electrum
I | logging | Python version: 3.7.3 (default, Jan 22 2021, 20:04:44)
[GCC 8.3.0]. On platform: Linux-5.10.17-v7l+-armv7l-with-debian-10.9
I | logging | Logging to file: None
I | logging | Log filters: verbosity '*', verbosity_shortcuts ''
I | daemon | failed to connect to JSON-RPC server Cannot connect to host 127.0.0.1:43567 ssl:default [Connect call failed ('127.0.0.1', 43567)]
I/p | plugin.Plugins | registering hardware bitbox02: ('hardware', 'bitbox02', 'BitBox02')
I/p | plugin.Plugins | registering hardware coldcard: ('hardware', 'coldcard', 'Coldcard Wallet')
I/p | plugin.Plugins | registering hardware digitalbitbox: ('hardware', 'digitalbitbox', 'Digital Bitbox wallet')
I/p | plugin.Plugins | registering hardware keepkey: ('hardware', 'keepkey', 'KeepKey wallet')
I/p | plugin.Plugins | registering hardware ledger: ('hardware', 'ledger', 'Ledger wallet')
I/p | plugin.Plugins | registering hardware safe_t: ('hardware', 'safe_t', 'Safe-T mini wallet')
I/p | plugin.Plugins | registering hardware trezor: ('hardware', 'trezor', 'Trezor wallet')
I/p | plugin.Plugins | registering wallet type ('2fa', 'trustedcoin')
D | util.profiler | Plugins.init 0.0414
I/n | network | blockchains [0]
I | exchange_rate.FxThread | using exchange CoinGecko
D | util.profiler | Daemon.init 0.0063
I | daemon.Daemon | launching GUI: qt
I/n | network | starting network
I/n | network | setting proxy {'mode': 'socks5', 'host': '127.0.0.1', 'port': '9050'}
I | daemon.Daemon | starting taskgroup.
I/n | network | connecting to :50002:s as new interface
I/n | network | starting taskgroup.
Unable to init server: Could not connect: Connection refused
Unable to init server: Could not connect: Connection refused

micto2021, 2021-04-22:
The same error with cetificate is present when running on the pi directly (not that I expected a different behaviour vs running remotely).

If I launch electrum on the pi itself with single server pointing to my onion address, it launches, is configured correctly but I am unable to connect to my electrs.  

the ncat message run locally returns the same error.  So i am convinced there is something I am missing between [UFW config  ](https://stadicus.github.io/RaspiBolt/raspibolt_21_security.html#enabling-the-uncomplicated-firewall )and the [NGINX config](https://stadicus.github.io/RaspiBolt/raspibolt_50_electrs.html#secure-communication).  

Everything else is working.  I am remoting in via SSH without password.  I am guessing there is something the instructions omit because if I knew better what I was doing it'd be idiot proof, but as I don't, I can't get the electrum connection running.

LightningTwo, 2021-04-22:
I do not know how to fix the problem with NGINX. But what I personally changed from this guide is the TOR hidden service for electrs. I point it to port `50001`, because the connection over TOR is already encrypted anyway. This also enables connecting from apps that do not allow self signed certificates, such as Phoenix Lightning wallet.

```
HiddenServiceDir /var/lib/tor/hidden_service_electrs/
HiddenServiceVersion 3
HiddenServicePort 50001 127.0.0.1:50001
```

I connect Electrum (on Windows) with `--oneserver --server <address>.onion:50001:t --proxy socks5:127.0.0.1:9150`

micto2021, 2021-04-22:
Well, I gave that a shot - uninstalled nginx, disabled UFW...  

No dice.  So now I am suspecting that one of the older pieces of the instructions is not playing nicely with one of the other pieces (given several have very outdated versions).  I have followed everything for RaspiBolt through to LND aside from setting up channels, then skipped to the ethereum portion and setup electrs...  

With UFW disabled and NGINX removed, still can't get to electrs even via tor.

$ ncat -v --ssl localhost 50001
Ncat: Version 7.70 ( https://nmap.org/ncat )
Ncat: Connection to ::1 failed: Connection refused.

$ ncat -v --ssl localhost 50002
Ncat: Version 7.70 ( https://nmap.org/ncat )
Ncat: Connection to ::1 failed: Connection refused.

$ torsocks ncat -v --ssl <mynode>.onion:50002
Ncat: Version 7.70 ( https://nmap.org/ncat )
1619126428 ERROR torsocks[6707]: Unable to resolve. Status reply: 1 (in socks5_recv_resolve_reply() at socks5.c:677)
Ncat: Could not resolve hostname "<mynode>.onion:50002": Non-recoverable failure in name resolution. QUITTING.

compane, 2021-04-30:
Please repeat the electrs installation once again --> https://stadicus.github.io/RaspiBolt/raspibolt_50_electrs.html
It is important to let electrs finish indexing!

Then check the nginx and electrs connections
`sudo lsof -i | grep "nginx\|electrs"`

electrs should listen on port 50001 and nginx on port 50002

`electrs     768        bitcoin    7u  IPv4    21973      0t0  TCP localhost:50001 (LISTEN)`
`nginx      5465       www-data    5u  IPv4 15895680      0t0  TCP *:50002 (LISTEN)`

If both services are listening and a connection is still not possible you should check the ufw rules for port 50002

`sudo ufw status  | grep 50002`

should display

`50002                      ALLOW IN    Anywhere                   # allow electrs SSL`

micto2021, 2021-04-30:
So electrs isn't listening on 50001, which leads me to believe based on your input that the indexing wasn't completed..  

To my untrained eye, it looks like it's re-indexing everything and getting no records.  Which I had taken to mean it was done and could proceed.  I'm using a modern SSD, so I had originally left it alone a few days and assumed the "should take 6 hours" was fune.  Likely that's my issue then.  Electrs isn't yet listening on 50001

output of

sudo journalctl -f -u electrs

Looks like this;

-- Logs begin at Tue 2021-04-27 00:17:28 EDT. --
Apr 30 05:24:09 RaspiBolt electrs[17174]: 2021-04-30T05:24:09.779-04:00 - TRACE - indexed "/mnt/ext/bitcoin/blocks/blk01927.dat": 0 rows

UFW checks out, so thanks for that.  I'll let it run until I see something like this;

2020-07-12T23:00:53.251+01:00 - WARN - reconnecting to bitcoind: disconnected from daemon while receiving
2020-07-12T23:03:07.478+01:00 - INFO - Electrum RPC server running on 127.0.0.1:50001 (protocol 1.4)

micto2021, 2021-04-30:
Hmm, now I am thinking something is preventing it from completing.  

-- Logs begin at Tue 2021-04-27 00:17:28 EDT. --
Apr 30 05:47:39 RaspiBolt electrs[17368]: 2021-04-30T05:47:39.234-04:00 - TRACE - indexed "/mnt/ext/bitcoin/blocks/blk00264.dat": 0 rows

micto2021, 2021-04-30:
Apr 30 08:54:11 RaspiBolt electrs[17908]: 2021-04-30T08:54:11.683-04:00 - TRACE - indexed "/mnt/ext/bitcoin/blocks/blk02556.dat": 0 rows
Apr 30 08:54:11 RaspiBolt electrs[17908]: 2021-04-30T08:54:11.689-04:00 - DEBUG - no more blocks to index
Apr 30 08:54:13 RaspiBolt electrs[17908]: 2021-04-30T08:54:13.028-04:00 - TRACE - indexed "/mnt/ext/bitcoin/blocks/blk02557.dat": 87251 rows
Apr 30 08:54:13 RaspiBolt electrs[17908]: 2021-04-30T08:54:13.029-04:00 - DEBUG - no more blocks to index
Apr 30 08:54:13 RaspiBolt electrs[17908]: thread 'main' panicked at 'indexer panicked: Any', src/bulk.rs:256:14
Apr 30 08:54:13 RaspiBolt electrs[17908]: 2021-04-30T08:54:13.658-04:00 - TRACE - closing DB at "/mnt/ext/electrs/db/mainnet"
Apr 30 08:54:16 RaspiBolt systemd[1]: electrs.service: Main process exited, code=exited, status=101/n/a
Apr 30 08:54:16 RaspiBolt systemd[1]: electrs.service: Failed with result 'exit-code'.

yeah, so electrs is struggling with the end of the index

micto2021, 2021-04-30:
So, I deleted electrsa and tried to reinstall electrs.  Failing to index is clearly the problem.  

Apr 30 08:54:10 RaspiBolt electrs[17908]: 2021-04-30T08:54:10.196-04:00 - TRACE - indexed "/mnt/ext/bitcoin/blocks/blk02555.dat": 0 rows
Apr 30 08:54:11 RaspiBolt electrs[17908]: 2021-04-30T08:54:11.683-04:00 - TRACE - indexed "/mnt/ext/bitcoin/blocks/blk02556.dat": 0 rows
Apr 30 08:54:11 RaspiBolt electrs[17908]: 2021-04-30T08:54:11.689-04:00 - DEBUG - no more blocks to index
Apr 30 08:54:13 RaspiBolt electrs[17908]: 2021-04-30T08:54:13.028-04:00 - TRACE - indexed "/mnt/ext/bitcoin/blocks/blk02557.dat": 87251 rows
Apr 30 08:54:13 RaspiBolt electrs[17908]: 2021-04-30T08:54:13.029-04:00 - DEBUG - no more blocks to index
Apr 30 08:54:13 RaspiBolt electrs[17908]: thread 'main' panicked at 'indexer panicked: Any', src/bulk.rs:256:14
Apr 30 08:54:13 RaspiBolt electrs[17908]: 2021-04-30T08:54:13.658-04:00 - TRACE - closing DB at "/mnt/ext/electrs/db/mainnet"
Apr 30 08:54:16 RaspiBolt systemd[1]: electrs.service: Main process exited, code=exited, status=101/n/a
Apr 30 08:54:16 RaspiBolt systemd[1]: electrs.service: Failed with result 'exit-code'.

compane, 2021-04-30:
You could check the core debug.log at startup for chain state errors. Did you set txindex=1 in bitcoin.conf? Did that finish without errors?

May be IBD got corrupted and you have to resync the blockchain.

micto2021, 2021-04-30:
> You could check the core debug.log at startup for chain state errors. Did you set txindex=1 in bitcoin.conf? Did that finish without errors?

Yes.  That was a painful few days waiting for it to complete.  current status;

$ bitcoin-cli getblockchaininfo
{
"chain": "main",
  "blocks": 681272,
  "headers": 681273,
  "bestblockhash": "000000000000000000069613f3efa15ffa984103bee8a6c1e6257212509ca7f7",
  "difficulty": 23581981443663.85,
  "mediantime": 1619805930,
  "verificationprogress": 0.9999957282782606,
  "initialblockdownload": false,
  "chainwork": "00000000000000000000000000000000000000001cba6d4e3c0a15527f97c63c",
  "size_on_disk": 388325045741,
...

> 
> May be IBD got corrupted and you have to resync the blockchain.

I've restarted the pi and made sure bitcoind is running.  Restarted electrs.  

Apr 30 15:37:01 RaspiBolt electrs[1007]: 2021-04-30T15:37:01.458-04:00 - INFO - indexing 2559 blk*.dat files
Apr 30 15:37:18 RaspiBolt electrs[1007]: 2021-04-30T15:37:18.954-04:00 - DEBUG - found 559743 indexed blocks
.....
Apr 30 16:32:58 RaspiBolt electrs[1007]: 2021-04-30T16:32:58.228-04:00 - TRACE - indexed "/mnt/ext/bitcoin/blocks/blk02557.dat": 0 rows
Apr 30 16:32:58 RaspiBolt electrs[1007]: 2021-04-30T16:32:58.236-04:00 - DEBUG - no more blocks to index
Apr 30 16:32:58 RaspiBolt electrs[1007]: 2021-04-30T16:32:58.754-04:00 - TRACE - indexed "/mnt/ext/bitcoin/blocks/blk02558.dat": 232392 rows
Apr 30 16:32:58 RaspiBolt electrs[1007]: 2021-04-30T16:32:58.754-04:00 - DEBUG - no more blocks to index
Apr 30 16:32:59 RaspiBolt electrs[1007]: thread 'main' panicked at 'indexer panicked: Any', src/bulk.rs:256:14
Apr 30 16:32:59 RaspiBolt electrs[1007]: 2021-04-30T16:32:59.209-04:00 - TRACE - closing DB at "/mnt/ext/electrs/db/mainnet"
Apr 30 16:33:00 RaspiBolt systemd[1]: electrs.service: Main process exited, code=exited, status=101/n/a
Apr 30 16:33:00 RaspiBolt systemd[1]: electrs.service: Failed with result 'exit-code'.
Apr 30 16:33:30 RaspiBolt systemd[1]: electrs.service: Service RestartSec=30s expired, scheduling restart.

It's getting hung up on the last blocks.  So Electrs just keeps restarting.

Stadicus, 2021-11-27:
Were you able to fix this issue?

closed 2022-02-02 by github-actions[bot]