raspibolt/raspibolt #725: OS - Recommend 64-bit OS, 32-bit is deprecated by LND and potentially dangerous

wankerstar, 2021-04-26:
The guide currently recommends 32-bit Raspbian. This can cause serious problems to lnd as its BBOLT database approaches 1GB, and as a result the lnd devs recommend 64-bit systems only at this point.

Please see the following issue for details and lnd dev recommendations.
https://github.com/lightningnetwork/lnd/issues/5126

lucasmoten, 2021-04-27:
Thanks. This is a known issue also recorded in #669 

I believe this is still the most recent version available for 64 bit
https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2020-08-24/
However, the Raspberry Pi Foundation is still not linking to it from https://www.raspberrypi.org/software/operating-systems/

If you are running a smaller node, you can likely continue use of 32 bit and run the `chantools compactdb`. Newer versions of LND also support settings to compact on startup.

wankerstar, 2021-04-28:
Thanks for the response. My lnd db is near the 1gb limit due to invoices created by my early circular rebalancing attempts; they're immune to --gc-cancelled-invoices and AFAIK I can't delete them over the grpc. To deal with this I've had to re-compact regularly, often having to do so on a different (64-bit) machine. So I'm forced to switch architectures. The hope here is spare other people this experience.

What about recommending something like DietPi instead? Still Debian, but 64-bit.

Swissnode, 2021-05-28:
This is a real issue for me running a huge node on a raspBian system (from the Stadicus guide). It would be really nice to have a walk-through HOW to upgrade and not just a link to the 64bit OS library. Is anybody planning to do this?

xanoni, 2021-08-10:
I don't think there's an easy upgrade path. I believe it would be a manual reinstall and then you can copy over individual config files and your home folders etc.

Sorry.

kristapsk, 2021-09-28:
I'm currently migrating my RaspiBolt setup to 64-bit Odroid-HC4 (which uses Ubuntu Server), here's some installation notes, may be useful for someone else too - https://gist.github.com/kristapsk/79049d6da791ac5f44c4b655b5d50e4e.

xanoni, 2021-09-28:
> 64-bit Odroid-HC4

Curious, why did you choose this hardware (vs. a Raspberry product)?

kristapsk, 2021-09-28:
Because it has two built-in SATA slots.

Stadicus, 2021-11-27:
I think this has been taken care of with the recent update to RaspiBolt v3. I'm closing this issue, but feel free to reopen it if you think it's still unsolved.

closed 2021-11-27 by Stadicus