raspibolt/raspibolt #728: apt-key is being deprecated, use replacement for tor pgp key

mutatrum, 2021-05-11:
https://stadicus.github.io/RaspiBolt/raspibolt_22_privacy.html#installing-tor

Key management with apt-key is getting phased out. Replacement is by downloading key into a keyring file, taken from https://www.linuxuprising.com/2021/01/apt-key-is-deprecated-how-to-add.html:

```
wget -O- https://deb.torproject.org/torproject.org/A3C4F0F979CAA22CDBA8F512EE8CBC9E886DDD89.asc | gpg --dearmor | sudo tee /usr/share/keyrings/torproject-keyring.gpg
```
and adding a signed-by field to a separate sources file (`/etc/apt/sources.list.d/torproject.list`):
```
deb [signed-by=/usr/share/keyrings/torproject-keyring.gpg] https://deb.torproject.org/torproject.org buster main
deb-src [signed-by=/usr/share/keyrings/torproject-keyring.gpg] https://deb.torproject.org/torproject.org buster main
```

I tested this on Ubuntu 20.10. I don't know this is relevant for Raspberry OS.

Stadicus, 2021-11-14:
The next update of the guide will install `tor` directly from the official Debian repo, as it's now current enough.

Stadicus, 2021-11-27:
I think this has been taken care of with the recent update to RaspiBolt v3. I'm closing this issue, but feel free to reopen it if you think it's still unsolved.

closed 2021-11-27 by Stadicus