raspibolt/raspibolt #729: Switch to rpcauth for bitcoind

mutatrum, 2021-05-11:
rpcuser/rpcpassword are being deprecated and replace by rpcauth:

```
wget https://raw.githubusercontent.com/bitcoin/bitcoin/master/share/rpcauth/rpcauth.py
python3 rpcauth.py bitcoind -
```
After a password prompt, this will return the rpcauth config line for bitcoin.conf.

vicariousdrama, 2021-08-18:
Is there a way to make that work so we can do bitcoin-cli from the command line without having to pass  the RPC user and password clear text?

mutatrum, 2021-08-19:
That would be outside the scope of this issue. rpcuser / rpcpassword configuration will be deprecated. For RaspiBolt, the RPC is only opened for local processes I assume, so IMO using rpcauth is fine.

See https://github.com/bitcoin/bitcoin/blob/0.18/src/httprpc.cpp#L227 for the deprecation warning. I couldn't find any specific timeframe for this change, but at least it removes the warning.

Stadicus, 2021-11-14:
This change has a lot of ripple effects on all parts of the guide. This is why I never updated it just for Bitcoin Core, as it also breaks easy `bitcoin-cli` usage from other accounts, and it's a pretty pointless if then every other service needs the rpcpassword in plaintext somewhere.

In combination with cookie authentication it becomes more useful. I plan to change this for the next bigger guide overhaul (version 3). See WIP here: 

https://stadicus.github.io/raspibolt-remote/raspibolt_30_bitcoin.html#generate-access-credentials

Stadicus, 2021-11-27:
I think this has been taken care of with the recent update to RaspiBolt v3. I'm closing this issue, but feel free to reopen it if you think it's still unsolved.

closed 2021-11-27 by Stadicus