raspibolt/raspibolt #732: lncli unlock but in next command says its locked?

Superhans79, 2021-06-02:
admin@raspibolt:~ $ lncli unlock Input wallet password:   lnd successfully unlocked! admin@raspibolt:~ $ lncli walletbalance [lncli] Wallet is encrypted. Please unlock using 'lncli unlock', or set password using 'lncli create' if this is the first time starting lnd.

I was trying to install electrum and in the indexing stage it failed as my HDD was full (500GB SSD)

COULD THIS BE AN UNDERLYING FACTOR - IF IT IS - CAN I AD A SECOND HDD?

Massive thanks - sorry if daft questions!

I dont mind starting again with a bigger hard drive but im not sure how I can get back into my wallet which has my satoshis in :-)

lucasmoten, 2021-06-22:
First, If you have actually made use of lightning, funding wallet and opening channels, I recommend making backups of your 
- channel.db (in ~/.lnd/data/graph/mainnet) 
- channel.backup (~/.lnd/data/chain/bitcoin/mainnet/channel.backup). 
Or you can simply backup all of ~/.lnd as its around 200MB.

Next, yes, if you want to install Bitcoin, LND, and Electrum Server (Electrs in this case) you are going to need a hard drive larger than 500GB.  A 1TB SSD drive is recommended.

I would then stop the electrs service, and make it so it doesnt run at startup.
```
sudo systemctl stop electr
sudo systemctl disable electrs
```
And to free up space, you can remove the electrs db folder, but leave the config
```
rm -rf /mnt/ext/electrs/db
```

That should hopefully free up enough room to allow LND to startup as it wants and leave you with basically bitcoind and lnd.  

To support electrs and block explorer, I'd then recommend acquiring a 1TB drive, and follow guidance for migrating to the new drive by following this
https://github.com/Stadicus/RaspiBolt/blob/master/raspibolt_77_upgrade-external-drive.md

Note that this guide will also walk you through stopping all services, and references dropbox. If you hadn't setup backups to use dropbox, i'd skip that for now, and simply make a copy of the channel.backup file just in case.  If you don't actually have funds in wallet for Lightning or channels open, then you can skip that as well

Stadicus, 2021-11-27:
I think this has been taken care of with the recent update to RaspiBolt v3. I'm closing this issue, but feel free to reopen it if you think it's still unsolved.

closed 2021-11-27 by Stadicus