raspibolt/raspibolt #734: Unable to remove *.macaroon when upgrading LND

bvrulez, 2021-06-17:
Not sure what is going on here:

```
admin@RPi4:/tmp $ sudo ls /home/bitcoin/.lnd/data/chain/bitcoin/mainnet
admin.macaroon	chainnotifier.macaroon	channel.backup	invoice.macaroon  invoices.macaroon  macaroons.db  readonly.macaroon  router.macaroon  signer.macaroon	wallet.db  walletkit.macaroon
admin@RPi4:/tmp $ sudo rm /home/bitcoin/.lnd/data/chain/bitcoin/mainnet/*.macaroon
rm: cannot remove '/home/bitcoin/.lnd/data/chain/bitcoin/mainnet/*.macaroon': No such file or directory

```

bvrulez, 2021-06-17:
I was able to remove the `*.macaroon` files by their individual paths. After installing and restarting `lnd` I now get this error. When/How should the files be created again? 

```
admin@RPi4:/tmp $ lncli getinfo
[lncli] unable to read macaroon path (check the network setting!): open /home/admin/.lnd/data/chain/bitcoin/mainnet/admin.macaroon: no such file or directory

```

bvrulez, 2021-06-17:
I realized that my `lnd.conf` was missing two new entries. So I added them: 

```
protocol.achors=true
protocol.wumbo-channels=true
```

I thought maybe I deleted my wallet by deleting the `*.macaroon` files. Since I did not use `lnd` this would not have been a problem. But when I try to create a new wallet I get the error that a wallet already exists:

```
bitcoin@RPi4:~/.bitcoin $ lncli create
Input wallet password: 
Confirm password: 

Do you have an existing cipher seed mnemonic you want to use? (Enter y/n): n

Your cipher seed can optionally be encrypted.
Input your passphrase if you wish to encrypt it (or press enter to proceed without a cipher seed passphrase): 

Generating fresh cipher seed...

[lncli] unable to generate seed: rpc error: code = Unknown desc = wallet already exists

```

The original installation guide says "Check if the TLS certificates and admin.macaroon have been created." So I think, after deleting them they have to be created again. In which part of the process is this happening?

bvrulez, 2021-06-17:
```
admin@RPi4:/mnt/bitcoin/lnd $ lnd --version
lnd version 0.11.1-beta commit=v0.11.1-beta

```

But 
```

admin@RPi4:/mnt/bitcoin/lnd $ lnd
/home/admin/.lnd/lnd.conf:10: unknown option: protocol.anchors

```

So, I guess the `protocol.anchors` option in the config is outdated?

bvrulez, 2021-06-17:
Commenting this (`#protocoll.anchors=true`) works. Also, I deleted `~/.lnd/data` as per [here](https://github.com/lightningnetwork/lnd/issues/1216) to beeing prompted for a new wallet creating.

```
bitcoin@RPi4:~/.bitcoin $ lnd
Attempting automatic RPC configuration to bitcoind
Automatically obtained bitcoind's RPC credentials
2021-06-17 14:54:47.723 [INF] LTND: Version: 0.11.1-beta commit=v0.11.1-beta, build=production, logging=default
2021-06-17 14:54:47.723 [INF] LTND: Active chain: Bitcoin (network=mainnet)
2021-06-17 14:54:47.769 [INF] LTND: Opening the main database, this might take a few minutes...
2021-06-17 14:54:47.769 [INF] LTND: Opening bbolt database, sync_freelist=false
2021-06-17 14:54:47.826 [INF] CHDB: Checking for schema update: latest_version=17, db_version=17
2021-06-17 14:54:47.826 [INF] LTND: Database now open (time_to_open=57.298256ms)!
2021-06-17 14:54:47.907 [INF] RPCS: password RPC server listening on 127.0.0.1:10009
2021-06-17 14:54:47.907 [INF] RPCS: password gRPC proxy started at 127.0.0.1:8080
2021-06-17 14:54:47.907 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.

```

bvrulez, 2021-06-17:
Using `lncli create` made a new wallet and created new macaroons. 

So, I guess the guide should be updated to reflect the fact that deleting `*.macaroons` means deleting your wallet? Not a problem for me since I do this just to stay up-to-date.

Pantamis, 2021-06-19:
LND normally recreate macaroons file if there is none when you start it  

For anchors you must write
`protocol.anchors=1`

in configuration files (anchors are default starting from LND 0.12)

Yasoj, 2021-06-21:
> admin@RPi4:/mnt/bitcoin/lnd $ lnd --version
> lnd version 0.11.1-beta commit=v0.11.1-beta
> [...]
> So, I guess the `protocol.anchors` option in the config is outdated?

On the contrary, the `protocol.anchors` setting was introduced in version 0.12-beta, so a more recent version that the one you installed. If/when you'll update your LND to the most recent version (0.13.0 as of now), you should then be able to use `protocol.anchors=1`, although I believe it is now the default).

xanoni, 2021-06-25:
> On the contrary, the `protocol.anchors` setting was introduced in version 0.12-beta, so a more recent version that the one you installed. If/when you'll update your LND to the most recent version [...]

`Version: 0.13.0-beta commit=v0.13.0-beta, build=production, logging=default, debuglevel=info`

`/home/lnd/.lnd/lnd.conf:10: unknown option: protocol.anchors`

tried `=1` and `=true`

sup?

Pantamis, 2021-06-25:
> > On the contrary, the `protocol.anchors` setting was introduced in version 0.12-beta, so a more recent version that the one you installed. If/when you'll update your LND to the most recent version (0.13.0 as of now), you should then be able to use `protocol.anchors=1`, although I believe it is now the default).
> 
> `Version: 0.13.0-beta commit=v0.13.0-beta, build=production, logging=default, debuglevel=info`
> 
> `/home/lnd/.lnd/lnd.conf:10: unknown option: protocol.anchors`
> 
> tried `=1` and `=true`
> 
> sup?
> 
> (arch = armv7)

Anchor are activated by default in LND release 0.13.0. So the `protocol.anchors` settings is now `protocol.no-anchors` and it has the opposite behavior.

So if you installed LND 0.13.0 there is nothing to do to use anchor outputs !

xanoni, 2021-06-27:
Thank you. Yes, I realized that. However, it's confusing that the example sample-lnd.conf still includes both settings.

Stadicus, 2021-11-27:
I think this has been taken care of with the recent update to RaspiBolt v3. I'm closing this issue, but feel free to reopen it if you think it's still unsolved.

closed 2021-11-27 by Stadicus