raspibolt/raspibolt #736: Watchtower Client: "active_session_candidate": false

d-hoffi, 2021-06-23:
I want to use a watchtower out of this list for my lnd node: https://github.com/openoms/lightning-node-management/issues/4
The setup of my `wtclient` was successful using this guide: https://openoms.gitbook.io/lightning-node-management/advanced-tools/watchtower#tor-watchtower-client-setup

For some reason I can't get an active session with any watchtower. I tried four different ones, all with the same result.
```
$ lncli wtclient towers
 {
    "towers": [
        {
            "pubkey": "02855af21acd3c1c6c7f4fccb4aef5acb88b448a38d29f72334c3667e4ed1bae4b",
            "addresses": [
                "4a6brdux7pelxv47u22lklfcpk4moa54qi5fa4vrkcyo45lsls4pa4qd.onion:9911"
            ],
            "active_session_candidate": false,
            "num_sessions": 0,
            "sessions": [
            ]
        },
}
``` 
my firewall allows connections to port 9911:
```
$ sudo ufw status
Status: active

To                         Action      From
--                         ------      ----
9911                       ALLOW       Anywhere                   # allow watchtower
9911 (v6)                  ALLOW       Anywhere (v6)              # allow watchtower
```
my lnd logs:
```
2021-06-23 10:10:14.984 [INF] WTCL: (legacy) Starting watchtower client
2021-06-23 10:10:14.984 [INF] WTCL: (legacy) Watchtower client started successfully
2021-06-23 10:10:14.984 [INF] WTCL: (anchor) Starting watchtower client
2021-06-23 10:10:14.984 [INF] WTCL: (legacy) Requesting new session.
2021-06-23 10:10:14.984 [INF] WTCL: (anchor) Watchtower client started successfully
2021-06-23 10:11:13.931 [INF] WTCL: (legacy) Client stats: tasks(received=0 accepted=0 ineligible=0) sessions(acquired=0 exhausted=0)
2021-06-23 10:11:13.931 [INF] WTCL: (anchor) Client stats: tasks(received=0 accepted=0 ineligible=0) sessions(acquired=0 exhausted=0)
```

I already tried to restart `lnd` and `tor` & reboot the whole node.

ChristianBi11, 2021-08-02:
I have the same issue. I have currently configured 8 watchtowers and no one works...

d-hoffi, 2021-08-19:
The issue is fixed now by myself. probably the node / watchtower was just offline

closed 2021-08-19 by d-hoffi

DerHuette, 2021-09-11:
Hmm, I have exactly the same problem. And it does not fix itself. Any hints what I could do to get the watchtower client working?