raspibolt/raspibolt #740: Channel backups should be encrypted (especially if sent to Dropbox)

xanoni, 2021-07-10:
I saw that the guide "Static Channel Backups" suggests sending unencrypted backups to Dropbox. I don't think that's the greatest idea.

If one decides to use Dropbox, then the lnd data directory should be stored in something like a Cryptomator container (https://github.com/cryptomator/cli).

The only required change is that you would then point the Dropbox script at the Cryptomator data folder (the one with the encrypted files). This is what Cryptomator was made for. It's simple yet effective.

Alternatively, you could update the script to bundle the files in a .tar.gz. archive and then pipe it through GPG, using your public key for encryption.

Stadicus, 2021-11-14:
SCBs are encrypted by default:

> Because the backup file is encrypted with a key from the seed the node was created with, it can safely be stored on a cloud storage or any other storage medium.

https://github.com/lightningnetwork/lnd/blob/master/docs/safety.md#keeping-static-channel-backups-scb-safe

xanoni, 2021-11-15:
Alright, wasn't aware when I opened this.

closed 2021-11-15 by xanoni