raspibolt/raspibolt #742: /home/bitcoin/.lnd/lnd.conf:10: unknown option: protocol.anchors

n-e-t-d-i-v-e-r, 2021-07-22:
Hi there

I’m fresh in this fullnode universe. I just configured a raspibolt bitcoin and lightning fullnode. Most of it was going easy. But as I tried to configure lightning I had this alert:` /home/bitcoin/.lnd/lnd.conf:10: unknown option: protocol.anchors` until I commented out #protocol.anchors=true #protocol.wumbo-channels=true in the lnd.conf file.

Now everything works fine but I’m unsure about my modifications in the conf-file, could this be a security issue for my installation?

VajraOfIndra, 2021-07-22:
I believe this is because anchor channels were introduced in LND v0.12 so `protocol.anchors=true` is not recognized in v0.11 which is the version presently used in the guide. However, `protocol.wumbo-channels=true` should work fine with LND v0.11 ([link](https://lightning.engineering/posts/2020-08-20-lnd-v0.11/)).

Actually, anchor channels have become default in LND v0.13 so `protocol.anchors=true` is also not recognized in v0.13 and above! So afaik, `protocol.anchors=true` can only be used with LND v0.12.

Wumbo allows to open channels bigger than ~0.16 BTC and anchor channels allow cheaper closing fees in most cases. They are two useful options to have, but there is no security risks associated with not using them either!

n-e-t-d-i-v-e-r, 2021-07-23:
thanks for your response. so in case I understand you right I can leave #protocol.anchors=true #protocol.wumbo-channels=true out commented without any security risk and can make an update after v0.12 and v0.13 are released. true?

VajraOfIndra, 2021-07-24:
LND v0.12 and v0.13 are already released. The [latest version](https://github.com/lightningnetwork/lnd/releases) is LND v0.13.1-beta.

Afaik you do not need to leave `protocol.wumbo-channels=true` commented out as it should work with v0.11. But it's safe to do so anyway, sure.

Regarding anchor channels, best thing is to update your LND to the latest version, as you will have them by default (you could even delete `protocol.anchors=true` as you'll never need it anymore once on LND v0.13 and above.

n-e-t-d-i-v-e-r, 2021-07-27:
Thx for supporting me.

closed 2021-07-27 by n-e-t-d-i-v-e-r