raspibolt/raspibolt #743: LND Auto-Unlock should be much easier these days (wallet-unlock-password-file)

xanoni, 2021-07-26:
I saw this guide: https://stadicus.github.io/RaspiBolt/raspibolt_6A_auto-unlock.html

This must have been written before the option `wallet-unlock-password-file` was added to `lnd.conf`.

It just needs to be pointed at the password file and that's it. It seems the proposed script is overkill at this point.

Or did I miss some advantage of the proposed method?

VajraOfIndra, 2021-07-26:
The first line of the guide seems to indicate that it was written before [LND v0.5](https://github.com/lightningnetwork/lnd/releases/tag/v0.5-beta), so before September 2018; while the `wallet-unlock-password-file` option was released only recently with [v.0.13](https://github.com/lightningnetwork/lnd/releases/tag/v0.13.0-beta) (last month).

@guggero has written a useful guide for a secure setup involving a password manager (`pass`) and a named pipe: [https://github.com/lightningnetwork/lnd/blob/master/docs/wallet.md#more-secure-example-with-password-manager-and-using-a-named-pipe](https://github.com/lightningnetwork/lnd/blob/master/docs/wallet.md#more-secure-example-with-password-manager-and-using-a-named-pipe)

Would that kind of setup be the most secure option for now for the Raspibolt?

xanoni, 2021-07-27:
Interesting solution! However, I'm not sure I understand whether this is safer than just storing the password in a text file in shared memory (i.e., a tmpfs ramdisk, e.g., via `echo -n EXAMPLEPASSWORD > /dev/shm/password`).

Yes, the FIFO pipe can only be read once. Thus, after LND has read the password, the pipe will be empty and can't be read by other processes.

However, I don't understand how the `pass` tool is protected. Couldn't a random process running under the same user just execute the pass tool again (instead of reading a file)?

BTW, this script would still have to be integrated with `systemd` to enable automatic restarts upon failure/crash/reboot.

Kixunil, 2021-08-04:
Correct, FIFO is useless in non-interactive unlocking. It could be used in interactive setups as a more reliable and secure alternative to RPC.

xanoni, 2021-08-05:
> useless

I wouldn't say entirely useless. The one benefit is that you can lock a plaintext PW behind a privileged account and give one-time access to an unprivileged account. Or is there a better way to achieve this?

Kixunil, 2021-08-05:
> give one-time access to an unprivileged account

I don't see how it helps with any realistic attack. If an unprivileged account running LND is compromised the malware will just extract it from memory or do one of many other possible tricks.

Stadicus, 2021-11-27:
I think this has been taken care of with the recent update to RaspiBolt v3. I'm closing this issue, but feel free to reopen it if you think it's still unsolved.

More secure options for the LND auto-unlock would be great to have as a bonus guide, linked directly from the main guide as an option. If you have a good solution, feel free to contribute. :)

closed 2021-11-27 by Stadicus