raspibolt/raspibolt #747: Electrs guide is quite outdated + other suggestions

Kixunil, 2021-07-30:
There are several things out-of-date in the `electrs` guide and I have some other suggestion

- [ ] While the guide suggests dynamically checking version using `curl | jq` (more on it later), the screen recording does **not* use that technique, which [confused at least one user](https://github.com/romanz/electrs/issues/448#issuecomment-889635368)
- [ ] The guide uses `cookie` option which is deprecated and **will be removed in the upcoming release**, please consider using `cookie_file`, which is safer.
- [ ] Consider recommending [dynamic linking](https://github.com/romanz/electrs/blob/master/doc/usage.md#dynamic-linking), which is faster - I don't have exact numbers but feels like the compile time is about half with dynamic linking. This may be especially helpful for RPi users since RPi is less powerful

Additional suggestions:

- [ ] Recommend getting @romanz GPG key (`gpg --recv-keys 15C8C3574AE4F1E25F3F35C587CAE5FA46917CBB` - don't trust, verify the fingerprint from multiple sources) and running `git verify-tag "electrsgit"` before compiling
- [ ] Put double quotes around `$(curl ...)` and `electrsgit` to prevent github from sending garbage value (with spaces) that could be dangerous
- [ ] Add `echo "$electrsgit"` and recommend checking from an independent source that this is the latest version of `electrs` to avoid downgrade attacks (@romanz how about adding latest electrs version to your Twitter bio?)
- [ ] Consider using `db_dir` option instead of symlinking `electrs` directory
- [ ] Suggest upgrading by only backing up electrs binary, running `git clean -x -f d` followed by `git fetch` and `git checkout` - this is faster because only changes need to be downloaded

Things for the future (unsupported by `electrs` but wanted):

- [ ] when `electrs` can skip default configs and `db_dir` is used add `ProtectHome`
- [ ] when electrs has `--version` flag run it instead of parsing `RELEASE-NOTES.md`

xanoni, 2021-08-07:
Please also note that it's not required to build/install your own custom Rust version ... the one from Raspbian is fine ...

See here: https://github.com/romanz/electrs/issues/207#issuecomment-881924630

Kixunil, 2021-08-07:
Yeah, LOL, I recommend everyone to install from packages which is much safer, yet I missed it here. Good point!

xanoni, 2021-08-07:
```
$ rustc --version
rustc 1.41.1
```
Default 32-bit Raspbian

kristapsk, 2021-10-07:
> The guide uses `cookie` option which is deprecated and **will be removed in the upcoming release**, please consider using `cookie_file`, which is safer.

It's already removed now, you will have error when starting `electrs --conf /mnt/ext/electrs/electrs.conf` with current guide:
```
Config { network: Bitcoin, db_path: "/mnt/ext/electrs/db/bitcoin", daemon_dir: "/mnt/ext/bitcoin", daemon_auth: CookieFile("/mnt/ext/bitcoin/.cookie"), daemon_rpc_addr: 127.0.0.1:8332, daemon_p2p_addr: 127.0.0.1:8333, electrum_rpc_addr: 127.0.0.1:50001, monitoring_addr: 127.0.0.1:4224, wait_duration: 10s, jsonrpc_timeout: 15s, index_batch_size: 10, index_lookup_limit: Some(200), auto_reindex: true, ignore_mempool: false, sync_once: false, server_banner: "Welcome to electrs 0.9.0-rc2 (Electrum Rust Server)!", args: [] }
[2021-10-07T19:39:14.296Z INFO  electrs::metrics::metrics_impl] serving Prometheus metrics on 127.0.0.1:4224
[2021-10-07T19:39:14.300Z DEBUG tiny_http] Server listening on 127.0.0.1:4224
[2021-10-07T19:39:14.302Z DEBUG tiny_http] Running accept thread
[2021-10-07T19:39:14.392Z INFO  electrs::db] "/mnt/ext/electrs/db/bitcoin": 0 SST files, 0 GB, 0 Grows
[2021-10-07T19:39:14.392Z DEBUG electrs::db] DB None
[2021-10-07T19:39:14.398Z INFO  electrs::db] closing DB at /mnt/ext/electrs/db/bitcoin
Error: failed to open bitcoind cookie file: /mnt/ext/bitcoin/.cookie

Caused by:
    No such file or directory (os error 2)
```

AsyncLuck, 2021-11-07:
Hi @Kixunil, very good initiative to go deeper on the electrs part. On my side, i was forced to whitelist my node for download in the bitcoin.conf  :
whitelist=download@127.0.0.1

Or the initial electrs sync will fail after some times. (pi4 ubuntu 64bits).
After that, it was smooth...

Kixunil, 2021-11-08:
Thanks for reminding! Added to the checklist.

Stadicus, 2021-11-27:
@Kixunil, I think most of your suggested improvements made it into the updated RaspiBolt v3. Can you check if anything critical is still missing?

https://raspibolt.org/electrs.html

Kixunil, 2021-11-30:
Was away for a while, will check. Is there a way to set light theme? I don't see the option and dark themes are quite hard for me to read.

Stadicus, 2021-11-30:
> Is there a way to set light theme?

I'm afraid not. Didn't think of that when I did the dark layout, sorry. Maybe the MD files in the repo might be better to follow?

stefan-wuest, 2021-12-29:
Hi @Kixunil and @Stadicus 
I have a further suggestion to update the guide. Please see https://github.com/raspibolt/raspibolt/issues/858 for details.
The suggestion is to update this part in the guide: https://raspibolt.org/electrs.html#local-connection
and replace raspibolt.local by the actual IP address ($ ip addr show).

Kixunil, 2021-12-29:
Agree, `.local` names are very unreliable in my experience.

Reminds me I wanted to review the changes but stuff got in the way. :(

stefan-wuest, 2021-12-29:
Thanks for contributing in the first place - I'm thrilled!
Enjoy the holidays! 🎄

visskiss, 2021-12-29:
I prefer .local names. Good networks shouldn’t have an issue with this.

stefan-wuest, 2022-01-01:
Hi @Kixunil and @Stadicus
I have a further suggestion to update the guide. Please see https://github.com/raspibolt/raspibolt/issues/859 for details.
The suggestion is to update this part in the guide: https://raspibolt.org/rtl.html#ride-the-lightning-1
and add the following as a first step:

sudo npm install npm@latest -g
(in order to install latest Version of NPM as a prerequisite for further steps)

Kixunil, 2022-03-02:
Oh, forgot about this, just looked into it. My thoughts:

* I suggest prepending `cargo build` command with `CARGO_NET_GIT_FETCH_WITH_CLI=true` which will workaround the bug on Debian systems with broken `cargo`.
* Why not `adduser --system`? It should be more bare-bones but maybe I'm missing something.
* `ProtectSystem=full` should work - it does on my machine
* `ProtectHome=true` will work even without messing with HOME variable in the upcoming release if you use `--skip-default-conf-files` argument. But setting `HOME=/data/electrs` is not too bad either
* Why runtime directory mode is `710` instead of `750`?
* I'm afraid without `Group=electrs` the group will be `root` DYOR
* "Expert: install Tor as a background service" I don't think this is really "expert" and it's less weird/annoying to use. At least on Linux where it's a single command (much easier and *more secure* than manual download), maybe other OSes are terrible because of lack of packaging...
* "the blue LED" I'm not sure I would call it LED, those who know what LED is are confused and those who don't will not understand it. Just call it "circle" or "dot" :) Also last time I checked it was green, not blue.
* `git verify-tag` is missing in upgrade instructions, I recommend to add it before checkout
* Nit: "# Start electrs" in upgrade instructions should be "# Restart electrs"

VajraOfIndra, 2022-03-22:
@Stadicus @RooSoft 

I think we should aim at updating the Electrs guide fairly soon. On top of the suggestions above by @Kixunil , there is also a need to address #951  and #970.

What about we create a shared "update-electrs" branch in the `raspibolt/raspibolt` repository to work on all these updates?

purpleKarrot, 2022-05-11:
Instead of installing Rust on the Raspberry Pi, I cross-compile Electrs in a Docker container.

This is my `Dockerfile`:
```docker
FROM debian:bullseye-slim
RUN dpkg --add-architecture arm64 \
    && apt-get -qq update \
    && apt-get -qq install -y --no-install-recommends \
        ca-certificates \
        cargo \
        g++-aarch64-linux-gnu \
        libc6-dev:arm64 \
        libclang-11-dev \
        librocksdb-dev:arm64 \
        libstd-rust-dev:arm64 \
        libstdc++6:arm64 \
    && rm -rf /var/lib/apt/lists/*
```

Those are the commands to build the container, cross-compile Electrs, and copy to the Pi:
```bash
docker build -t raspibuilder .

docker run --rm \
  --env BINDGEN_EXTRA_CLANG_ARGS="-target gcc-aarch64-linux-gnu" \
  --env RUSTFLAGS="-C linker=aarch64-linux-gnu-gcc" \
  --volume $(pwd)/electrs:/source:ro \
  --volume $(pwd)/target:/target \
  --workdir /source \
  raspibuilder \
  cargo build \
  --locked \
  --release \
  --no-default-features \
  --target-dir /target \
  --target aarch64-unknown-linux-gnu

scp target/aarch64-unknown-linux-gnu/release/electrs admin@raspibolt.local:/home/admin
```

Then ssh to the Pi, test whether the binary can be executed, and `sudo mv` it to the final destination.

VajraOfIndra, 2022-05-15:
> Instead of installing Rust on the Raspberry Pi, I cross-compile Electrs in a Docker container.

Thanks @purpleKarrot! We don't user Docker or other types of containers in the RaspiBolt guides, but it's nice to have your instructions here in case someone is looking to do the same :+1:

VajraOfIndra, 2022-07-15:
Above updates (https://github.com/raspibolt/raspibolt/issues/747#issuecomment-1056786273) still to be implemented.

Kixunil, 2022-09-13:
Still not stale, I think.

Kixunil, 2022-09-13:
The instructions by @purpleKarrot don't require Docker, just any Debian 11 system.

VajraOfIndra, 2022-09-24:
_Trying to clarify where we're at with this thread and out Electrs setup._

1) ALREADY ADDRESSED

I ticked the recommendations that have already been addressed in the OP.
In the [second list down the thread](https://github.com/raspibolt/raspibolt/issues/747#issuecomment-1056786273), the Tor and blue led suggestions have already been addressed when the desktop wallet section was removed and a new Sparrow wallet guide was created.

---

2) ADDRESSED IN ON-GOING PR

The following recommandations are addressed in PR #1102 and can be discussed there:
- Dynamic linking when compiling
- Use of `CARGO_NET_GIT_FETCH_WITH_CLI=true` when compiling to prevent possible bug
- Hardening systemd service with `ProtectSystem=full` and `ProtectHome`
- Use of `git clean` and `git verify-tag` when updating & reworded title to "# Restart electrs"

---

3) NOT ADDRESSED YET, TO BE DISCUSSED

The following recommendations have not been addressed yet and requires further discussions (@Stadicus @kristapsk @RooSoft @blckbx):

**1) Double encryption**
> Don't double-encrypt traffic via onion+ssl, use onion only; suggest skipping nginx+TLS if not needed.
I can't comment on that one.. Is there an advantage (or disadvantage) to direct the Tor request to port 50002 (the reverse proxy for Electrs) rather than 50001?

**2) raspibolt.local**
@s-wuest: 
> replace raspibolt.local by the actual IP address ($ ip addr show).

@Kixunil: 
> Agree, .local names are very unreliable in my experience.

@visskiss: 
> I prefer .local names. Good networks shouldn’t have an issue with this.

Again, can't comment much on that one (except that I can't use raspibolt.local). Just using the IP would simplify the guide a bit, but at the same time offering the 2 options is not too hard (as we do now).

**3) System user**
> Why not adduser --system? It should be more bare-bones but maybe I'm missing something.
I looked a bit into user vs system user.. Looks like it's more semantic than anything else (system users are more for daemons). However, system user don't have a home directory by default.
As we never user system users in the guide, I would think we should stick to a normal user for Electrs. 
Would it make sense/be beneficial to start using system users in the guide?

**4) Runtime directory permissions**
> Why runtime directory mode is 710 instead of 750?
> I'm afraid without Group=electrs the group will be root DYOR

$ cd /run
$ ls -la
> drwx--x---  2 electrs     electrs       40 Sep 24 10:16 electrs

$ cd /run/electrs
$ ls -la
> drwx--x---  2 electrs electrs  40 Sep 24 10:16 .
> drwxr-xr-x 33 root    root    960 Sep 24 10:16 ..

@Kixunil what do you mean by the group will be root?

@Stadicus et al, any thoughts on that point?

blckbx, 2022-09-24:
> Again, can't comment much on that one (except that I can't use raspibolt.local). Just using the IP would simplify the guide a bit, but at the same time offering the 2 options is not too hard (as we do now).

Agree with you. Personally don't use .local addresses as well but it's easy to resolve them and follow the guide using IP only. 

> As we never user system users in the guide, I would think we should stick to a normal user for Electrs.
> Would it make sense/be beneficial to start using system users in the guide?

Again agreeing with you. 

> 4) Runtime directory permissions

There's a sample file we could adopt: https://github.com/romanz/electrs/blob/master/contrib/electrs.service and/or combine this with the [RaspiBlitz' approach](https://github.com/rootzoll/raspiblitz/blob/v1.8/home.admin/config.scripts/bonus.electrs.sh#L405-L426).

Kixunil, 2022-09-28:
1) Double encryption

There is no advantage I can think of and a load of disadvantages:

* requires the whole `nginx`, generating certificate, setting up and maintaining proxy...
* one more process that can fail
* consumes more CPU
* may cause bogus "invalid TLS cert" errors on the clients or similar problems
* some clients may even refuse to support it
* increased attack surface

2) raspibolt.local

I remember other people say it's unreliable. Both options sound good and the unreliability should be noted. "Good networks" means pretty much "networks operated by networking geeks". Someone with $20 router probably doesn't have "Good network"

3) System user

> However, system user don't have a home directory by default.

This isn't a problem for electrs since it doesn't need it. Just set appropriate flags and it'll work.

> Would it make sense/be beneficial to start using system users in the guide?

I think it could. It'd give off "don't use this as a usual user" vibe which is good. OTOH it adds more information to readers to think about so depending on how much the guide is supposed to teach people it may be less ideal.

4) Runtime directory permissions

> `drwx--x--- 2 electrs electrs 40 Sep 24 10:16 electrs`

Yep, listing is not allowed for group but access is. I don't think it achieves anything interesting. 750 makes more sense to me.

> $ cd /run/electrs
> $ ls -la

It looks like you ran it before electrs created the database. Could you run it after the database is created?

Stadicus, 2022-10-26:
I'm happy to let you guys lead the charge on revamping the Electrs guide. I'll just note some thoughts of mine here.

### Double encryption

Using SSL over Tor is unnecessary, which I learned well after first implementing it. But that only means that the Tor config should point directly to the HTTP web service. I'm not a fan of Tor-only web services, as these are slow and come with additional headaches on connecting devices.

Connecting to Electrs from within your own home network should not route over Tor. I prefer connecting over HTTPS for added privacy, but HTTP itself would also suffice for this use case. Other applications, e.g. RTL, however never expose their interfaces over plain HTTP, as it's trivial to capture login credentials. For this, NGINX is definitely needed.

Is the goal to get rid of the "double encryption", or provide just Electrs via HTTP and disregard this local NGINX config, or get rid of NGINX completely?

### raspibolt.local

I've never had any issues with local domain names, and they're very handy when using web interfaces (who remembers the IP address to connect daily?). For internal config, IP addresses should be better, but only if they don't change over time. The only guarantee for this is to map the device MAC address to a static IP address in the router config, which is not something I'm keen on doing. In practice, most IP addresses won't change, though, I'm just highlighting that IP addresses could also create issues down the line within "poor" networks.

I'm fine with whatever solution as long as the user-facing web services remain reachable with rapibolt.local

### System users

Semantically, system users might be better for system daemons that only use standard system directories. I'm not so sure about larger applications which are user configured. For Electrs, it might be easy, as it doesn't use a home directory. But I still value the option to log into the electrs user and start the application myself for debugging purposes.

With other applications, like Bitcoin Explorer or RTL, that are downloaded and installed in their user's home directory, a system user won't work. So we'll end up with a mix of different users, which could lead to confusion overall. 

### Runtime directory permissions

#### Group "electrs" 
Is there a need for users of group "electrs" (except user "electrs") to access the directory? If not, group "root" is default and more strict than setting it to "electrs", as root can access anything anyway.

#### Permissions `710` vs `750`
* The `R` allows to "read" this directory, e.g. list its content
* The `X` "execute" permission is a bit more tricky and allows to "traverse" the directory without listing its content

I'm not quite sure if I remember this correctly, but I think with `710` you can't list the directory (if you user is just in the "group"), but you can go to a sub-directory and list it if this has additional permissions. E.g. with `700` you can't list a subdirectory, even if it has `750` permissions.

I don't think it matters much. In the end, I'd just try and see what works .

VajraOfIndra, 2022-10-26:
> ### Double encryption
> Is the goal to get rid of the "double encryption", or provide just Electrs via HTTP and disregard this local NGINX config, or get rid of NGINX completely?

In my understanding, it was to just get rid of the double encryption, i.e. to change the Tor hidden service config file for Electrs by using port using 50001. Connection within the LAN would still be using the nginx reverse proxy.

> ### raspibolt.local 
> [...]
> I'm fine with whatever solution as long as the user-facing web services remain reachable with rapibolt.local
Ok, maybe let's continue as it is: i.e. use raspibolt.local by default but remind users to use their IP address if .local does not work for them.

> ### System users
> [...] But I still value the option to log into the electrs user and start the application myself for debugging purposes.
> [...] So we'll end up with a mix of different users, which could lead to confusion overall.

Ok, to avoid confusion and because it would apply to only one user, maybe let's continue with using normal users.

> ### Runtime directory permissions
> [...]
> I don't think it matters much. In the end, I'd just try and see what works .

Is your runtime directory empty as well? if so, is that expected behaviour? I don't know much about how runtime directories are used..

I'll update my PR with the above and with the subsequent comments by others, depending on what the consensus seem to be on the various topics.

closed 2023-01-25 by github-actions[bot]