raspibolt/raspibolt #748: LND configuration chapter small issue

derrickdeek, 2021-08-02:
A small issue while trying to set up lnd.conf
On line 10 of the lnd.conf I got any error:
/home/bitcoin/.lnd/lnd.conf:10: unknown option: protocol.anchors
When checking sample lnd conf files it seems that you need to use protocol.**no-anchors**=true
Then lnd can configure correctly

VajraOfIndra, 2021-08-03:
Yes, there is a mismatch between the LND version in the guide (v.0.11) and the mentionned anchor option which afaik only works for v.0.12.

I'm working on a PR that updates the guide to the latest v0.13 version which has anchor channels by default (so the anchor channel option can be deleted altogether, unless one does NOT want to use anchor channels, and in that case would need to use the no-anchors option that you mention - but I believe leaving the default is best for the guide, right?)

xanoni, 2021-08-07:
I actually sent a tiny PR to `lnd` for this a month ago, which was accepted .. see here .. https://github.com/lightningnetwork/lnd/pull/5439 

TL;DR: the new setting is called `protocol.no-anchors=false` (default)

PR #752 (2021-08-29): Update raspibolt_40_lnd.md (many updates of the lnd/litd guide)
Summary: the updates aim at:
1. using the latest lnd and litd releases,
2. correcting typos and broken links,
3. adding useful options to the lnd/litd configuration files,
4. replacing outdated commands and examples and warning for outdated videos, and;
5. adding a few safety information for some commands.

The full list of changes and rationales for them:
1. Added warning above most videos: (Warning: the video below is outdated and does not correspond exactly to the commands above)
2. Updated the 'download and install' section wi [... cut]
closed 2021-08-29 by Stadicus