raspibolt/raspibolt #749: Problem with syncing bitcoin core to external ssd with raspberry pi 4

n-e-t-d-i-v-e-r, 2021-08-03:
I’ve got a problem with initial sync from bitcoin blockchain to external ssd. Since days it loops between 8.6.2015 and 28.6.2015 and dosen’t go further. I read in this forum a reindex will solve this issue. Can somebody tell me the exactly terminal commands for this action on MacOS?

2021-08-03T06:55:19Z Cache configuration:
2021-08-03T06:55:19Z * Using 2.0 MiB for block index database
2021-08-03T06:55:19Z * Using 127.8 MiB for transaction index database
2021-08-03T06:55:19Z * Using 8.0 MiB for chain state database
2021-08-03T06:55:19Z * Using 886.2 MiB for in-memory UTXO set (plus up to 286.1 MiB of unused mempool space)
2021-08-03T06:55:19Z init message: Loading block index...
2021-08-03T06:55:19Z Switching active chainstate to Chainstate [ibd] @ height -1 (null)
2021-08-03T06:55:19Z Opening LevelDB in /mnt/ext/bitcoin/blocks/index
2021-08-03T06:55:21Z Opened LevelDB successfully
2021-08-03T06:55:21Z Using obfuscation key for /mnt/ext/bitcoin/blocks/index: 0000000000000000
2021-08-03T06:55:34Z LoadBlockIndexDB: last block file = 281
2021-08-03T06:55:34Z LoadBlockIndexDB: last block file info: CBlockFileInfo(blocks=202, size=102187753, heights=359632...360653, time=2015-06-06...2015-06-12)
2021-08-03T06:55:34Z Checking all blk files are present...
2021-08-03T06:55:35Z Opening LevelDB in /mnt/ext/bitcoin/chainstate
2021-08-03T06:55:35Z Opened LevelDB successfully
2021-08-03T06:55:35Z Using obfuscation key for /mnt/ext/bitcoin/chainstate: d2dabf5319b9493a
2021-08-03T06:55:35Z Loaded best chain: hashBestChain=000000000000000003449d6490882fbe2e02e5ddf0725b7c0704eb76e87584e0 height=359991 date=2015-06-08T13:00:33Z progress=0.107062
2021-08-03T06:55:35Z init message: Rewinding blocks...
2021-08-03T06:55:37Z FlushStateToDisk: write coins cache to disk (0 coins, 0kB) started
2021-08-03T06:55:37Z FlushStateToDisk: write coins cache to disk (0 coins, 0kB) completed (0.00s)
2021-08-03T06:55:37Z init message: Verifying blocks...
2021-08-03T06:55:37Z Verifying last 6 blocks at level 3
2021-08-03T06:55:37Z [0%]...[16%]...[33%]...[50%]...[66%]...[83%]...[99%]...[DONE].
2021-08-03T06:55:47Z No coin database inconsistencies in last 6 blocks (5166 transactions)
2021-08-03T06:55:47Z  block index           27711ms
2021-08-03T06:55:47Z Opening LevelDB in /mnt/ext/bitcoin/indexes/txindex
2021-08-03T06:55:48Z Opened LevelDB successfully
2021-08-03T06:55:48Z Using obfuscation key for /mnt/ext/bitcoin/indexes/txindex: 0000000000000000
2021-08-03T06:55:48Z txindex thread start
2021-08-03T06:55:48Z txindex is enabled at height 359991
2021-08-03T06:55:48Z txindex thread exit
2021-08-03T06:55:48Z loadblk thread start
2021-08-03T06:55:48Z block tree size = 693897
2021-08-03T06:55:48Z nBestHeight = 359991
2021-08-03T06:55:48Z torcontrol thread start
2021-08-03T06:55:48Z Bound to 127.0.0.1:8333
2021-08-03T06:55:48Z Bound to 127.0.0.1:8334
2021-08-03T06:55:48Z init message: Loading P2P addresses...
2021-08-03T06:55:48Z tor: Got service ID (private), advertising service (tor-config-private).onion:8333
2021-08-03T06:55:48Z AddLocal(tor-config-private)..onion:8333,4)
2021-08-03T06:55:48Z Loaded 45066 addresses from peers.dat  496ms
2021-08-03T06:55:48Z ERROR: DeserializeFileDB: Failed to open file /mnt/ext/bitcoin/anchors.dat
2021-08-03T06:55:48Z 0 block-relay-only anchors will be tried for connections.
2021-08-03T06:55:48Z init message: Starting network threads...
2021-08-03T06:55:48Z net thread start
2021-08-03T06:55:48Z dnsseed thread start
2021-08-03T06:55:48Z Waiting 300 seconds before querying DNS seeds.
2021-08-03T06:55:48Z addcon thread start
2021-08-03T06:55:48Z init message: Done loading
2021-08-03T06:55:48Z msghand thread start
2021-08-03T06:55:48Z opencon thread start
2021-08-03T06:55:51Z UpdateTip: new best=0000000000000000060fe13a6a2eba5c666469037ddb849e29b6060e1bd2273f height=359992 version=0x00000002 log2_work=82.905108 tx=71328349 date='2015-06-08T13:28:01Z' progress=0.107064 cache=0.8MiB(6572txo)
2021-08-03T06:55:54Z Update

n-e-t-d-i-v-e-r, 2021-08-05:
in meanwhile i reconfigured the whole installation several times but always with the same result. Syncing loops from the beginning of june till end june 2015 without any progress. i changed the external ssd and the ssd card, same result. may somebody help me here.
last week it worked but i tried to install coinjoin and that broke my whole installation with the result that i can’t sync bitcoin core anymore ...
thx for supporting me

n-e-t-d-i-v-e-r, 2021-08-05:
HALLELUJA!!! 

1) cd /mnt/ext/bitcoin/
2) rm -rf /mnt/ext/bitcoin/peers.dat
3) sudo reboot

Solved my problem finaly!

n-e-t-d-i-v-e-r, 2021-08-08:
After asking for help in various forums for over a week, i have now found the solution myself. Unfortunately, i haven't felt any of this often-cited helpfulness in this community. The problem that the initial synchronisation of the bitcoin blockchain can stuck and get into a loop is well known, however. it happens rarely, but if you get into the situation yourself, you'll be glad to find out the solution to this problem quickly.

Anyway, the sync is finally running smoothly again for me and I'm looking forward to being able to connect my fullnode to the bitBox2 soon.

So here are two possible solutions for all non-programmers and IT professionals on how to get the sync working again:

Solution 1: delete the file peer.dat
- log in as admin user and enter the following terminal commands
- $ sudo service bitcoind stop
- $ sudo su
- $ cd /mnt/ext/bitcoin/
- $ ls - la (check if you can find the file peer.dat in the directory)
- $ rm peer.dat
- $ ls - la (check if the file peer.dat has been deleted from the directory)
- $ sudo service bitcoind start

with this solution, my sync unfortunately only ran for a month at a time, only to get lost in an endless loop again.

Solution 2: reindex
- log in as admin and enter the following terminal commands
- $ sudo service bitcoind stop
- $ sudo su - bitcoin
- $ nano /mnt/ext/bitcoin/bitcoin.conf
- $ replace here txindex=1 with reindex=-1 (save, exit)
- $ sudo reboot

Wait a few minutes log in as admin user again and check if the reindex has started: sudo tail -f /mnt/ext/bitcoin/debug.log

Don't forget to replace reindex=-1 with txindex=1 in the bitcoin.conf file after a successful reindexing, otherwise your fullnode will be reindexed after every reboot again and again ...

Translated with www.DeepL.com/Translator (free version)

closed 2021-08-08 by n-e-t-d-i-v-e-r

Stadicus, 2021-08-08:
Thanks for providing the solution here after figuring out how to solve issue, much appreciated!

VajraOfIndra, 2021-08-09:
Glad you fixed your issue.

In case you're not aware of them, we now have two communities platforms where you can ask for help: the [Raspibolt Reddit sub](https://www.reddit.com/r/raspibolt/) and a [Telegram group](https://t.me/raspibolt). Still small so you might not get any replies to issues you post (that was the case for me! :) but worth trying.

I actually mentionned your issue 6 days ago in a [Reddit post ](https://www.reddit.com/r/raspibolt/comments/ox0vb0/new_github_issues_suggestions_for_outdated/ )linked to our TG group, but it got 0 replies unfortuantely. I posted a link to your solution there for future reference.