raspibolt/raspibolt #757: Bitcoind connection refused

Mexxblitz, 2021-08-28:
Hello Community, 

My full node stopped synching the blockchain about 4 days ago. It does not connect to any peers and so it does not synch further than block 697259. It just stops at this point. 

I really don't know how to fix this problem. I am a complete tech noob. What I did was to try monitoring bitcoind log files but the following message occurred: 

2021-08-28T12:54:08Z connect() to 127.0.0.1:9050 failed after select(): Connection refused (111)

Does anyone know why my node does not connect to others? How can I fix this problem so my node can start to fully synch the blockchain again? 

I appreciate all your help!

VajraOfIndra, 2021-09-03:
I don't know how to help you with your issue, but I will share it on the Raspibolt Reddit sub and TG group in case someone might be able to help.
Good luck troubleshooting it!

Stadicus, 2021-11-27:
Were you able to solve this issue?

Mexxblitz, 2022-01-26:
Yes sir, Problem solved

Stadicus, 2022-02-01:
Nice, thanks for your feedback. Closing.

closed 2022-02-01 by Stadicus