raspibolt/raspibolt #758: Issue with verifying SHA256SUS.asc from release bitcoin-core-22.0

n-e-t-d-i-v-e-r, 2021-09-16:
Hi there
i just wanted to verify the current bitcoin release bitcoin-core-22.0 and got the following message:
[image]

Is this a problem or can i continue with the upgrade without hesitation?

bmml1, 2021-09-16:
Hello,

The verification process changed
There is now a SHA256SUMS & SHA256SUMS.asc

First you need to download and import some developer keys like described here:
https://github.com/bitcoin/bitcoin/tree/master/contrib/builder-keys
`wget https://raw.githubusercontent.com/bitcoin/bitcoin/master/contrib/builder-keys/keys.txt`

Then download the other files
```
wget https://bitcoincore.org/bin/bitcoin-core-22.0/SHA256SUMS.asc
wget https://bitcoincore.org/bin/bitcoin-core-22.0/SHA256SUMS
```

Then do the sha256sum check
`sha256sum --check SHA256SUMS --ignore-missing`
Then 
`gpg --verify SHA256SUMS.asc `

more context:
https://github.com/bitcoin/bitcoin/issues/22634

dominicus, 2021-10-03:
I downloaded bitcoin core 22.0, adapting the commands as @bmml1 states.
The binary is verified OK, but some of the signatures were not.
Am I missing a needed step here?

```
# download Bitcoin Core binary
$ wget https://bitcoincore.org/bin/bitcoin-core-22.0/bitcoin-22.0-arm-linux-gnueabihf.tar.gz
$ wget https://bitcoincore.org/bin/bitcoin-core-22.0/SHA256SUMS.asc
$ wget https://bitcoincore.org/bin/bitcoin-core-22.0/SHA256SUMS

# check that the reference checksum matches the real checksum
# (ignore the "lines are improperly formatted" warning)
$ sha256sum --check SHA256SUMS --ignore-missing
> bitcoin-22.0-arm-linux-gnueabihf.tar.gz: OK

# import the public key of developers, verify the signed checksum file
# and check the fingerprint again in case of malicious keys
$ wget https://raw.githubusercontent.com/bitcoin/bitcoin/master/contrib/builder-keys/keys.txt
$ gpg --refresh-keys
$ gpg --verify SHA256SUMS.asc
> gpg: assuming signed data in 'SHA256SUMS'
> gpg: Signature made Fri 10 Sep 2021 04:29:17 PDT
> gpg:                using RSA key 0CCBAAFD76A2ECE2CCD3141DE2FFD5B1D88CA97D
> gpg: Can't check signature: No public key
> gpg: Signature made Thu 09 Sep 2021 13:09:04 PDT
> gpg:                using RSA key 152812300785C96444D3334D17565732E08E5E41
> gpg:                issuer "achow101@gmail.com"
> gpg: Good signature from "Andrew Chow (Official New Key) <achow101@gmail.com>" [unknown]
> gpg:                 aka "Andrew Chow <achow101-github@achow101.com>" [unknown]
> gpg:                 aka "Andrew Chow <achow101-lists@achow101.com>" [unknown]
> gpg:                 aka "Andrew Chow <achow101@pm.me>" [unknown]
> gpg:                 aka "Andrew Chow <achow101@protonmail.com>" [unknown]
> gpg:                 aka "Andrew Chow <achow101@yahoo.com>" [unknown]
> gpg:                 aka "Andrew Chow <andrew@achow101.com>" [unknown]
> gpg: WARNING: This key is not certified with a trusted signature!
> gpg:          There is no indication that the signature belongs to the owner.
> Primary key fingerprint: 1528 1230 0785 C964 44D3  334D 1756 5732 E08E 5E41
> gpg: Signature made Thu 09 Sep 2021 13:16:18 PDT
> gpg:                using RSA key 0AD83877C1F0CD1EE9BD660AD7CC770B81FD22A8
> gpg:                issuer "benthecarman@live.com"
> gpg: Good signature from "Ben Carman <benthecarman@live.com>" [unknown]
> gpg: WARNING: This key is not certified with a trusted signature!
> gpg:          There is no indication that the signature belongs to the owner.
> Primary key fingerprint: 0AD8 3877 C1F0 CD1E E9BD  660A D7CC 770B 81FD 22A8
> gpg: Signature made Fri 10 Sep 2021 06:00:35 PDT
> gpg:                using RSA key 590B7292695AFFA5B672CBB2E13FC145CD3F4304
> gpg:                issuer "darosior@protonmail.com"
> gpg: Good signature from "Antoine Poinsot <darosior@protonmail.com>" [unknown]
> gpg:                 aka "darosior <darosior@protonmail.com>" [unknown]
> gpg: WARNING: This key is not certified with a trusted signature!
> gpg:          There is no indication that the signature belongs to the owner.
> Primary key fingerprint: 590B 7292 695A FFA5 B672  CBB2 E13F C145 CD3F 4304
> gpg: Signature made Thu 09 Sep 2021 13:54:01 PDT
> gpg:                using RSA key 28F5900B1BB5D1A4B6B6D1A9ED357015286A333D
> gpg: Can't check signature: No public key
> gpg: Signature made Fri 10 Sep 2021 07:26:03 PDT
> gpg:                using RSA key 637DB1E23370F84AFF88CCE03152347D07DA627C
> gpg: Good signature from "Stephan Oeste (it) <it@oeste.de>" [unknown]
> gpg:                 aka "Emzy E. (emzy) <emzy@emzy.de>" [unknown]
> gpg:                 aka "Stephan Oeste (Master-key) <stephan@oeste.de>" [unknown]
> gpg: WARNING: This key is not certified with a trusted signature!
> gpg:          There is no indication that the signature belongs to the owner.
> Primary key fingerprint: 9EDA FF80 E080 6596 04F4  A76B 2EBB 056F D847 F8A7
>      Subkey fingerprint: 637D B1E2 3370 F84A FF88  CCE0 3152 347D 07DA 627C
> gpg: Signature made Thu 09 Sep 2021 18:04:14 PDT
> gpg:                using RSA key CFB16E21C950F67FA95E558F2EEB9F5CC09526C1
> gpg:                issuer "fanquake@gmail.com"
> gpg: Good signature from "Michael Ford (bitcoin-otc) <fanquake@gmail.com>" [unknown]
> gpg: WARNING: This key is not certified with a trusted signature!
> gpg:          There is no indication that the signature belongs to the owner.
[... 18 lines cut ...]
> gpg:                issuer "jon@atack.com"
> gpg: Good signature from "Jon Atack <jon@atack.com>" [unknown]
> gpg:                 aka "jonatack <jon@atack.com>" [unknown]
> gpg: WARNING: This key is not certified with a trusted signature!
> gpg:          There is no indication that the signature belongs to the owner.
> Primary key fingerprint: 8292 1A4B 88FD 454B 7EB8  CE3C 796C 4109 063D 4EAF
> gpg: Signature made Fri 10 Sep 2021 10:33:30 PDT
> gpg:                using RSA key 9DEAE0DC7063249FB05474681E4AED62986CD25D
> gpg: Good signature from "Wladimir J. van der Laan <laanwj@visucore.com>" [unknown]
> gpg:                 aka "Wladimir J. van der Laan <laanwj@gmail.com>" [unknown]
> gpg:                 aka "Wladimir J. van der Laan <laanwj@protonmail.com>" [unknown]
> gpg: WARNING: This key is not certified with a trusted signature!
> gpg:          There is no indication that the signature belongs to the owner.
> Primary key fingerprint: 71A3 B167 3540 5025 D447  E8F2 7481 0B01 2346 C9A6
>      Subkey fingerprint: 9DEA E0DC 7063 249F B054  7468 1E4A ED62 986C D25D
> gpg: Signature made Thu 09 Sep 2021 13:22:36 PDT
> gpg:                using RSA key 9D3CC86A72F8494342EA5FD10A41BDC3F4FAFF1C
> gpg:                issuer "aaron@sipsorcery.com"
> gpg: Can't check signature: No public key
> gpg: Signature made Fri 10 Sep 2021 02:59:33 PDT
> gpg:                using RSA key 74E2DEF5D77260B98BC19438099BAD163C70FBFA
> gpg:                issuer "will8clark@gmail.com"
> gpg: Can't check signature: No public key

```