raspibolt/raspibolt #766: UPGRADE LND: rm: cannot remove '/home/bitcoin/.lnd/data/chain/bitcoin/mainnet/readonly.macaroon':

n-e-t-d-i-v-e-r, 2021-10-09:
another question. wanted to upgrade lnd. for this i have to delete *.macaroon. but i always get this alert: rm: cannot remove '/home/bitcoin/.lnd/data/chain/bitcoin/mainnet/readonly.macaroon': Read-only file system no matter as admin-, bitcoin- or root-user where and how can i change rights for admin-user. probably you can write in here the right commands. thx

VajraOfIndra, 2021-10-21:
I don't think there is a need anymore to delete macaroons before upgrading LND. See for example @guggero's [reply](here:https://github.com/lightninglabs/lightning-terminal/issues/241#issuecomment-869848449).

Otherwise, not sure why you get this error message. Did you have a program running that was currently using this read-only macaroon maybe?

Stadicus, 2021-11-14:
Did you see this entry in the Troubleshooting section?

https://raspibolt.github.io/raspibolt/raspibolt_70_troubleshooting.html#is-you-root-filesystem-read-only

Stadicus, 2021-11-27:
I think this has been taken care of with the recent update to RaspiBolt v3. I'm closing this issue, but feel free to reopen it if you think it's still unsolved.

https://raspibolt.org/lnd.html#for-the-future-upgrade-lnd

closed 2021-11-27 by Stadicus