raspibolt/raspibolt #779: PROPOSAL: Create separate users for LND and Electrs (+ an "operator" user?)

VajraOfIndra, 2021-11-07:
This is a proposal to have separate users for LND, Electrs (as is already done for the BTC-RPC-Explorer).

Using one user per program can improve the safety of the overall node as permissions are easier to manage.
It also makes it easier to delete a program from the device.

This approach is used for example by the Raspiblitz and Nix Bitcoin projects (see for e.g. Nix Bitcoin security section ['Principle of Least Privilege'](https://github.com/fort-nix/nix-bitcoin#security)).

So for the main guide, there would be the following users:
- "bitcoin": for `bitcoind`
- "lnd": for `lnd`
- "electrs": running `electrs`
- "btcrpcexplorer": for `btc-rpc-explorer`

_Additional suggestion:_ Nix Bitcoin uses a non-root "operator" user that have access to client tools such as `bitcoin-cli`, `lncli` etc. Maybe we could also consider adding such a user?

As these changes would require a signifcant modifications of many parts of the guide (and careful considerations of how to implement least privileges without breaking something), I wanted to start a discussion here first to see if that is something that the community would like (or not) to see implemented.

Thanks in advance for reviewing and commenting on this proposal.

ephestione, 2021-11-08:
I am all for a separate user for each separate service, I like compartmentalization of privileges, that is inevitably going to make things more complicated to plan though, and possibly to manage (for example I have had to chmod g+rwx several times in some parts of external drive because it wasn't being inherited for new files, and that's just a minor issue) and I won't be able to help much as I'm all self taught about Linux and currently drained by LN for free time.
It's also possible this would look easy for someone else.

asvdf, 2021-11-08:
I'm all for segregating services in different users, I like this both semantically and from a security perspective, and yet in my specific case I wouldn't know where to start given my linux skills, this would require planning first, and then some workarounds to be able to use all services from a maintainer account as you said.
Maybe the task would be trivial to someone else where it looks a significant endeavor to me.

wankerstar, 2021-11-09:
Seems reasonable. An "operator" user would help simplify the user interaction and maintenance post-install, so that definitely make sense if we're adding more users.

Stadicus, 2021-11-14:
> This is a proposal to have separate users for LND, Electrs...

For **Electrs**, since the sync with Bitcoin Core is done through RPC, this is now possible. Before that, Electrs needed read access to the actual block files, which were not consistently created with the right group permissions. RPC sync fixes this.

I added this change to the upcoming RaspiBolt v3 guide, that also changes the Bitcoin Core authentication, with many other (breaking) changes. I don't think it's worth it to update the current v2 guide.

https://stadicus.github.io/raspibolt-remote/raspibolt_35_electrs.html

For **LND**, the change is pretty easy as well.

https://stadicus.github.io/raspibolt-remote/raspibolt_40_lnd.html

> Nix Bitcoin uses a non-root "operator" user that have access to client tools such as bitcoin-cli, lncli etc. Maybe we could also consider adding such a user?

I'm not sure if that's necessary, to be honest. With the "admin" user, you log in using SSH keys (no password), and to do any system damage, you'd actively need to run a `sudo` command with entering the password. Should be easy to add, though, now the cookie authentication of Bitcoin Core is enabled and the cookie is readable by the "bitcoin" group thanks to the `sysperms` option.

VajraOfIndra, 2021-11-14:
Yes, thinking of it, the 'operator' user makes maybe more sense for a setup like Nix Bitcoin rather than for the RaspiBolt since the RaspiBolt users built the whole thing and are probably well aware of what is risky or not to do.

Great to see the new users in v3!

closed 2021-11-14 by VajraOfIndra