raspibolt/raspibolt #783: PROPOSAL - Move "Static Channel Backups" from Bonus section to Lightning section, replace Dropbox with FOSS

wankerstar, 2021-11-10:
An automated system for keeping an off-site, up-to-date copy of the SCB file is a critical safeguard. Not having one puts the user at a definite risk of loss of funds due to an inability to close channels in case of total node failure.

I think this should be part of the standard lnd setup process, lest users think that the can safely skip it because it is in the Bonus section of the guide.

We might also replace Dropbox with something more open while we're at it. Two alternatives come to mind:

- @VajraOfIndra suggested using NextCloud, as RaspiBlitz does. This has the advantage of being simple and similar to Dropbox while also being self-hostable FOSS.
- On my own node Cornelius, I run a bash script as a service which uses inotifywait to monitor the channel.backup file. When it detects a change, it copies the file into a backup directory and then does a git commit and push. I use Bitbucket as the backup repository, but essentially the same script could be used with any git repo including self-hosted ones. This approach has the disadvantage that the user has to know or be instructed how to set up a git repository, but the advantage that it only relies on common Linux software and not a specific hosted backup solution.

I'm happy to start working on a PR for this using either solution, and I can share my script and service files and modify them as necessary.

@Stadicus , @VajraOfIndra , what do you think?

VajraOfIndra, 2021-11-10:
Thanks @wankerstar, I think this is an excellent proposal, as I also believe that the automatic backup of the SCB file is critical and therefore should be part of the main guide rather than as a bonus guide.

Moving from Dropbox to a FOSS alternative is a great idea!
Concerning NextCloud, I was merely noting that this is what the Raspiblitz project decided to move towards to ([v1.71](https://github.com/rootzoll/raspiblitz/blob/v1.7/CHANGES.md#whats-new-in-version-171-of-raspiblitz)). However, your solution looks very elegant and I don't know enough about the pros and cons of the two solutions/setups to be able to have an informed opinion on which one woud be most suitable for the RaspiBolt. So I'll just say I'm all for this proposal whether it's with option 1 or 2 :)

Stadicus, 2021-11-14:
I love the idea of storing the SCB in a purely FOSS online storage only using standard Linux CLI tools. Git seems like an obvious solution and has the added advantage that all backups would be versioned.

I tried to think of the easiest way to do this. When using GitHub, there's no way around creating an account, of course. But standard authentication is a pain. I think creating dedicated SSH key on the Pi for user "bitcoin", adding it to the GitHub repository as a "deploy key", and then cloning/pushing the repository through  SSH is easiest.

Quick guide draft, works for me :)

```
On github.com
* Create user
* Go to Repositories
* New / name & description / set to private

$ sudo su - bitcoin

$ ssh-keygen -t rsa -b 4096 -C "root@stadicus.com"
> Generating public/private rsa key pair.
> Enter file in which to save the key (/home/bitcoin/.ssh/id_rsa): 
> Enter passphrase (empty for no passphrase): 
> Enter same passphrase again: 
> Your identification has been saved in /home/bitcoin/.ssh/id_rsa
> Your public key has been saved in /home/bitcoin/.ssh/id_rsa.pub

$ cat /home/bitcoin/.ssh/id_rsa.pub

* In repository: Settings / Deploy keys
  * Add deploy key
  * Title: SCB"
  * Key: copy/paste output of `github_scb_deploy_key.pub`
  * Check "Allow write access"

$ cd /data

# name & email are required, you can put any value 
$ git config --global user.name "Stadicus"
$ git config --global user.email "root@stadicus.com"

$ git clone git@github.com:Stadicus/lnd-backup.git
$ cd lnd-backup

$ git add .
$ git commit -m "SCB"
$ git push --set-upstream origin master
```

The last three commands could be put in a shell script and triggered on change. Alternatively, the script could be run at regular intervals, as checking for file changes through `git` should be pretty simple.

VajraOfIndra, 2021-11-22:
That looks like a clean and neat solution. Do you plan to add it to the main guide in v3?

Stadicus, 2021-11-25:
To start with, the main guide just saves the SCB to the SSD, but I'd like to add at least some redundancy to store it on a USB thumbdrive. Could even be versioned (filename containing the date) to avoid overwriting it with a corrupted version.

This guide would be a cool (optional) addition to that, best as a bonus guide directly recommended in the main guide.

VajraOfIndra, 2021-11-25:
> To start with, the main guide just saves the SCB to the SSD, but I'd like to add at least some redundancy to store it on a USB thumbdrive. Could even be versioned (filename containing the date) to [...]

LGTM :) The USB thumbdrive backup for the main guide is a really great idea!

And an optional guide for the remote backup on Github in case of physical damage of the hardware stack seems like a nice solution to have additional resilience for people ok to create a Github account.

AsyncLuck, 2021-12-02:
hi @Stadicus 
`$ sudo su - bitcoin`

with v3 the ownership of the chan backup folder is set to lnd user (and for the systemd service too if i remember well)
so maybe `$ sudo su - lnd` for the git operations

Stadicus, 2021-12-03:
Updated the above instructions. These were from before the v3 guide introduced the "lnd" user.

Stadicus, 2022-02-01:
Nooooo, don't fade. We're making progress over here: https://github.com/raspibolt/raspibolt/pull/844

closed 2022-02-10 by VajraOfIndra