raspibolt/raspibolt #784: Recommend using Tor's own Debian repos instead of distribution's

wankerstar, 2021-11-12:
Raspbian Tor support isn't up to date. I've run into issues upgrading Tor before, as has a redditor just [recently](https://www.reddit.com/r/raspibolt/comments/qsb682/errors_upgrading_bitcoin_core/).

The solution on Raspbian is to enable Tor's own Debian repositories instead of relying on Raspbian to keep them updated. While changing OS might resolve this for the moment, I think we should recommend installing Tor from its own repos in the first place, both to eliminate the risk of this problem and to eliminate one link in the dependency stack.

There are instructions for this at https://support.torproject.org/apt/ , but they're a bit complex and overly general. I think we could recommend that user simply:

1. Add the following line (assuming Raspbian) to their /etc/apt/sources.list:
`deb https://deb.torproject.org/torproject.org buster main`

2. Run the following command (or something like it) to acquire the needed public keys:
`wget -O- https://deb.torproject.org/torproject.org/A3C4F0F979CAA22CDBA8F512EE8CBC9E886DDD89.asc | gpg --dearmor | tee /usr/share/keyrings/tor-archive-keyring.gpg >/dev/null`

3. Run apt update, apt install tor

Stadicus, 2021-11-14:
Isn't this exactly what the current guide describes?   
https://raspibolt.github.io/raspibolt/raspibolt_22_privacy.html#installing-tor

[image]

Stadicus, 2021-11-27:
Thanks for your contribution. I think this has been taken care of with the recent update to RaspiBolt v3. I'm closing this issue, but feel free to reopen it if you think it's still unsolved.

closed 2021-11-27 by Stadicus