raspibolt/raspibolt #787: Zeus not connecting to wallet after installing LiT

raelschiffler, 2021-11-14:
I don't have many evidence to show that would not impact OpSec, but since I installed LiT following the [tutorial](https://raspibolt.github.io/raspibolt/raspibolt_40_lnd.html#optional-using-lit-lightning-terminal-instead-of-lnd), Zeus won't connect anymore.
To be sure it was not a problem in tor and test it, I added a new node in Zeus with the same Macaroon and REST Port 8080, but without checking "Use Tor" and filling the Host with Raspibolt's internal LAN IP, and received a "Request timed out" error. Yes, the raspibolt and the device were on the same network, double-checked. The block explorer works fine on Android phone's browser using the same internal IP on port 3002 and Electrum mobile also uses perfectly raspibolt's Electrs as only server to connect through Tor (One-server mode: ON).
I need either help uninstalling LiT and reconfiguring LND as my unique Lightning service OR help configuring LiT to work with Zeus.
Does /home/bitcoin/.lit/lit.conf have an extra configuration or does Tor hidden services have to be configured to work with LiT?
Any help is appreciated.

`admin ~ ₿ sudo systemctl stop lnd`
`Failed to stop lnd.service: Unit lnd.service not loaded.`

`admin ~ ₿ sudo systemctl status litd`
`● litd.service - LND Lightning Network Daemon`
`   Loaded: loaded (/etc/systemd/system/litd.service; enabled; vendor preset: enabled)`
`   Active: active (running) since Sat 2021-11-13 21:06:04 -03; 38min ago`

VajraOfIndra, 2021-11-14:
Thinking about one possible issue: did you replace `--lnddir=/home/bitcoin/.lnd` by `--lnddir=/home/bitcoin/.litd` when trying to generate the LND connect URI QR code (or String) with the command: `lndconnect --lnddir=/home/bitcoin/.lnd --host=z1234567890abc.onion --port=8080`?

raelschiffler, 2021-11-16:
> Thinking about one possible issue: did you replace `--lnddir=/home/bitcoin/.lnd` by `--lnddir=/home/bitcoin/.litd` when trying to generate the LND connect URI QR code (or String) with the command: `lndconnect --lnddir=/home/bitcoin/.lnd --host=z1234567890abc.onion --port=8080`?

I surely tried this before opening the issue. First of all, the ".litd" directory does not exist, it's ".lit". Anyway, of course it could not work due to lndconnect search for lnd.conf and tls.cert files, and those are not present in .lit directory.

`$ lndconnect --lnddir=/home/bitcoin/.lit --host=z1234567890abc.onion --port=8080`
`open /home/bitcoin/.lit/lnd.conf: no such file or directory`
`open /home/bitcoin/.lit/tls.cert: no such file or directory`

VajraOfIndra, 2021-11-16:
Oh well, forget what I said then. I had Zeus running with LiT (with integrated LND) on the node in the past but have stopped using Zeus and LiT since so can't check my files to help, sorry.

raelschiffler, 2021-11-28:
Please, I really need help uninstalling LiT. I already closed all my channels to be sure I won't lose anything. But I don't know how to go ahead right now. @Stadicus @lucasmoten any help is welcome.

VajraOfIndra, 2021-12-14:
> Please, I really need help uninstalling LiT. I already closed all my channels to be sure I won't lose anything. But I don't know how to go ahead right now. @Stadicus @lucasmoten any help is welcome.

I switched back to LND from LiT in the past. What I did was to stop and disable litd (I also deleted/moved the lit.conf to avoid any possible restart even if there was no risk as the service was disabled), then checked my old lnd.conf that was still there, updated with my desried new options and then to started LND again. There is no need to close down the channels.

It might be too late now but if you join us on Telegram, several of us might be able to guide you through the process to completely uninstall LiT and use LND only if that's what you want to do.

closed 2022-02-24 by github-actions[bot]