raspibolt/raspibolt #788: PROPOSAL: some ideas to reorganize the Bonus section

VajraOfIndra, 2021-11-15:
So I had a go at reorganizing and updating the Bonus section, keeping in mind the vision of @Stadicus (see for example key points from the [recent AMA](https://t.me/raspibolt/3981)) and with the goals of:
- Correcting the mismatch between the left side Menu vs  the Bonus page's Table of Content
- Organizing the guides in logical categories for ease of navigation (particularly for new users)
- Making it easy to maintain and add new guides
- Making it clear that the core guide files are treated differently than the bonus guides

The draft version of the repo can be viewed here: [https://github.com/VajraOfIndra/RaspiBolt/tree/restructured_bonus_section](https://github.com/VajraOfIndra/RaspiBolt/tree/restructured_bonus_section)
How it would look in the website can be seen here: [https://vajraofindra.github.io/RaspiBolt/raspibolt_60_bonus.html](https://vajraofindra.github.io/RaspiBolt/raspibolt_60_bonus.html)

This is just a draft proposal to share some ideas and trigger some discussions on the topic of updating the Bonus section. There might be a much better way to structure it or many ways to improve upon it so feedback, improvement suggestions and alternative solutions more than welcomed!

---

_**Summary of the changes and rationales behind them:**_

- Classified the guides in 3 main baskets:
  1. **Node management**: anything related to OS, Linux admin, hardware etc
  2. **Bitcoin tools**:  any guides related to the base layer
  3. **Lightning tools**: any guides related to the LN
- Created additional layers of classification to help new users understand where does a bonus guide fits in within the entire stack. The entire structure is:
  1. Node management
     1. Dashboards & Appearance
     2. Privacy
     3. Resilience
     4. Security
     5. Hardware
  2. Bitcoin tools
     1. Electrum Servers
     2. Coinjoin (privacy)
     3. Multisig (security)
  3. Lightning tools
     1. Maintenance
     2. Dashboards & Wallets
        1. CLI-only
        2. GUI - Web
        3. GUI - Desktop
        4. GUI - Mobile
     3. Liquidity management (rebalancing)
     4. Fee management   
     5. Even more extras

- Removed the use of the Jekyll automatic Menu & TOC creation (using parent/children metadata) as it is 1) difficult to keep updated and matching; and 2) does not allow for the creation of a clean, compact and easy to navigate organized list of Bonus guides.
- Instead, used a simple manual table of content with links to guides
- Shortened the blurbs in the main 'Bonus' page and removed the illustrations to make it compact and easy to scroll up and down and understand where a guide fits within the overall architecture
- Created a new 'bonus' folder in the repo and moved all bonus guides to this folder: this is to make it easier for contributors to know which files are 'core' and which files are 'bonus'
- Replaced the difficulty ranking text by a colour-coded book icon to make the section more compact (see point 5) and more visisble

Stadicus, 2021-11-15:
Sounds like a great plan! :)

How would you feel about splitting the Bonus section in the three sub-sections you proposed?

* Bonus section: _Quick introduction, color scheme, maintenance guarantees (all high-level)_
  * Node maintenance: _separate page with list of relevant guides_ 
  * Bitcoin tools: _dito_
  * Lightning tools: _dito_

In my opinion that might be a bit better for navigation. And it would add the additional level to the breadcrumb path, e.g. 

> _Bonus section / Node maintenance / Pimp the CLI_

Just an idea, I think this proposal is already much better than today.

Stadicus, 2021-11-17:
FYI: instead of emojis, you could also use the labels provided by the "Just the docs" theme:

https://pmarsceill.github.io/just-the-docs/docs/ui-components/labels/

VajraOfIndra, 2021-11-19:
> How would you feel about splitting the Bonus section in the three sub-sections you proposed?
>
>     * Bonus section: _Quick introduction, color scheme, maintenance guarantees (all [...]

Yes, sounds great. So I split the bonus section in 3 sub-sections. I renamed the section to 'Raspberry Pi', 'Bitcoin' and 'Lightning' (instead of 'Node maintenance', 'Bitcoin tools' and 'Lightning tools' to match and correlated with the sections of the main guide.

I'm not veyr familiar with Jekyll and 'Just the docs' so slowly discovering how it all works. I've used the `nav_exclude=true ` in the yaml header of each bonus guides to ensure they don't appear in the main Menu on the left.
I've placed the index page of each section in the section folder, as shown in the Just the docs examples, but we could rename them and move them aroung if you prefer another structure. Right now, the structure looks like this:

|-- (main guide .md files, .yml, license, readme etc)
|-- bonus
|   |-- index.md (parent page)
|   |-- bitcoin
|   |   |-- index.md  (parent page)
|   |   |-- (bitcoin bonus .md files)
|   |-- lightning
|   |   |-- index.md  (parent page)
|   |   |-- (lightning bonus .md files)
|   |-- raspberry_pi
|   |   |-- index.md  (parent page)
|   |   |-- (raspberry pi bonus .md files)

I also added some collapsible menus at the top of the bonus guides as I think these menu are quite useful particulalry for long guides. 

> FYI: instead of emojis, you could also use the labels provided by the "Just the docs" theme

Do you mean as a replacement of the green, orange and red book emojis? I tried but I couldn't make it work as it seems there can be only one or more labels on one line but they can't be mixed with normal text... or if they can I didn't figure out how yet!
So I've removed the emojis for now.

We could use the labels at the top of each bonus guide though (green, yellow and red maybe, as there is no orange option afaik), I added them for now as a test, what do you think?

Let me know what you'd like to keep, what you'd prefer to change or any other suggestions on structure, naming convention and design; as the above changes is just to test ideas

VajraOfIndra, 2021-11-22:
Latest version can be seen here: https://vajraofindra.github.io/RaspiBolt/
It's a proof of concept for the navigation structure of the Bonus section.
I'll wait for comments and suggestions before changing anything more.

Later on, we could discuss additional things to do, such as:
* removing the guides that apply only to RaspiBolt v1 and v2 (as they are available in the respective v1 and v2 archive repos), i.e.
  *  Anonymous with Tor (by default v3)
  * System recovery (no use of SD card in v3)
  * Auto unlock LND on startup (by default in v3)
  * Automatic Static Channel Backup (SCB) to Dropbox (by default in v3, using Github)
* Putting a label at the top of each guide indicating if it has been tested on v3 and updated (e.g. "Tested & Working") or if it's outdated and not tested yet (e.g. "Possibly outdated") or any other more semantically meaningful labels - explanations for the labels could be given in the Bonus Section index page
* The sub-sections and individual guides could be added in the left Menu but it might become not easy to navigate (too many indentations/levels), hence for now they are not included in the left Menu  
* All guides to be tested and updated to work on v3

Structure of the section and titles of each sub-section to be reviwed and improved upon with possibly even more semantically meaningful categories (some categories will be added as new bonus guides will be merged)

Stadicus, 2021-11-23:
Me likes, and everything you write makes sense.

I noted that the breadcrumbs of the individual bonus guides don't work. Maybe worth looking into that.

And I think it would be worthwhile to replace the top-level bullets of the individual Bonus pages with proper H2 headings:

**Current structure**
---
* **Dashboards & Appearance**
  * System overview
  * Pimp the command line

* **Privacy**
  * Anonymous node with Tor
  * ...

---
**with H2 tags (`##`)
---
## Dashboards & Appearance
* System overview
* Pimp the command line
## Privacy
* Anonymous node with Tor
* ...

VajraOfIndra, 2021-11-23:
> I noted that the breadcrumbs of the individual bonus guides don't work. Maybe worth looking into that.

You mean in the left Menu? or the absence of the categories (e.g. 'Security')? 

The headings and sub-headings in each section (e.g. "Dashboards & Wallets", "GUI - Mobile") are not .md files as I feel it would lead to too many clicks to get to a specific guide and also make the left Menu very indented and crowded. But becasue of this, they don't appear in the breadcrumbs at the top of each guide nor in the left Menu..

Do you have some ideas how ot best handle this?
 
> And I think it would be worthwhile to replace the top-level bullets of the individual Bonus pages with proper H2 headings:

Absolutely, updated!

---
I removed the 'System recovery' and 'Tor' bonus guides as they are already part of the v3 main guide. For the autounlock LND wallet and the automatic SCB backup, I've left them for now and will remove then when, and if, you add these topics to the core guide of v3.

Regarding labels, is it what you had in mind when you mentionned them?
Should we consider adding some more?

For now we have:

Easy
{: .label .label-green }
Intermediate
{: .label .label-yellow }
Difficult
{: .label .label-red }

What about also:

Tested
{: .label .label-green }
Possibly outdated
{: .label .label-yellow }
Warning: Privacy
{: .label .label-red }
Warning: Security
{: .label .label-red }

with:
* "Tested" = guides that have been tested and are working as is on a RaspiBolt 3 setup
* "Possibly outdated" = guides that might be outdated or need some tweaks by the user to make it work on the RaspiBolt 3
* "Warning: Privacy" = Using this guide might lead to some loss of privacy, to be used only if you are aware and ok with the privacy implications
* "Warning: Security" = Using this guide might lead to some loss of security, to be used only if you are aware and ok with the secuirty implications

These are just a few examples, I'm sure you could find more useful and better worded labels! What do you think?

Finally, could we review the on-going pull-requests of bonus guides and see if we could include them in the reworked Bonus Section before v3 publication? it's be nice to have a few new bonus-guides from the get-go!

The PRs are:
* Balance Of Satoshis: it would go in Bonus Lightning / Liquidity Management
* rebalance-lnd: it would go in same sub-heading as above (PR to be published soon)
* charge-lnd: it would go in Bonus Lightning / Fee Management
* circuit-breaker: it would go in Bonus Lightning / Security
* Download the white paper: it would go in Bonus Bitcoin / Fun (or White Paper or something like that)

VajraOfIndra, 2021-11-24:
The breadcrumbs are now fixed! The issue came from a typo: I used `grand-parent` instead of `grand_parent`...

As discussed yesterday, I won't touch the branch (unless a change in structure is desired) and until it is merged into v3. Any additional improvements will be made after the merge as this will become a pull-request focused on mostly structure.

closed 2021-11-24 by VajraOfIndra