raspibolt/raspibolt #790: PROPOSAL (bonus section): decrease writing on SD Card with tmpfs

AsyncLuck, 2021-11-17:
For the guys who have problem with their sd cards.

On my side, I loose the "sudo" lib because of a bad disk error (worst error ever).

I m trying to use tmpfs like this 
`sudo nano /etc/fstab`

Add this lines, to mount system folders that need writing directly in RAM 
```
tmpfs    /tmp    tmpfs    defaults,noatime,nosuid,size=100m    0 0

tmpfs    /var/tmp    tmpfs    defaults,noatime,nosuid,size=100m    0 0

tmpfs    /var/log    tmpfs    defaults,noatime,nosuid,mode=0755,size=100m    0 0

tmpfs    /var/run    tmpfs    defaults,noatime,nosuid,mode=0755,size=2m    0 0
```

You can tweak the size (the space used in ram)

`sudo reboot`

If you use nginx, you can make a cron job to create its log folders/files because it doesn't create its structure by itself
`sudo crontab -e`

And add 
`@reboot mkdir /var/log/nginx
@reboot touch /var/log/nginx/access.log
@reboot touch /var/log/nginx/error.log'

Maybe can be integrated in the backup/restore section.

++

Stadicus, 2021-11-17:
Thanks for this, very interesting. I'm currently working on the version 3 guide, however, that gets rid of the microSD card entirely. It's definitely great to have this information documented somewhere, but I'm not sure where it would still be relevant in the next version.

AsyncLuck, 2021-11-18:
hi @Stadicus 
I think it is a great idea to get ride of the microSD => The weak point for a long term reliable pi node.
I will not have the motivation, for now, to migrate the system on the SSD and will try this ram method for some times. 

But for sure, it will be great, if you go with this approach (read only boot partition on the microSD + system on SSD or even no microSD at all) for the new happy followers of the raspibolt tuto ;)

Stadicus, 2021-11-27:
Thanks for your contribution. I think this has been taken care of with the recent update to RaspiBolt v3. I'm closing this issue, but feel free to reopen it if you think it's still unsolved.

closed 2021-11-27 by Stadicus

AsyncLuck, 2021-12-02:
@Stadicus it is not working good... SD card failure again on strategic files. I m going with Raspibolt v3, without SD card, just now :)