raspibolt/raspibolt #792: failed to load config: ValidateConfig: either bitcoin.active or litecoin.active must be set to 1 (true)

n-e-t-d-i-v-e-r, 2021-11-24:
Not sure if it’s a but or a mal configuration from my side but if i’ll run lnd as bitcoin user first time i get following allert:
«failed to load config: ValidateConfig: either bitcoin.active or litecoin.active must be set to 1 (true)»

I did configuration as described in the new manual here: https://stadicus.github.io/raspibolt-remote/lnd.html
And as far as I understand bitcoin.active is set true:
[image]

[image]

Any suggestions?

Stadicus, 2021-11-27:
Is this still an unsolved issue? I ran through the RaspiBolt v3 setup several times from start to finish and never experienced this.

xpsx, 2021-12-20:
I have the same problem, new setup

n-e-t-d-i-v-e-r, 2021-12-21:
problem was that lnd-user isn’t operable with admin user anymore. running lnd must be executed with the lnd-user on raspibolt v3

closed 2021-12-21 by n-e-t-d-i-v-e-r

Stadicus, 2021-12-21:
Maybe you set up your v3 before this was merged? User "admin" should definitely be able to run `lncli`:

https://raspibolt.org/lnd.html#allow-user-admin-to-work-with-lnd