raspibolt/raspibolt #794: Request to support NVME drives

tobysharp, 2021-11-26:
I'm using the guide to set up a different micro-PC with an M2 NVME SSD. It would be great to update the guide to support this kind of drive. I was able to run the command `sudo hdparm -t --direct /dev/nvme0n1p1` but it did include an error message of `NVMe status: INVALID_FIELD: A reserved coded value or an unsupported value in a defined field(0x2002)`. On looking this up, I discovered the `nvme` command, which has its own `nvme format` subcommand. So I do not know if that should be used here instead of `mkfs.ext4` or other commands. A brief update for this scenario would be very helpful.

Stadicus, 2021-11-29:
I think the `mkfs.ext` command works for all sorts of drives, I never heard that it's incompatible with NVME. But if you're formatting the drive anyway, I would just try and see if it works. I'd assume so.

Providing instructions for Non-Raspberry Pi hardware is out-of-scope for the "core" guide. The Raspberry Pi has no direct NVME support, and if you attache such a drive over USB (like I did myself), the instructions work. I think it would be a never-ending rabbit hole, distracting from the "beginner's guide" aspect that assumes a Raspberry Pi build.

If you have a good external link addressing this topic, however, or you'd like to provide a bonus guide that explains all the little differences for your specific micro-PC, that's be great!

VajraOfIndra, 2021-12-14:
> I'm using the guide to set up a different micro-PC with an M2 NVME SSD. It would be great to update the guide to support this kind of drive. I was able to run the command `sudo hdparm -t --direct /dev/nvme0n1p1` but it did include an error message of `HDIO_DRIVE_CMD(identify) failed: Inappropriate ioctl for device`. On looking this up, I discovered the `nvme` command, which has its own `nvme format` subcommand. So I do not know if that should be used here instead of `mkfs.ext4` or other commands. A brief update for this scenario would be very helpful. [image]

I'd be very interested to read any notes you might take about using the RaspiBolt but on on a microPC with M2 NVME SSD. I'd be happy to help with formatting them as a bonus guide if you'd like.. just let me know.

closed 2022-02-19 by github-actions[bot]