raspibolt/raspibolt #797: Any idea how to prevent electrum to connect autom

n-e-t-d-i-v-e-r, 2021-11-27:
Hi there

I try to connect my electrum wallet (macOS) with my private fullnode on raspibolt vs3. but allways if I try to connect electrum connects automatically, also if I uncheck the related box. As I can see there is another «rpc user and password filled in». As you can see on screenshot it is not my user which connect with electrum server:

[image]

and server is up and running:
[image]

also newest version of tor-browser is installed and running on personal computer.

is there a chance to fill in «rpc user and PW» manually into the electrum.config?  I think I’m near on a running system, but it seems I can’t find the solution for my problem. With the new manual there is no chance to change rpc credentials manually anymore.

Thx for any support.

Stadicus, 2021-11-27:
I'm not sure I understand what the problem is. 

Yes, Electrs must be fully synced, otherwise, you can't connect. You're right, that's not mentioned in the guide. Pull request https://github.com/raspibolt/raspibolt/pull/799 adds this remark

n-e-t-d-i-v-e-r, 2021-11-28:
thx for response. yes, the «issue» was the missing info in the manual to sync electrum server fully before trying to connect wallet and server. I will close this.

closed 2021-11-28 by n-e-t-d-i-v-e-r