raspibolt/raspibolt #801: JoinMarket is currently incompatible with RaspiBolt v3

kristapsk, 2021-11-27:
b845122d6e98b958d5aa6893fcef67599cdd0d57 disables Bitcoin Core wallet functionality, which is required for JoinMarket (watchonly, for address / transaction monitoring). So, if this isn't changed, link to JoinMarket guide in bonus section should be removed.

Stadicus, 2021-11-28:
Good point. Unfortunately, AFAIK, the Bitcoin Core wallet must be disabled to allow the `sysparms` option. And only then the Bitcoin Core authorization cookie can be made accessible through group permissions. 

Can you think of a better way to achieve permissions across multiple service users? I'd hate to go back to putting the rpcuser/pw into every configuration file.

kristapsk, 2021-11-28:
There is maybe alternative approach, but that requires development work from JM side. JoinMarket originally had Electrum Server backend support as alternative to Bitcoin Core, but it has been unmaintained for a long time and is currently broken. Need to look how much effort it would take to get it back running. Have never ever used it myself. Also have never looked too much into Electrum JSON-RPC.

It would have potentially other benefits, for example, RaspiBolt v3 also exports electrs to the outside world by default, but not so with bitcoind RPC, which would make easier to connect JoinMarketQt GUI running on your PC to the node.

kristapsk, 2021-11-29:
Looked at existing JM `ElectrumInterface` code, it's useless here. It's from the old times when JoinMarket plugin for Electrum was developed, it does not implement Electrum Server / Stratum protocol, it's just meant for that plugin, so it calls Electrum Python functions directly.

Still think Electrum Server backend support for JM would be cool, but it's not coming too soon.

> Can you think of a better way to achieve permissions across multiple service users?

Not a better way, but I found one hackish solution that seems to work in my tests - add `-startupnotify="chmod g+r /home/bitcoin/.bitcoin/.cookie"` argument to `ExecStart=...` in `/etc/systemd/system/bitcoind.service`. That will make cookie file group readable after `bitcoind` has finished it's startup sequence.

Stadicus, 2021-11-29:
Thanks for the info. Yeah, that seems pretty hackish, I wouldn't want the whole Bitcoin Core  API accessibility depend on that.

kristapsk, 2021-12-02:
The same change also breaks Specter Desktop Wallet bonus guide (haven't tested, but it requires wallet support in Bitcoin Core).

VajraOfIndra, 2021-12-04:
How do the other node projects handle this? ...by not using cookie authorization for bitcoin?

kristapsk, 2021-12-04:
@openoms You have experience with RaspiBlitz, how it's solved there?

openoms, 2021-12-11:
RaspiBlitz uses rpcuser and rpcpassword, not cookie auth.

VajraOfIndra, 2021-12-12:
> RaspiBlitz uses rpcuser and rpcpassword, not cookie auth.

Thanks @openoms. Is it because of these issues that you decided to not move to cookie auth for now..?

> The same change also breaks Specter Desktop Wallet bonus guide 

I don't understand the various options and pros and cons enough to contribute to the disucssion.. but losing both JoinMarket and Specter seems a heavy price to pay.

Stadicus, 2021-12-12:
I think it would be great to solve this issue, to enable compatibility with these services. I'm not a big fan of explicitly putting the rpcuser & rpcpassword into all config files, though.

Unfortunately, I'm not able to make the above suggestion from @kristapsk work, maybe because it's triggered too early?

> add -startupnotify="chmod g+r /home/bitcoin/.bitcoin/.cookie" argument to ExecStart=... in /etc/systemd/system/bitcoind.service. That will make cookie file group readable after bitcoind has finished it's startup sequence.

A solution that works reliably in my tests is the following addition to `/etc/systemd/system/bitcoind.service`, enforcing a 5s delay before setting the correct group permissions on the auth cookie:

```
ExecStart=/usr/local/bin/bitcoind -daemon \
                                  -pid=/run/bitcoind/bitcoind.pid \
                                  -conf=/home/bitcoin/.bitcoin/bitcoin.conf \
                                  -datadir=/home/bitcoin/.bitcoin
ExecStartPost=/bin/sleep 5
ExecStartPost=/bin/chmod g+r /data/bitcoin/.cookie
```

That allows to remove the following arguments from `/data/bitcoin/bitcoin.conf`:

```
#sysperms=1
#disablewallet=1
```

openoms, 2021-12-12:
@VajraOfIndra we did not "decide to not move to cookie auth", simply just did not try to implement it so far. We also reuse the rpcpassword as the Password_B in places.
An alternative is using rpcuth:
https://raw.githubusercontent.com/bitcoin/bitcoin/master/share/rpcauth/rpcauth.py
It would provide a different password for each app.
Discussed also in: https://github.com/openoms/joininbox/issues/30

Stadicus, 2021-12-14:
If anyone would like to give this a try, here's what to do. No worries, it's nothing permanent and all changes are easily reverted.

The goal is to avoid disabling the Bitcoin Core wallet, while still using cookie authentication, and make the .cookie file readable by the group "bitcoin". If this works reliably (it does for me), I think that's a viable alternative to the current setup, restoring compatibility for JoinMarket, Electrum Personal Server, Specter Wallet etc. again. I'd then update the main guide.

* Make sure Bitcoin Core is configured to use the `rpcauth` method [used in the RaspBolt v3](https://raspibolt.org/bitcoin-core.html#generate-access-credentials), and not the `rpcuser` / `rpcpassword` of older guides
* Disable the following options in the `bitcoin.conf` file:
  ```
  #sysperms=1
  #disablewallet=1
  ```
* Restart Bitcoin Core and make sure the `.cookie` file is generated. The `-rw-------` part means that only the user "bitcoin", but not the group "bitcoin" has (r)ead and (w)rite access to the file.
  ```
  $ sudo systemctl restart bitcoind
  $ $ ls -la /data/bitcoin | grep .cookie
  > -rw------- 1 bitcoin bitcoin      75 Dec 12 12:25 .cookie
  ```
* Add the following two lines right under the `ExecStart` line in the Bitcoin Core systemd unit that starts the application automatically:
  ```
  $ sudo nano /etc/systemd/system/bitcoind.service
  ```
  ```
  ExecStartPost=/bin/sleep 5
  ExecStartPost=/bin/chmod g+r /data/bitcoin/.cookie
  ```

* Reload the system unit and restart Bitcoin Core again. Then check if the group "bitcoin" now has (r)ead access to the file, as the additional `r` indicates in the output below:
  ```
  $ sudo systemctl daemon-reload
  $ sudo systemctl restart bitcoind
  $ ls -la /data/bitcoin | grep .cookie
  > -rw-r----- 1 bitcoin bitcoin      75 Dec 12 12:25 .cookie

  ```

kristapsk, 2021-12-16:
@Stadicus Not sure why it's not working for you. My service file:
```
admin ~ ₿ sudo cat /etc/systemd/system/bitcoind.service
[sudo] password for admin: 
# RaspiBolt: systemd unit for bitcoind
# /etc/systemd/system/bitcoind.service

[Unit]
Description=Bitcoin daemon
After=network.target

[Service]

# Service execution
###################

ExecStart=/usr/local/bin/bitcoind -daemon \
                                  -pid=/run/bitcoind/bitcoind.pid \
                                  -conf=/home/bitcoin/.bitcoin/bitcoin.conf \
                                  -datadir=/home/bitcoin/.bitcoin \
                                  -startupnotify="chmod g+r /home/bitcoin/.bitcoin/.cookie"

# Process management
####################
Type=forking
PIDFile=/run/bitcoind/bitcoind.pid
Restart=on-failure
TimeoutSec=300
RestartSec=30

# Directory creation and permissions
####################################
User=bitcoin
UMask=0027

# /run/bitcoind
RuntimeDirectory=bitcoind
RuntimeDirectoryMode=0710

# Hardening measures
####################
# Provide a private /tmp and /var/tmp.
PrivateTmp=true

# Mount /usr, /boot/ and /etc read-only for the process.
ProtectSystem=full

# Disallow the process and all of its children to gain
# new privileges through execve().
NoNewPrivileges=true

# Use a new /dev namespace only populated with API pseudo devices
# such as /dev/null, /dev/zero and /dev/random.
PrivateDevices=true

# Deny the creation of writable and executable memory mappings.
MemoryDenyWriteExecute=true

[Install]
WantedBy=multi-user.target
admin ~ ₿ ls -al /home/bitcoin/.bitcoin/.cookie
-rw-r----- 1 bitcoin bitcoin 75 Dec  5 12:36 /home/bitcoin/.bitcoin/.cookie
```

Your approach should work too, but, what worries me, are 5 seconds always enough? Need to look closer at Bitcoin Core startup sequence, at which point cookie file is created.

Stadicus, 2021-12-17:
Ha, I tried again, and now it works. Not sure what I did wrong before... I'll add this to the main config, so we can still have the wallet enabled.

PR #846 (2021-12-18): bitcoind: use 'startupnotify' for group perms
### What

This pull request changes the method to ensure that the .cookie file is group accessible.

### Why

Bitcoin Core is accessible to applications run by different users through the authentication cookie. Users that are a member of the group "bitcoin" need read-only access to /data/bitcoin/.cookie and can then authenticate themselves without username and password.

By default, Bitcoin Core does not create the .cookie file with group permissions. As of now, the bitcoind argument `sysperms=1`
is used with UMask=0027 to override this. Bitcoind o [... cut]
closed 2021-12-18 by Stadicus