raspibolt/raspibolt #804: Can’t start lnd

n-e-t-d-i-v-e-r, 2021-11-28:
Hello there

If I’ll run lnd, I get this allert:
[image]

That’s strange because lnd.conf seems to be OK:
[image]

Do you have an idea, what caused my problem with starting lnd? In the config-script for lnd.conf users have to fill in «YOUR_FANCY_ALIAS» isn’t that raspibolt? If not, where will I find this alias?

Thx for your support.

Stadicus, 2021-11-28:
Do you start `lnd` manually? Make sure to start it with user "lnd". There's a typo in the guide, that is addressed here: 

https://github.com/raspibolt/raspibolt/pull/798/files#diff-bf8910c1d3c557bce681fe6ac82b7fc9f8e45223b72c42ec5d2319c0ae47df4aR217

Mayte that's the issue?

n-e-t-d-i-v-e-r, 2021-11-28:
you’re right thx for the info. lnd is now up and running:
[image]

But if i go further in the manual and will generating a new bitcion address or checking wallet balance i get this:
[image]

No matter if user is admin, lnd, bitcoin or root ...

How can I solve this?

Stadicus, 2021-11-28:
As user "lnd", it definitely needs to work. If it doesn't, then there's something wrong with the directory / file permissions. Check if you can access `/data/lnd` and the subdirectories listed in the error message (just `cd /data/lnd` and go from there).

User "admin" can't use `lncli` at the moment, due to missing access privileges the same directories.

There are some ways to do that, not sure if it's a) necessary, and b) which one is best:

1. create the above directories in "admin" home and copy the necessary files (probably best)
1. add "admin" to the group "lnd" and set proper read-only permissions
1. link to `/data/lnd` from the "root" home directory and then run `sudo lncli`

@n-e-t-d-i-v-e-r @VajraOfIndra @Pantamis WDYT?

n-e-t-d-i-v-e-r, 2021-11-29:
thx for response. I tried to change file permissions. But if I try to point a symbolic link in the bitcoin directory with lnd user, i get this:

[image]

[image]

May you tell me witch which exactly command and from which user I can change this permissions?

With autostart i get this message:

[image]

Stadicus, 2021-11-29:
* User "lnd" must be a member of the group "bitcoin", that gives it access to the bitcoin data directory. Please check with `$ groups lnd`. Also note that changes to groups only take effect in a new user session, so if you add a group, log out and log in again.      
  https://raspibolt.org/lnd.html#data-directory

* I also see that the permissions on the `password.txt` are not set correctly   
  https://raspibolt.org/lnd.html#wallet-password

Can you provide the output of `sudo ls -la /data/bitcoin`? It should look like this:

```sh
drwxr-xr-x 5 bitcoin bitcoin    4096 Nov 29 07:17 .
drwxr-xr-x 6 bitcoin bitcoin    4096 Nov 25 12:41 ..
-rw-r----- 1 bitcoin bitcoin      75 Nov 25 12:45 .cookie
-rw-r----- 1 bitcoin bitcoin       0 Nov 25 06:54 .lock
-rw------- 1 bitcoin bitcoin      37 Nov 24 22:45 banlist.dat
-rw------- 1 bitcoin bitcoin      34 Nov 24 22:45 banlist.json
-rw-r--r-- 1 bitcoin bitcoin     656 Nov 25 06:55 bitcoin.conf
-rw-r--r-- 1 bitcoin bitcoin     416 Nov 24 22:45 bitcoin.conf.bak
drwx------ 3 bitcoin bitcoin  147456 Nov 29 02:10 blocks
drwx------ 2 bitcoin bitcoin   86016 Nov 28 22:30 chainstate
-rw------- 1 bitcoin bitcoin       0 Nov 24 22:45 db.log
-rw-r--r-- 1 bitcoin bitcoin  621457 Nov 29 07:24 debug.log
-rw------- 1 bitcoin bitcoin  247985 Nov 25 12:40 fee_estimates.dat
drwx------ 3 bitcoin bitcoin    4096 Nov 25 00:48 indexes
-rw-r----- 1 bitcoin bitcoin 5432180 Nov 25 12:40 mempool.dat
-rw-r----- 1 bitcoin bitcoin      99 Nov 25 12:45 onion_v3_private_key
-rw-r----- 1 bitcoin bitcoin 2750304 Nov 29 07:17 peers.dat
-rw-r----- 1 bitcoin bitcoin       7 Nov 25 12:45 settings.json
-rw------- 1 bitcoin bitcoin 4988928 Nov 24 22:45 wallet.dat
```

n-e-t-d-i-v-e-r, 2021-11-29:
result from my sudo ls -la /data/bitcoin is:
[image]

looks like there is a gap. Can you probably give me here the exact commands to fix this? So that in the end all directories and files are on the right place and with the needed permissions for every user? Or should I delete lnd and start from the scratch? If yes, how to delete all lnd related stuff proper?

Stadicus, 2021-11-29:
> looks like there is a gap

What do you mean exactly? Looks good to me, at least permission-wise. 

Please try the following commands starting with user "admin", just to be sure:

```sh
$ sudo su - lnd
$ ls -la
$ cd /data/bitcoin
$ ls -la
$ ln -s /data/bitcoin /home/lnd/.bitcoin
$ cd ~/.bitcoin
$ cd ~/.lnd
$ ls -la
$ cd /data/lnd/data/chain/bitcoin/mainnet
$ ls -la
```

Please copy/paste the output as a text here, using a code block using three backticks:

````
```
your code output
```
````

n-e-t-d-i-v-e-r, 2021-11-29:
`admin@raspibolt:~ $ sudo su - lnd
[sudo] password for admin: 
lnd@raspibolt:~ $ ls -la

total 28
drwxr-xr-x 2 lnd  lnd  4096 Nov 28 09:34 .
drwxr-xr-x 8 root root 4096 Nov 28 09:32 ..
-rw------- 1 lnd  lnd   324 Nov 29 08:22 .bash_history
-rw-r--r-- 1 lnd  lnd   220 Nov 28 09:32 .bash_logout
-rw-r--r-- 1 lnd  lnd  3523 Nov 28 09:32 .bashrc
lrwxrwxrwx 1 lnd  lnd    13 Nov 28 09:34 .bitcoin -> /data/bitcoin
lrwxrwxrwx 1 lnd  lnd     9 Nov 28 09:33 .lnd -> /data/lnd
-rw-r--r-- 1 lnd  lnd  1670 Nov 28 09:32 .mkshrc
-rw-r--r-- 1 lnd  lnd   807 Nov 28 09:32 .profile

lnd@raspibolt:~ $ cd /data/bitcoin
lnd@raspibolt:/data/bitcoin $ ls -la

total 14648
drwxr-xr-x 5 bitcoin bitcoin     4096 Nov 29 09:30 .
drwxr-xr-x 6 bitcoin bitcoin     4096 Nov 28 09:33 ..
-rw-r--r-- 1 bitcoin bitcoin       34 Nov 25 07:49 banlist.json
-rw-r----- 1 bitcoin bitcoin      655 Nov 27 08:19 bitcoin.conf
drwxr-xr-x 3 bitcoin bitcoin   151552 Nov 28 20:19 blocks
drwxr-xr-x 2 bitcoin bitcoin    81920 Nov 29 08:23 chainstate
-rw-r----- 1 bitcoin bitcoin       75 Nov 28 09:46 .cookie
-rw-r--r-- 1 bitcoin bitcoin 10229764 Nov 29 09:39 debug.log
-rw-r----- 1 bitcoin bitcoin   247985 Nov 28 09:46 fee_estimates.dat
drwxr-xr-x 3 bitcoin bitcoin     4096 Nov 25 07:49 indexes
-rw-r--r-- 1 bitcoin bitcoin        0 Nov 25 07:49 .lock
-rw-r----- 1 bitcoin bitcoin  1499791 Nov 28 09:46 mempool.dat
-rw-r--r-- 1 bitcoin bitcoin       99 Nov 28 10:15 onion_v3_private_key
-rw-r----- 1 bitcoin bitcoin  2729485 Nov 29 09:30 peers.dat
-rw-r--r-- 1 bitcoin bitcoin     1534 Nov 25 07:45 rpcauth.py
-rw-r----- 1 bitcoin bitcoin        7 Nov 28 09:46 settings.json

lnd@raspibolt:/data/bitcoin $ ln -s /data/bitcoin /home/lnd/.bitcoin
ln: failed to create symbolic link '/home/lnd/.bitcoin/bitcoin': Permission denied

lnd@raspibolt:/data/bitcoin $ cd ~/.bitcoin

lnd@raspibolt:~/.bitcoin $ cd ~/.lnd

lnd@raspibolt:~/.lnd $ ls -la

total 40
drwxr-xr-x 5 lnd     lnd     4096 Nov 29 05:32 .
drwxr-xr-x 6 bitcoin bitcoin 4096 Nov 28 09:33 ..
drwx------ 5 lnd     lnd     4096 Nov 28 11:34 data
drwx------ 2 lnd     lnd     4096 Nov 28 11:34 letsencrypt
lrwxrwxrwx 1 lnd     lnd        9 Nov 29 05:32 lnd -> /data/lnd
-rw-r--r-- 1 lnd     lnd     1061 Nov 28 09:43 lnd.conf
drwx------ 3 lnd     lnd     4096 Nov 28 11:34 logs
-rw-r--r-- 1 lnd     lnd       13 Nov 28 09:39 password.txt
-rw-r--r-- 1 lnd     lnd      847 Nov 28 11:34 tls.cert
-rw------- 1 lnd     lnd      227 Nov 28 11:34 tls.key
-rw------- 1 lnd     lnd       99 Nov 28 13:17 v3_onion_private_key

lnd@raspibolt:~/.lnd $ cd /data/lnd/data/chain/bitcoin/mainnet

lnd@raspibolt:/data/lnd/data/chain/bitcoin/mainnet $ ls -la

total 1200
drwx------ 2 lnd lnd    4096 Nov 28 11:57 .
drwx------ 3 lnd lnd    4096 Nov 28 11:34 ..
-rw-r----- 1 lnd lnd     293 Nov 28 11:55 admin.macaroon
-rw-r--r-- 1 lnd lnd      83 Nov 28 11:55 chainnotifier.macaroon
-rw-r--r-- 1 lnd lnd     132 Nov 28 11:55 invoice.macaroon
-rw-r--r-- 1 lnd lnd      91 Nov 28 11:55 invoices.macaroon
-rw-r--r-- 1 lnd lnd   32768 Nov 28 12:06 macaroons.db
-rw------- 1 lnd lnd       8 Nov 28 11:57 macaroons.db.last-compacted
-rw-r--r-- 1 lnd lnd     217 Nov 28 11:55 readonly.macaroon
-rw-r--r-- 1 lnd lnd      91 Nov 28 11:55 router.macaroon
-rw-r--r-- 1 lnd lnd      92 Nov 28 11:55 signer.macaroon
-rw------- 1 lnd lnd 2097152 Nov 29 09:25 wallet.db
-rw-r--r-- 1 lnd lnd     114 Nov 28 11:55 walletkit.macaroon`

Stadicus, 2021-11-29:
From all I can see you should be able to call `lncli` with user "lnd": have the right symlinks, you can enter all directories manually, list contents and the permissions look right.

What happens if you try it as user "lnd"?

```sh
$ sudo su - lnd
$ lncli getinfo
```

VajraOfIndra, 2021-11-29:
> @n-e-t-d-i-v-e-r @VajraOfIndra @Pantamis WDYT?

I tried to catch up with the thread and issue.. I can't help that much right now as I didn't reach the lnd install stage on my new node. 

From the outputs above, user "lnd" should be able to run `lncli` as the symlink is there in the home folder and the cert and maacroons are all there...
As user "admin", as shown in [ message 3](https://github.com/raspibolt/raspibolt/issues/804#issuecomment-981075616), it is normal it doesn't work as there did not seem to be any symlink in the admin home folder to go to the lnd folder (plus possibly missing group permissions).

But I don't see why user "lnd" wouldn't be able to run `lncli` sucessfully...

n-e-t-d-i-v-e-r, 2021-11-29:
In case I call lincli i get this:

```
lncli getinfo
{
    "version": "0.14.0-beta commit=v0.14.0-beta",
    "commit_hash": "df0b82f0165f19bde8832bacd1e35544b0a2990d",
    "identity_pubkey": "0207005dd4154b87e381a5ccdd9d1a903236cd32e0f74a21494607b89765601994",
    "alias": "raspibolt",
    "color": "#3399ff",
    "num_pending_channels": 0,
    "num_active_channels": 0,
    "num_inactive_channels": 0,
    "num_peers": 5,
    "block_height": 711805,
    "block_hash": "0000000000000000000a4e7b3dc2bd1562a97ace676597791db16f9be8be7240",
    "best_header_timestamp": "1638190094",
    "synced_to_chain": true,
    "synced_to_graph": true,
    "testnet": false,
    "chains": [
        {
            "chain": "bitcoin",
            "network": "mainnet"
        }
    ],
    "uris": [
        "abcTOR_address.onion:9735"
    ],
    "features": {
        "0": {
            "name": "data-loss-protect",
            "is_required": true,
            "is_known": true
        },
        "5": {
            "name": "upfront-shutdown-script",
            "is_required": false,
            "is_known": true
        },
        "7": {
            "name": "gossip-queries",
            "is_required": false,
            "is_known": true
        },
        "9": {
            "name": "tlv-onion",
            "is_required": false,
            "is_known": true
        },
        "12": {
            "name": "static-remote-key",
            "is_required": true,
            "is_known": true
        },
        "14": {
            "name": "payment-addr",
            "is_required": true,
            "is_known": true
        },
        "17": {
            "name": "multi-path-payments",
            "is_required": false,
            "is_known": true
        },
        "19": {
            "name": "wumbo-channels",
            "is_required": false,
            "is_known": true
        },
        "23": {
            "name": "anchors-zero-fee-htlc-tx",
            "is_required": false,
            "is_known": true
        },
        "30": {
            "name": "amp",
            "is_required": true,
            "is_known": true
        },
        "31": {
            "name": "amp",
            "is_required": false,
            "is_known": true
        },
        "45": {
            "name": "explicit-commitment-type",
            "is_required": false,
            "is_known": true
        },
        "2023": {
            "name": "script-enforced-lease",
            "is_required": false,
            "is_known": true
        }
    }
}
```

In case I call status with admin user i see this:

```
ystemctl status lnd
* lnd.service - LND Lightning Network Daemon
     Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: e>
     Active: active (running) since Sun 2021-11-28 12:06:48 GMT; 24h ago
   Main PID: 2007 (lnd)
      Tasks: 13 (limit: 8986)
        CPU: 6h 16min 57.130s
     CGroup: /system.slice/lnd.service
             `-2007 /usr/local/bin/lnd
```

If I call journalctl looks also find for me:

```
sudo journalctl -f -u lnd
-- Journal begins at Sat 2021-10-30 13:43:13 BST. --
Nov 29 12:49:53 raspibolt lnd[2007]: 2021-11-29 12:49:53.896 [INF] WTCL: (legacy) Client stats: tasks(received=0 accepted=0 ineligible=0) sessions(acquired=0 exhausted=0)
Nov 29 12:49:53 raspibolt lnd[2007]: 2021-11-29 12:49:53.896 [INF] WTCL: (anchor) Client stats: tasks(received=0 accepted=0 ineligible=0) sessions(acquired=0 exhausted=0)
Nov 29 12:49:59 raspibolt lnd[2007]: 2021-11-29 12:49:59.189 [INF] CRTR: Processed channels=0 updates=125 nodes=1 in last 1m0.000174745s
Nov 29 12:50:48 raspibolt lnd[2007]: 2021-11-29 12:50:48.120 [INF] DISC: Broadcasting 162 new announcements in 17 sub batches
Nov 29 12:50:53 raspibolt lnd[2007]: 2021-11-29 12:50:53.896 [INF] WTCL: (anchor) Client stats: tasks(received=0 accepted=0 ineligible=0) sessions(acquired=0 exhausted=0)
Nov 29 12:50:53 raspibolt lnd[2007]: 2021-11-29 12:50:53.896 [INF] WTCL: (legacy) Client stats: tasks(received=0 accepted=0 ineligible=0) sessions(acquired=0 exhausted=0)
Nov 29 12:50:59 raspibolt lnd[2007]: 2021-11-29 12:50:59.189 [INF] CRTR: Processed channels=1 updates=99 nodes=10 in last 59.999451866s
Nov 29 12:51:53 raspibolt lnd[2007]: 2021-11-29 12:51:53.895 [INF] WTCL: (anchor) Client stats: tasks(received=0 accepted=0 ineligible=0) sessions(acquired=0 exhausted=0)
Nov 29 12:51:53 raspibolt lnd[2007]: 2021-11-29 12:51:53.895 [INF] WTCL: (legacy) Client stats: tasks(received=0 accepted=0 ineligible=0) sessions(acquired=0 exhausted=0)
Nov 29 12:51:59 raspibolt lnd[2007]: 2021-11-29 12:51:59.189 [INF] CRTR: Processed channels=0 updates=93 nodes=8 in last 59.999803116s
Nov 29 12:52:18 raspibolt lnd[2007]: 2021-11-29 12:52:18.120 [INF] DISC: Broadcasting 178 new announcements in 18 sub batches
```

But in case I will fund my wallet with admin user i get this:

```
lncli newaddress p2wkh
[lncli] could not load global options: could not load TLS cert file: open /home/admin/.lnd/tls.cert: no such file or directory
```

VajraOfIndra, 2021-11-29:
Do you have the .lnd symlink in the "admin" user home directory?

n-e-t-d-i-v-e-r, 2021-11-29:
As far as I understand yes:

admin@raspibolt:/home $ ls -la
```
total 32
drwxr-xr-x  8 root           root           4096 Nov 28 09:32 .
drwxr-xr-x 19 root           root           4096 Nov 25 08:53 ..
drwxr-xr-x  6 admin          admin          4096 Nov 28 09:39 admin
drwxr-xr-x  3 bitcoin        bitcoin        4096 Nov 25 07:49 bitcoin
drwxr-xr-x  7 btcrpcexplorer btcrpcexplorer 4096 Nov 28 09:04 btcrpcexplorer
drwxr-xr-x  3 electrs        electrs        4096 Nov 27 08:59 electrs
drwxr-xr-x  2 lnd            lnd            4096 Nov 28 09:34 lnd
drwxr-xr-x  2 pi             pi             4096 Nov 25 07:24 pi
```

VajraOfIndra, 2021-11-29:
> admin@raspibolt:/home $ ls -la

It wold be in `/home/admin/`.
Could you run this wuth user "admin":

```
$ cd ~/
$ ls -la
```

n-e-t-d-i-v-e-r, 2021-11-29:
admin@raspibolt:~ $ cd ~/
admin@raspibolt:~ $ ls -la
```
total 52
drwxr-xr-x 6 admin admin 4096 Nov 28 09:39 .
drwxr-xr-x 8 root  root  4096 Nov 28 09:32 ..
-rw------- 1 admin admin 7904 Nov 29 06:14 .bash_history
-rw-r--r-- 1 admin admin  220 Nov 25 07:23 .bash_logout
-rw-r--r-- 1 admin admin 3539 Nov 25 07:25 .bashrc
drwxr-xr-x 3 admin admin 4096 Nov 27 08:22 .cargo
drwx------ 4 admin admin 4096 Nov 28 09:31 .gnupg
drwxr-xr-x 3 admin admin 4096 Nov 28 09:39 .local
-rw-r--r-- 1 admin admin 1670 Nov 25 07:23 .mkshrc
-rw-r--r-- 1 admin admin  807 Nov 25 07:23 .profile
-rw-r--r-- 1 admin admin  317 Nov 28 09:31 .wget-hsts
drwxr-xr-x 3 admin admin 4096 Nov 27 08:18 rust
```

VajraOfIndra, 2021-11-29:
> admin@raspibolt:~ $ cd ~/ admin@raspibolt:~ $ ls -la

It seems you don't have the lnd symlink there. When you invoke` lncli` with user "admin", the client tries to go to `/home/admin/.lnd/tls.cert` but can't find `.lnd` in /home/admin.

You could simply use the "lnd" user instead of admin as it seems to be recommended in the guide right now. 
Otherwise, you'd have to add the symlink in the /home/admin folder and then also make the "admin" user a member of the "lnd" group to give permissions (but I'm still unsure on how all the permissions work so better have @Stadicus comment on the above points).

n-e-t-d-i-v-e-r, 2021-11-29:
As far as I remember there is no call in the manual to add a lnd symlink to /home/admin directory or was there something i missed?

Anyway should i delete everything and restart installation from the scretch or can you enter me here the right commands/calls which help me to fix this issue and add a symlink to /home/admin with this tls.cert inside and the right permissions? I lost the overview ... ;)

VajraOfIndra, 2021-11-29:
No you're right, there is no such instructions in RaspiBolt v3.

I personally wouldn't try to add a symlink for the "admin" user, I would just use the "lnd" user.
Did you try to create a new address with the "lnd" user? should work without isssue.

n-e-t-d-i-v-e-r, 2021-11-29:
BTW ligthning node is up but there is no connection to the wallet and also no lincli connect link:
[image]

n-e-t-d-i-v-e-r, 2021-11-29:
Tried to create a new address with lnd user session:

lnd@raspibolt:~ $ lncli create
```
Input wallet password: 
Confirm password: 

Do you have an existing cipher seed mnemonic or extended master root key you want to use?
Enter 'y' to use an existing cipher seed mnemonic, 'x' to use an extended master root key 
or 'n' to create a new seed (Enter y/x/n): n

Your cipher seed can optionally be encrypted.
Input your passphrase if you wish to encrypt it (or press enter to proceed without a cipher seed passphrase): 

Generating fresh cipher seed...

[lncli] unable to generate seed: rpc error: code = Unknown desc = wallet already unlocked, WalletUnlocker service is no longer available
```

Stadicus, 2021-11-29:
I think we should keep the focus of issues small, close them if solved, and open new ones.

So, for the original question: it works with user "lnd", as it should). But does not with user "admin" which is expected, as mentioned above:

> User "admin" can't use lncli at the moment, due to missing access privileges the same directories.
>
> There are some ways to do that, not sure if it's a) necessary, and b) which one is best:
>
> 1. [...]

Giving user "admin" permissions to use `lncli` can be a good thing, but not part of the current guide. We'd need to decide on the method to use. I'd prefer number 1. --> new issue, feature request

> Tried to create a new address with lnd user session:
> lnd@raspibolt:~ $ lncli create

With that, you create a new wallet, not a new address: 

```
$ $ lncli help create
NAME:
   lncli create - Initialize a wallet when starting lnd for the first time.

USAGE:
   lncli create [command options] [arguments...]

CATEGORY:
   Startup

DESCRIPTION:
   
  The create command is used to initialize an lnd wallet from scratch for
  the very first time. This is interactive command with one required
  argument (the password), and one optional argument (the mnemonic
  passphrase).
```

n-e-t-d-i-v-e-r, 2021-11-29:
OK. I see and what is the right command/call to create a new address? Can’t finde something like this in the manual.

Stadicus, 2021-11-29:
Bitcoin on-chain address or Lightning invoice? 

* Bitcoin: `newaddress` https://raspibolt.org/lnd.html#funding-your-lightning-node
* Invoice: `addinvoice` https://raspibolt.org/lnd.html#more-commands

n-e-t-d-i-v-e-r, 2021-11-30:
lnd node is up and running now:

```
* lnd.service - LND Lightning Network Daemon
     Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2021-11-29 19:14:38 GMT; 10h ago
   Main PID: 629 (lnd)
      Tasks: 11 (limit: 8986)
        CPU: 26min 51.979s
     CGroup: /system.slice/lnd.service
             `-629 /usr/local/bin/lnd
```
everything seems to be OK:
```
-- Journal begins at Sat 2021-10-30 13:43:13 BST. --
Nov 30 06:03:08 raspibolt lnd[629]: 2021-11-30 06:03:08.541 [INF] WTCL: (anchor) Client stats: tasks(received=0 accepted=0 ineligible=0) sessions(acquired=0 exhausted=0)
Nov 30 06:03:40 raspibolt lnd[629]: 2021-11-30 06:03:40.209 [INF] CRTR: Processed channels=0 updates=106 nodes=0 in last 59.999928033s
Nov 30 06:04:08 raspiboltlnd[629]: 2021-11-30 06:04:08.541 [INF] WTCL: (anchor) Client stats: tasks(received=0 accepted=0 ineligible=0) sessions(acquired=0 exhausted=0)
Nov 30 06:04:08 raspibolt lnd[629]: 2021-11-30 06:04:08.541 [INF] WTCL: (legacy) Client stats: tasks(received=0 accepted=0 ineligible=0) sessions(acquired=0 exhausted=0)
Nov 30 06:04:10 raspibolt lnd[629]: 2021-11-30 06:04:10.692 [INF] DISC: Broadcasting 193 new announcements in 18 sub batches
Nov 30 06:04:40 raspibolt lnd[629]: 2021-11-30 06:04:40.209 [INF] CRTR: Processed channels=0 updates=165 nodes=1 in last 59.999598512s
```

And  I was able to open a new wallet in a lnd user session:
```
{
    "total_balance": "0",
    "confirmed_balance": "0",
    "unconfirmed_balance": "0",
    "account_balance": {
        "default": {
            "confirmed_balance": "0",
            "unconfirmed_balance": "0"
        }
    }
}
```

Next adventure will be funding this wallet and open and connect some channels. Thx for supportting me and maybe it And sorry that I took your time. the «issue» was that i was used to the admin being able to start lnd and create wallets from version 2.

n-e-t-d-i-v-e-r, 2021-11-30:
Thx for supportting me and maybe it And sorry that I took your time. the «issue» was that i was used to the admin being able to start lnd and create wallets from version 2.

Stadicus, 2021-11-30:
No worries, I was just struggling to keep up with the different issues mentioned all in the same thread... :)

I also think admin should be able to run `lncli`, I'll try to update the guide soonish.