raspibolt/raspibolt #806: SSD or Sdkarte

Ubuntu51, 2021-11-29:
hallo people,
I'm 53 years old and tried to rebuild after your lead.
first problem what I had is that I tried it with a ssd and a memory card ds it was broken off with the time since the place was too little, clear.
the second problem is that you write you should connect to an external ssd the image of os raspberypi then to one of the usb and start, the raspberrypi can not just start from an usb without an entry in the boot.txt please help what I don't understand here thank you.

Stadicus, 2021-11-29:
The Raspberry Pi Imager takes care of all the boot configuration. You should be able to just write Raspberry Pi OS to your SSD, connect it as a USB3 device (no microSD card connected), and start the Pi.

Ubuntu51, 2021-11-29:
hello, thank you for your answer but that can't be that does it automatically when I read in the internet it must rewrite the boot.config.txt that it starts from the usb. more detailed information where I can read this. have a rpi 4 with 8GB ram and a 1 tb scandisk ssd
Thank you.

Ubuntu51, 2021-11-29:
> The Raspberry Pi Imager takes care of all the boot configuration. You should be able to just write Raspberry Pi OS to your SSD, connect it as a USB3 device (no microSD card connected), and start the Pi.

hello, thank you for your answer but that can't be that does it automatically when I read in the internet it must rewrite the boot.config.txt that it starts from the usb. more detailed information where I can read this. have a rpi 4 with 8GB ram and a 1 tb scandisk ssd
Thank you.

Stadicus, 2021-11-29:
Have you tried it? This is a more recent feature, and previously, it really was not very straight-forward to do that. I tried it with many different SSD models, and so did other. It even works booting from a standard USB thumbdrive.

Ubuntu51, 2021-11-29:
> Have you tried it? This is a more recent feature, and previously, it really was not very straight-forward to do that. I tried it with many different SSD models, and so did other. It even works booting from a standard USB thumbdrive.

yes it tried with two different ssd, two raspberrypi and two different cables that go to ssd. will try it tomorrow night with a normal usb stick. I could hang the ssd normal in mnt and copy the bitcoin kernel on it would have to go or
thank you for your help

Stadicus, 2021-11-30:
If booting from SSD does not work for you, it's pretty easy to use the microSD card. I"ll probably write a bonus guide to add this as a fallback option.

Do everything like the guide mentions, but simply mount the external drive to `/data` as explained in the version 2 guide here:  
https://v2.raspibolt.org/raspibolt_20_pi.html#attach-external-drive

Ubuntu51, 2021-11-30:
> If booting from SSD does not work for you, it's pretty easy to use the microSD card. I"ll probably write a bonus guide to add this as a fallback option.
>
> Do everything like the guide mentions, but [...]

Good Morning,
Thanks for the help, I also did an rpi update yesterday, unfortunately everything did not lead to success, I will try a USB in the evening, just for the sake of interests. At the weekend I will try the V2, if everything works, I will then get back to you Thank you

Stadicus, 2021-11-30:
I just created a pull request for a new bonus guide. Feel free to test, it should solve your issue:

https://github.com/raspibolt/raspibolt/pull/807/files#diff-d32d1001912ece545fee2e93e6a71a7e85d50cefb2b01d053da6cf9504cff7ad

Ubuntu51, 2021-11-30:
> I just created a pull request for a new bonus guide. Feel free to test, it should solve your issue:
> 
> https://github.com/raspibolt/raspibolt/pull/807/files#diff-d32d1001912ece545fee2e93e6a71a7e85d50cefb2b01d053da6cf9504cff7ad

Thank you very much will try and report
Thank you

Ubuntu51, 2021-12-01:
> > I just created a pull request for a new bonus guide. Feel free to test, it should solve your issue:
> > https://github.com/raspibolt/raspibolt/pull/807/files#diff-d32d1001912ece545fee2e93e6a71a7e85d50cefb2b01d053da6cf9504cff7ad
> 
> Thank you very much will try and report Thank you

closed 2021-12-01 by Ubuntu51

Ubuntu51, 2021-12-01:
> I just created a pull request for a new bonus guide. Feel free to test, it should solve your issue:
> 
> https://github.com/raspibolt/raspibolt/pull/807/files#diff-d32d1001912ece545fee2e93e6a71a7e85d50cefb2b01d053da6cf9504cff7ad

the bircoin core was downloaded but I found a short way to incorporate a ssd.
thanks again for the help will start with the lnd.

#ssd search
lsblk
#ssd mount
sudo mount /dev/sda1 /mnt/
#typing on
sudo chmod 775 /mnt/

cd /mnt
sudo mkdir /data/
#auto mount
sudo nano /etc/fstab

/dev/sda1 /mnt ext4 defaults 0 0

reboot

if there should be a question I would like to make

Stadicus, 2021-12-03:
Sounds good! Mouting via device name (e.g, `dev/sda1`) is not quite as stable because the name can change if you connect other drives or a USB thumbdrive. But as long as it works, it's a good solution as well.

Ubuntu51, 2021-12-03:
> Sounds good! Mouting via device name (e.g, `dev/sda1`) is not quite as stable because the name can change if you connect other drives or a USB thumbdrive. But as long as it works, it's a good solution as well.

that is absolutely right what they write.I only use the raspibolt and for this purpose it should be enough.
Thank you

PR #807 (2021-12-02): bonus: boot from microsd card
#### What

This change adds a bonus guide on how to boot from a microSD card and mount an external drive to /data/ to use for data storage. This method should work with the broadest range of devices.

### Why

The new RaspiBolt version 3 assumes that all external drives can be used to boot the Raspberry Pi.

This seems to hold true for most modern drives, but if this does not work with a specific drive, the reader is stuck without any fallback instructions.

#### How

The instructions from the v2 guide are simplified, and the external drive mounted to `/data/`. Ma [... cut]