raspibolt/raspibolt #808: Verification of tesnet config file

thunderbiscuit, 2021-11-30:
Hi there! Thanks for the awesome guide. I didn't really understand what raspibolt was about until a few days ago when I saw a tweet about it (I just thought it was the ancestor to the Raspiblitz... 🤦 but it's so much more). 

### Describe the issue
I'm trying to setup a testnet node in the cloud, and just want to validate my `bitcoin.conf` file. After some trial and error (you get a "bind error" if you just add `testnet=1` and nothing else to the config file), I settled on this config, which seems to work so far (but I have not set up Electrum nor LND yet, hence my question here):
```txt
# RaspiBolt: bitcoind configuration
# /home/bitcoin/.bitcoin/bitcoin.conf

# Testnet
testnet=1

# Bitcoin daemon
server=1
txindex=1

# File permissions, necessary for cookie authentication
sysperms=1
disablewallet=1

# Network
#listen=1
#listenonion=1
#proxy=127.0.0.1:9050
#bind=127.0.0.1

# Connections
rpcauth=raspiboltadmin:string
zmqpubrawblock=tcp://127.0.0.1:28332
zmqpubrawtx=tcp://127.0.0.1:28333
whitelist=download@127.0.0.1          # for Electrs

# Raspberry Pi optimizations
maxconnections=40
maxuploadtarget=5000

# Initial block download optimizations
dbcache=2000
blocksonly=1

[test]
rpcbind=127.0.0.1
rpcport=18332
listen=1
listenonion=1
proxy=127.0.0.1:9050
bind=127.0.0.1
```
Note that the only changes to the recommended config file from the guide are: `testnet=1`, and the commented out section under `# Network`, which was brought down under the `[test]` section. But I just copy pasted the lines... I'm not sure if they indeed should be there (for example, should `listen=1` and `listenonion=1` be under `[test]`? Is the `proxy=127.0.0.1:9050` the right proxy port for when one uses testnet?)

Thank you! 

### Additional comments
The guide doesn't give the testnet equivalent config file for users who wish to run purely testnet nodes. I'm thinking it would be a great addition to the `Bonus` section, and would be happy to write something up once I got my stuff working!

### Environment
The node is on Debian 11.

Stadicus, 2021-12-03:
When looking at your `bitcoin.conf`, I took the chance to clean up the structure a bit. Your config looks good, but there's some redundancy.

I think this would be a bit cleaner and easier to switch between chains:

```ini
# [chain]
# main, test, signet, regtest
chain=test

# [core]
sysperms=1
blocksonly=1
dbcache=2000
txindex=1

# [wallet]
disablewallet=1

# [network]
listen=1
listenonion=1
proxy=127.0.0.1:9050
maxconnections=40
maxuploadtarget=5000
whitelist=download@127.0.0.1          # for Electrs

# [rpc]
rpcauth=raspiboltadmin:string
server=1

# [zeromq]
zmqpubrawblock=tcp://127.0.0.1:28332
zmqpubrawtx=tcp://127.0.0.1:28333

# Options specific to chain
# ('rpcport' shows default values)
#
[main]
#rpcport=8332
bind=127.0.0.1
rpcbind=127.0.0.1

[test]
#rpcport=18332
bind=127.0.0.1
rpcbind=127.0.0.1

[signet]
#rpcport=38332

[regtest]
#rpcport=18443
```

thunderbiscuit, 2021-12-03:
Thank you! I had my sections mixed up a bit.

I'm working through some of the book [Learning Bitcoin from the Command Line](https://github.com/BlockchainCommons/Learning-Bitcoin-from-the-Command-Line) and one problem I was running into is that if `disablewallet` is commented out in order to generate wallets, the systemctl bitcoind service errors out on restart. After fiddling around a bit I found that it appears to be due to the `sysperms=1` config, which when commented out fixes the issue. This, however, turns off the ability for the cookie to work and the `bitcoin-cli` to be accessed without a password (I now have to provide `-rpcuser` and `-rpcpass` on commands).

I'm wondering if you can confirm that my understanding is correct:
1. `disablewallet=0` requires you also set `sysperms=0`
2. this in turns prevents the "auto-authentication" that is otherwise happening with raspibolt when you follow the guide on mainnet

Thank you!

Stadicus, 2021-12-04:
Your understanding is 100% correct. It's necessary to use `sysperms` to allow the authentication cookie to be readable by group "bitcoin", otherwise it's only accessible to the user "bitcoin". Because the file is newly created every time bitcoins starts, it's not possible to change the permissions permanently (unless with some [hacky systemd scripts](https://github.com/raspibolt/raspibolt/issues/801#issuecomment-981748019)). 

Disabling the wallet breaks compatibility with several other projects, like JoinMarket or EPS, which is a pity. But I'm not aware of a clean way to enable group access permissions without using `sysperms`.

Stadicus, 2021-12-10:
Great, thanks! :+1:

thunderbiscuit, 2021-12-10:
I'm unable to connect a Sparrow wallet to my testnet node using Tor with my current configuration. I think it's on my RaspiBolt side of things that my problem is, because I can connect my Sparrow to other nodes using Tor (only providing a Tor address and a port).

Here are some of the things I realize are not clear for me yet:
1. Is Tor affected by by `ufw` or does it circumvent it? I have opened up the port just in case but it's not making a difference
2. Is the NGINX reverse proxy also affecting Tor connections? In other words, should I give Sparrow port `60001` or `60002`? (I tried both but no luck so far)

My current mental model is that you connect using Tor to the `60002` port, which NGINX forwards to `60001`, which is the local electrs RPC port.

Stadicus, 2021-12-10:
> Is Tor affected by by ufw?

No, Tor establishes the connections going out, no need to listen for incoming connections.

> Is the NGINX reverse proxy also affecting Tor connections?

Yes, the config above accepts SSL port `60002` over Tor, and NGINX routes it internally to port `60001` without SSL. That should work as it does for mainnet (can't test currently, though). You can create another Tor hidden service without SSL in the `/etc/tor/torrc` file to try:

```
HiddenServiceDir /var/lib/tor/hidden_service_electrs_nossl/
HiddenServiceVersion 3
HiddenServicePort 60001 127.0.0.1:60001
```

thunderbiscuit, 2021-12-10:
Thanks @Stadicus! I had not understood I needed to turn on the SSL toggle even with Tor. Your recommendation is I think a good step in the right direction, because now I'm getting the same error in Sparrow on both (`60001` with the SSL `off` and `60002` with the SSL `on`).

Stadicus, 2021-12-11:
I might have found the issue: using `blocksonly` for Bitcoin Core causes it to disable fee estimation. That is then mirrored by Electrs.

Can you check if you have that enabled? And if yes, try to disable it in `bitcoin.conf` by commenting it out or setting it to `#blocksonly=0`?

thunderbiscuit, 2021-12-11:
That was it!  Cheers.

Stadicus, 2021-12-11:
Nice! I'm linking the Sparrow wallet issue for completeness https://github.com/sparrowwallet/sparrow/issues/354 before closing

closed 2021-12-11 by Stadicus

Stadicus, 2021-12-11:
I extended comment https://github.com/raspibolt/raspibolt/issues/808#issuecomment-989910246 slightly:

* added `cookie_file = "/data/bitcoin/testnet3/.cookie"` to `electrs.conf`
* added file locations

I think that would make a neat bonus guide :)