raspibolt/raspibolt #815: [BUG] Block Explorer version / update status in MOTD is wrong if auth password is set

kristapsk, 2021-12-02:
### Describe the bug
If `BTCEXP_BASIC_AUTH_PASSWORD` in `/home/btcrpcexplorer/btc-rpc-explorer/.env` is set, version number / update status info in MOTD is wrong.
[image]

Reason is that `curl -s http://127.0.0.1:3002/tools` in such case will return `Unauthorized`.

Simplest solution likely would be to ignore blank `btcrpcexplorerversion` version (result of `curl -s http://127.0.0.1:3002/tools | grep -oP '(?<=version: )[^</]*' | awk '{print "v"$1}'`). Will not be ideal but will not show wrong "Update!" warning. Alternative approach is to detect version number differently (compare checked out git tag with most recent version?).

### Location in guide
https://raspibolt.org/bonus/raspberry-pi/system-overview.html

Stadicus, 2021-12-03:
Thanks, good point. I'm currently working on updating the MOTD in this staging branch and can incorporate this change:

https://github.com/raspibolt/raspibolt/tree/welcome-script-staging

Might be a while, though, as I feel I need to improve the overall structure and code quality of the script quite a bit. :)

VajraOfIndra, 2021-12-03:
> Might be a while, though, as I feel I need to improve the overall structure and code quality of the script quite a bit. :)

What about we add a warning at the top the bonus guide to let users that the guide is not up-to-date with v3 yet/contains bugs  and that a new version will come soon?

viertelb, 2021-12-14:
@Stadicus I recognized the link to the script is currently unavaillable in your guide. I was trying to understand why checking my bitcoin status with the alternate option (NAT not beeing available) does not work. Maybe you could also check if this line works as expected `#Choose the top two lines if you experience 'not reachable' while blockchain syncs. Could be due to nat-loopback being not available. 
public_check=$(curl -s https://bitnodes.io/api/v1/nodes/me-${public_port}/ | jq .success)`

Also, adding a message that the script is currently under development might be good for the overall impression of the otherwise awesome guide.

closed 2021-12-21 by Stadicus