raspibolt/raspibolt #819: [FEATURE REQ] Add RaspiBolt v3.0 to the repo 'Releases' section

VajraOfIndra, 2021-12-03:
### What & why

The 'Releases' section of the repo is still showing v2.1.  
Since v3 is now live we should probably update the section by adding this latest version.

### How

Create new release.

### Additional context

N/A

Stadicus, 2021-12-04:
Yes, good point. I thought I'd give it some time in case there are still some bugs, but I think we're getting there :)

Stadicus, 2021-12-04:
Added a PRE-RELEASE for now:

https://github.com/raspibolt/raspibolt/releases/tag/3.0-pre-release

VajraOfIndra, 2021-12-04:
> I thought I'd give it some time in case there are still some bugs

I see, makes sense!

> Added a PRE-RELEASE for now:

Nice :)

Stadicus, 2021-12-04:
Let's leave it open as a reminder to actually create the v3 soonish... :)

Stadicus, 2021-12-07:
RaspiBolt v3 officially tagged! https://github.com/raspibolt/raspibolt/releases/tag/3.0

closed 2021-12-07 by Stadicus